marvinroger/async-mqtt-client #93: Strange symbols on publishing and subscribing

dejanra, 2018-03-20:
Library is great and all working, but I get strange symbols on 'publishing' and 'subscribing', like this (Serial Monitor):

Connecting to Wi-Fi...
Connected to Wi-Fi.
Connecting to MQTT...
Connected to MQTT.
Session present: 1
Subscribing at QoS 1, packetId: 1
Publishing at QoS 1, packetId: 2
Subscribe acknowledged.
  packetId: 1
  qos: 1
test/lol
test 2@                             should be only test 2
Publish acknowledged.
  packetId: 2
test/lol
test 2
Publish acknowledged.
  packetId: 3
Publish acknowledged.
  packetId: 4
SSDJ/ZVONO/SWITCH
ON⸮a⸮ @򜁇 ⸮ ⸮                     should be only ON
Publish acknowledged.
  packetId: 5
SSDJ/ZVONO/SWITCH
OFF j⸮ A⸮⸮⸮ ⸮ ⸮                    should be only OFF
Publish acknowledged.
  packetId: 6

marvinroger, 2018-03-20:
You've been contacting me by e-mail, I don't have time for support there. 😉 
The `payload` field is a char array, not an actual NULL terminated string, hence what you see.

If you want to print it, you have to allocate a char array and NULL terminate it yourself.

See https://github.com/marvinroger/async-mqtt-client/issues/73

closed 2018-03-20 by marvinroger

dejanra, 2018-03-20:
Thank You, verry much!

dejanra, 2018-03-20:
Thank You, verry much!

Regards!

Dejan Rakijašić