marvinroger/async-mqtt-client #94: wifimanager + asyncmqttclient = problem!

sohaibilyas, 2018-03-21:
first of all thank you so much for this amazing library.

the thing is i dont want to hard-code the wifi details in the sketch so i decided to use wifimanager with your library asyncmqttclient and instead of using WiFi.begin() function i used wifiManager.autoConnect() in connectToWifi() function and now esp8266 keeps resetting specially when autoConnect() function cant connect to last connected wifi so please could you give me some hint or if you already have used both libraries together then share with us the code, thanks.

timpur, 2018-03-21:
Can you send us the stack dump decoded ?

sohaibilyas, 2018-03-21:
Decoding 26 results
0x402013b4: __pinMode at C:\Users\roomi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 55 (discriminator 1)
0x402013bf: __pinMode at C:\Users\roomi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 55 (discriminator 3)
0x40203600: system_get_chip_id at ?? line ?
0x4020360e: system_get_chip_id at ?? line ?
0x402033f4: system_restoreclock at ?? line ?
0x40207ab5: espconn_tcp_accept at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/app/espconn_tcp.c line 1253
0x402035c0: system_print_meminfo at ?? line ?
0x402035ad: system_print_meminfo at ?? line ?
0x40207e53: espconn_kill_oldest at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/app/espconn_tcp.c line 79
:  (inlined by) espconn_kill_oldest_pcb at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/app/espconn_tcp.c line 137
0x4020be63: tcp_output at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 924
0x4020beaf: tcp_output at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 983
0x4020bee1: tcp_output_segment at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 1066
:  (inlined by) tcp_output at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 988
0x4020bf2e: tcp_output_segment at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 1071
:  (inlined by) tcp_output at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 988
0x40208f47: dhcp_bind at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/dhcp.c line 983
0x4020bb00: tcp_write at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 599
0x401045e4: rcGetRate at ?? line ?
0x402024fd: ets_vsnprintf at ?? line ?
0x4020c84b: sys_untimeout at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c line 365

tarzan115, 2018-03-22:
@sohaibilyas  can you give me more detail, such as a simple sketch to reproduce your problem.

sohaibilyas, 2018-03-22:
using this example https://github.com/marvinroger/async-mqtt-client/blob/master/examples/FullyFeatured-ESP8266/FullyFeatured-ESP8266.ino i just commented the WiFi.begin() and used the wifiManager.autoConnect() function:

void connectToWifi() {
  Serial.println("Connecting to Wi-Fi...");
  wifiManager.setAPStaticIPConfig(IPAddress(1,1,1,1), IPAddress(1,1,1,1), IPAddress(255,255,255,0));
  wifiManager.autoConnect("eSwitchy", "12345678");
  Serial.println("connected...yeey :)");
  // WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
}

if i use WiFi.begin() then its fine but once again i dont want to hard-code wifi details

tarzan115, 2018-03-23:
@sohaibilyas `connectToWifi()` is called in a callback of an event. So you can't `delay` or run `while` too long inside.  
Method `autoConnect()` is run `while(1)` inside.
You can resolve by adding `wifiManager` stub before this line https://github.com/marvinroger/async-mqtt-client/blob/0605524875dd4bf3004282812a203f8189d83005/examples/FullyFeatured-ESP8266/FullyFeatured-ESP8266.ino#L107 and left `connectToWiFi()` use `Wi-Fi.begin()`

sohaibilyas, 2018-03-23:
okay it did solve that error but it will also keep trying to connect to wifi using connectToWifi() @ wifiReconnectTimer.once(3, connectToWifi);

and why should i still use WiFi.begin()!

tarzan115, 2018-03-26:
please read about `interrupt` to know more information.

ristomatti, 2018-04-09:
@sohaibilyas Have you tried with https://github.com/alanswx/ESPAsyncWiFiManager ?

bertmelis, 2021-02-24:
Nothing indicates an issue with this lib.

closed 2021-02-24 by bertmelis