marvinroger/async-mqtt-client #96: Receiving (subscribing) message does not working anymore

dejanra, 2018-04-01:
All was working (publish and subscribe) great, but suddenly my ESP8266 (Wemos D1 mini pro) could not receiving message. It seems that 'onMqttMessage' function des not work anymore. It is strange, becouse Wemos send (publish) data normally. I don't know is it problem with update core or library. I'am using Arduino IDE 1.8.5. and ESP core 2.4.1. and latest async MQTT library. 

Other library - pubsubclinet.h and  -MQTT.h (async) work normally with IDE1.8.5. and core 2.4.1. But Marvin's async MQTT library is best for my project, so please help!

tarzan115, 2018-04-02:
could you please write small of your code. it better to help.

dejanra, 2018-04-02:
#include <ESP8266WiFi.h>
#include <Ticker.h>
#include <AsyncMqttClient.h>

#define WIFI_SSID "**************"
#define WIFI_PASSWORD "**********************"

char mqtt_server[40] = "*****************";
char mqtt_port[6] = "*************";
char mqtt_user[40] = "**************";
char mqtt_pass[60] = "**********************";

#define RELAY_PIN D6

#define MQTT_RELAY1_TOPIC     "SSDJ_zvono/Relay"
#define MQTT_FEEDBACK1_TOPIC  "SSDJ_zvono/RelayFB"
#define MQTT_LASTWILL_TOPIC   "SSDJ_zvono/offline"

AsyncMqttClient mqttClient;
Ticker mqttReconnectTimer;

WiFiEventHandler wifiConnectHandler;
WiFiEventHandler wifiDisconnectHandler;
Ticker wifiReconnectTimer;

void connectToWifi() {
  wifiReconnectTimer.detach();
  Serial.println("Connecting to Wi-Fi...");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
}

void onWifiConnect(const WiFiEventStationModeGotIP& event) {
  Serial.println("Connected to Wi-Fi.");
  connectToMqtt();
}

void onWifiDisconnect(const WiFiEventStationModeDisconnected& event) {
  Serial.println("Disconnected from Wi-Fi.");
  mqttReconnectTimer.detach(); // ensure we don't reconnect to MQTT while reconnecting to Wi-Fi
  wifiReconnectTimer.attach(2, connectToWifi);
}

void connectToMqtt() {
  Serial.println("Connecting to MQTT...");
  mqttReconnectTimer.detach();
  mqttClient.connect();
}

void onMqttConnect(bool sessionPresent) {
  Serial.println("Connected to MQTT.");
  Serial.print("Session present: ");
  Serial.println(sessionPresent);
  uint16_t packetIdSub = mqttClient.subscribe("test/lol", 1);
  Serial.print("Subscribing at QoS 1, packetId: ");
  Serial.println(packetIdSub);
  uint16_t packetIdPub1 = mqttClient.publish("test/lol", 1, true, "test 2");
  Serial.print("Publishing at QoS 1, packetId: ");
  Serial.println(packetIdPub1);
}

void onMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
  Serial.println("Disconnected from MQTT.");
  if (WiFi.isConnected()) {
    Serial.println("Connecting to MQTT...");
    mqttReconnectTimer.attach(2, connectToMqtt);
  }
}

void onMqttSubscribe(uint16_t packetId, uint8_t qos) {
  Serial.println("Subscribe acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
  Serial.print("  qos: ");
  Serial.println(qos);
}

void onMqttUnsubscribe(uint16_t packetId) {
  Serial.println("Unsubscribe acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
}

void onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {
   Serial.println(topic);
  Serial.println(payload);
char new_payload[len+1];
new_payload[len] = '\0';
strncpy(new_payload, payload, len);
Serial.print("**new_payload: ");
  Serial.println(new_payload);
if((String(topic) == MQTT_RELAY1_TOPIC) && (String(new_payload) =="ON"))
{
   digitalWrite(RELAY_PIN, HIGH);   
    mqttClient.publish(MQTT_FEEDBACK1_TOPIC, 1, false, new_payload);
}
else
{
  mqttClient.publish(MQTT_FEEDBACK1_TOPIC, 1, false, new_payload);  
  digitalWrite(RELAY_PIN, LOW);
}
}
void onMqttPublish(uint16_t packetId) {
  Serial.println("Publish acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
}

void setup() {
  Serial.begin(115200);

  pinMode(RELAY_PIN, OUTPUT);
  digitalWrite(RELAY_PIN, LOW);
  
  wifiConnectHandler = WiFi.onStationModeGotIP(onWifiConnect);
  wifiDisconnectHandler = WiFi.onStationModeDisconnected(onWifiDisconnect);

  mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  mqttClient.onSubscribe(onMqttSubscribe);
  mqttClient.onUnsubscribe(onMqttUnsubscribe);
  mqttClient.onMessage(onMqttMessage);
  mqttClient.onPublish(onMqttPublish);
  mqttClient.setServer(mqtt_server, atoi(mqtt_port));
  mqttClient.setCredentials(mqtt_user, mqtt_pass);
  mqttClient.setKeepAlive(5).setCleanSession(false).setWill(MQTT_LASTWILL_TOPIC, 2, true, "zadnja poruka").setCredentials(mqtt_user, mqtt_pass).setClientId("m");
  Serial.println("Connecting to MQTT...");
  connectToWifi();
}

void loop() {
}

tarzan115, 2018-04-02:
I had seen nothing wrong. maybe your topic when you publish data does not match in your code.
please give me log when you run these code.

dejanra, 2018-04-02:
Topic is the same, I did not change nothing, I am using cloudMQTT and MQTTdash on my Android phone. This is happened suddenly afther update code, becouse I was planing save some parameters in EEPROM.

Serial Monitor:

Connecting to Wi-Fi...
Connected to Wi-Fi.
Connecting to MQTT...
Connected to MQTT.
Session present: 1
Subscribing at QoS 1, packetId: 1
Publishing at QoS 1, packetId: 2
Subscribe acknowledged.
  packetId: 1
  qos: 1
test/lol
test 2@
**new_payload: test 2
Publish acknowledged.
  packetId: 2
test/lol
test 2nU⸮>:⸮
**new_payload: test 2
Publish acknowledged.
  packetId: 3
Publish acknowledged.
  packetId: 4

Wemos can send data, but not receive. Topic 'MQTT_RELAY1_TOPIC' and function 'onMqttMessage' not working.

tarzan115, 2018-04-02:
following your code, I just saw you subscribe only `test/lol` topic not `MQTT_RELAY1_TOPIC`
```cpp
void onMqttConnect(bool sessionPresent) {
  Serial.println("Connected to MQTT.");
  Serial.print("Session present: ");
  Serial.println(sessionPresent);
  uint16_t packetIdSub = mqttClient.subscribe("test/lol", 1);
  Serial.print("Subscribing at QoS 1, packetId: ");
  Serial.println(packetIdSub);
  uint16_t packetIdPub1 = mqttClient.publish("test/lol", 1, true, "test 2");
  Serial.print("Publishing at QoS 1, packetId: ");
  Serial.println(packetIdPub1);
}
```  

try this:
```cpp
void onMqttConnect(bool sessionPresent) {
  Serial.println("Connected to MQTT.");
  Serial.print("Session present: ");
  Serial.println(sessionPresent);

  uint16_t packetIdSub = mqttClient.subscribe("test/lol", 1);
  Serial.print("Subscribing at QoS 1, packetId: ");
  Serial.println(packetIdSub);

  packetIdSub = mqttClient.subscribe(MQTT_RELAY1_TOPIC, 1);
  Serial.print("Subscribing at QoS 1, packetId: ");
  Serial.println(packetIdSub);

  uint16_t packetIdPub1 = mqttClient.publish("test/lol", 1, true, "test 2");
  Serial.print("Publishing at QoS 1, packetId: ");
  Serial.println(packetIdPub1);
}
```  

I hope this code work for you.

dejanra, 2018-04-02:
It' working! Thanks!

closed 2018-04-02 by dejanra

timpur, 2018-04-02:
@tarzan115 thanks.

emnabk123, 2022-03-13:
> #include <ESP8266WiFi.h> #include <Ticker.h> #include <AsyncMqttClient.h>
>
> #define WIFI_SSID "**************" #define WIFI_PASSWORD "**********************"
>
> char mqtt_server[40] = [...]

Hey is it okay if loop function is empty ?

bertmelis, 2022-03-14:
Yes, that is just fine.

emnabk123, 2022-03-14:
> Yes, that is just fine.

Thank you but, would it run everytime i click on the (ON/OFF) button in the "onMqttMessage" function ? or it only runs once? because the role  of void loop is to repeat the code over and over as far as i know.