marvinroger/async-mqtt-client #104: onMessage payload contains extra bytes after data!

robig, 2018-05-01:
Hi
I'm using async-mqtt-client with Basecamp on esp32 boards for a while now.
Until now, i've only subscribed to boolean topics.
Now I'm trying to get numbers. But the thing is, i'm getting more bytes in the char* payload than set on the broker. Seems like the string end char is missing.

Example:
onMqttMessage
66 = f
61 = a
6C = l
73 = s
65 = e
received delaySleep... false
setting delaySleep=false
onMqttMessage
31 = 1
38 = 8
30 = 0
30 = 0
30 = 0
23 = #
1000
received sleeptime... 18000#
setting sleeptime=18000

sleeptime is set to 1800 in the broker.
Somehow its sometimes converted to 18000!

I'm dumping the bytes using this function:
void dump(char *payload)
{
  int i;
  for (i = 0; i < strlen(payload); i++) {
    Serial.print(payload[i], HEX);
    Serial.print(" = ");
    Serial.write(payload[i]);
    Serial.println();
  }
}

I do really have no idea how to handle that!

Any suggestions?

timpur, 2018-05-02:
Sorry, can you try create a raw sketch just using async-mqtt-client from our examples and try replicate, just to eliminate Basecamp as the issue and thus we can also use to test.

ananyas1498, 2019-03-29:
MY SOLUTION:
because len will give the (max of index no+1) ie. we need to write '\0' at last of string by:

payload[len] = '\0';

now you can use payload[].

bertmelis, 2019-03-29:
mqtt payloads are binary and therefore have nothing to do with c-strings. If you want the payloads to ben null-terminated you have to take care yourself.

CyanoFresh, 2019-08-16:
Please add a note with this to documentation:
```c++
payload[len] = '\0';
```

bertmelis, 2019-08-16:
That would be very wrong. payload[len] is not a memory location you should write to.

If you want your payload to be a string then you should properly convert (copy!) it into a string. The raw payload is just a series of bytes.

CyanoFresh, 2019-08-17:
as I know esp8266 has slow memory operations, so copying takes some time and affects rps rate.
string is a series of bytes, isn't it?
can I use strncmp instead of copying?

bertmelis, 2019-08-18:
no. And string is not just a series of bytes. c-string is a series of bytes which is null terminated. String could also be an object (standard or custom implementation).

You can prevent copying but only if your sender guarantees the payload is null-terminated, and even then...

Google a bit about Strings in C and C++ and learn. That how I did it.

If speed is an issue, why would you use MQTT in the first place?

bertmelis, 2021-07-22:
Closing, added a note in the docs

closed 2021-07-22 by bertmelis