marvinroger/async-mqtt-client #105: Half Open Connections Happening Sometimes

arihantdaga, 2018-05-09:
I think there is still a possibility of half open connection in the library.
It happened 3-4 times randomly. I am not able to reproduce this issue, as it is happening sometimes randomly. It happened with different devices.
IN my broker's dashboard it was showing that device is not connected. But the device always pretended like its connected. 
For example this code - 

```
After Every 2 Minutes -

if(myMqttClient.connected()){
    Serial.println("Sending MQTT Message ");
    myMqttClient.publish(SOME_TOPIC,SOME_MESSAGE);
}
```
I set my keepAlive to 30 Seconds and clean session to false. 
In the meanwhile i restarted my broker also, on restarting the broker my all other devices got disconnected and again connected to the broker. 
But this affected device never disconnected or tried to reconnect to the broker. 
I read something about half open connections here - https://www.hivemq.com/blog/mqtt-essentials-part-10-alive-client-take-over I think we may have this issue in this library.

timpur, 2018-05-09:
I'll add it to my list of things to do, but very busy, anyone else here got time to debug. Thanks for the info :)

Cosford, 2018-06-16:
I believe I may also have experienced this issue - the library reports connection to the broker, however does not receive any data coming through. Hopping to a different access point - the missing data floods through (QoS 2).

josep112, 2018-07-02:
I have the same problem, the wifi is connected with you to drib the esp but the mqtt does not work, this happens always after a good time, I think I'll have to change my mqtt library, I really liked this but I do not see a solution at the moment

arihantdaga, 2019-01-07:
I dont understand why this could be happening, but this problem is appearing many times. The Broker Shuts down the connection due to keep alive timeout, but the device keeps thinking that its connected because`mqtt.connected()` is returning true. I don't quiet understand the internals of this function and whats happening inside.
I saw logs on my broker - 

```
Jan  6 13:33:14 emq-broker-2-1c-2g emqttd[2785]: [info] <0.846.7151>@emqttd_session:handle_info:566 Session(dev_807d3a5a43c7): Client <0.26187.4107> EXIT for {shutdown,keepalive_timeout}
Jan  6 15:33:14 emq-broker-2-1c-2g emqttd[2785]: [info] <0.846.7151>@emqttd_session:handle_info:555 Session(dev_807d3a5a43c7): Expired, shutdown now.
```
Note the - `EXIT for {shutdown,keepalive_timeout}` . After which it's been 24 hours i did not plug out the device and it still believes its connected.

mcspr, 2019-06-06:
Just an observation, I did not do a more extensive testing.
Based on xoseperez/espurna#1759. If we physically remove mqtt broker server from the network and continue to keep sending data, we end up filling the network send buffer and essentially leave the connection forever stuck.

onPoll tracks last ping time:
https://github.com/marvinroger/async-mqtt-client/blob/8bd918bfeb2f197afa8b0cb5d278ab2942c62ccb/src/AsyncMqttClient.cpp#L447-L461
But sendPing function will never update last ping time if it can't send ping (as established above):
https://github.com/marvinroger/async-mqtt-client/blob/8bd918bfeb2f197afa8b0cb5d278ab2942c62ccb/src/AsyncMqttClient.cpp#L614-L619
And we end up in a loop, because time last ping time was reset to 0 and next ping ends up stalling because it is never sent.
https://github.com/marvinroger/async-mqtt-client/blob/8bd918bfeb2f197afa8b0cb5d278ab2942c62ccb/src/AsyncMqttClient.cpp#L477

arihantdaga, 2019-06-14:
@mcspr  WIll it result in the esp being stuck in the mqtt loop forever or will it trigger watchdog ? 
In my case i am having a strange problem, Some of my devices suddenly get stuck and the functions in the main loop (led, relays, buttons) etc stops working, however actions from mqtt messages happen just fine.

bertmelis, 2019-06-14:
Are you using ESP 8266 or 32?

arihantdaga, 2019-06-15:
Sorry @bertmelis. I dint mention clearly. :) I am using Esp8266.

mcspr, 2019-06-15:
No resets, just that mqtt poll will continue to work and not notify about the disconnect. And not really forever as it turns out, because the connection will eventually will be dropped when lower level timeout comes up.
**but** that only times-out on older esp8266 2.3.0 platform and current esp32. Running a test right now with esp8266/arduino 2.5.2 and it still thinks it is connected after ~10 min. Maybe lwip2 port breaks something here for esp8266

arihantdaga, 2019-06-18:
@mcspr  I am using latest esp8266 core. And i can verify that device just gets stuck in the mqtt loop forever and no functions from main loop are executed. 
Is there any temporary or permanent solution for this problem ?

arihantdaga, 2019-06-18:
@mcspr Do you think 13a1254e425e35933f8d188b28bb2ef078a0d701 can fix it ?

arihantdaga, 2019-06-18:
@mcspr  you think this 4742a0be9117d67299674fec3695a3c797d72cb0 will fix it ?

arihantdaga, 2019-06-18:
Although i believe that in my case problem is something else. Because  after reading this repository i have understood that there is nothing because of which main loop functions will stop working.

mcspr, 2019-06-18:
For async-mqtt-client it is probably sufficient to move _lastPingRequestTime before the space() check here:
https://github.com/marvinroger/async-mqtt-client/blob/8bd918bfeb2f197afa8b0cb5d278ab2942c62ccb/src/AsyncMqttClient.cpp#L619
Before space() check
https://github.com/marvinroger/async-mqtt-client/blob/8bd918bfeb2f197afa8b0cb5d278ab2942c62ccb/src/AsyncMqttClient.cpp#L612-L615
I think I tried this exact change and it worked for me. Note that ping is sent in (keepAlive * 0.7) seconds and connection will drop in (keepAlive * 2) seconds

If you mean main loop of espurna, we can use the issue tracker there? Since you do not reset, something is calling yield() and we can do some debugging in async context. (e.g. using Ticker to print something). Especially if web or telnet stuff still works

arihantdaga, 2019-06-18:
Yeah. I'll create another issue there for this problem.

mcspr, 2019-06-18:
[Small test](https://gist.github.com/mcspr/be1a92827f71d9702ebb3d800d82f344) also shows that timeout callback comment here is not really true: 
https://github.com/marvinroger/async-mqtt-client/blob/8bd918bfeb2f197afa8b0cb5d278ab2942c62ccb/src/AsyncMqttClient.cpp#L367
https://github.com/me-no-dev/ESPAsyncTCP/blob/7e9ed22ed0078097f895dfee6ebcec90046b81b9/src/ESPAsyncTCP.cpp#L439-L445
disconnect callback is not called, it is up to us to force client close if we want to
(do we? or was it intended to avoid competing with ping?)
```
aaaaaaaaaa608729 connected=1, space=39
aaaaaaaaa609730 connected=1, space=30
aaaaaaaaaa610731 connected=1, space=20
aaaaaaaaaa611732 connected=1, space=10
aaaaaaaaaa612733 connected=1, space=0
aaaaaaaaaa613734 connected=1, space=0
aaaaaaaaaa614735 connected=1, space=0
aaaaaaaaaa615736 connected=1, space=0
aaaaaaaaaa616737 connected=1, space=0
aaaaaaaaaa617738 connected=1, space=0
aaa618101 onTimeout(time=5434)
618121 connect(ip=..., port=1883)
618739 connected=1, space=0
619740 connected=1, space=0
```

bertmelis, 2019-06-19:
So when the ack timeouts (broker is dead) we have to disconnect ourselves.

what happens if we add
```C++
disconnect(true);
```
here https://github.com/marvinroger/async-mqtt-client/blob/8bd918bfeb2f197afa8b0cb5d278ab2942c62ccb/src/AsyncMqttClient.cpp#L370-L374

mcspr, 2019-06-19:
If we add disconnect to onAck, we will disconnect after the first time data is sent.

Since the is no specific documentation for async libs but only lwip, I think this is warranted to explain the flow a bit (and to help my understanding, too)

When we write more frequently than ack timeout (~10 seconds) and also move it's window. Timeout will happen only if write() calls stop being successful. In our case, when space() is zero. Last successful write would be treated as timeout, because network did not receive ack packet for it

Same test on Android phone running `nc -l <ip> 1883` in termux, waiting for the first packet and turning off WiFi. Also we will increase tick time to 15000 ms:
```
20285 connect(ip=10.10.10.50, port=1883)
20286 connected=0, space=0
20443 onConnect()
...reports...
a35285 connected=1, space=1008 < first tick write
35927 onAck(len=64, time=642)
...cut the network on the server...
a50285 connected=1, space=1008 < next tick write
51285 connected=1, space=1008
52285 connected=1, space=1008
53285 connected=1, space=1008
54285 connected=1, space=1008
55285 connected=1, space=1008
55780 onTimeout(time=5495)
...but we are still hanging here, because TCP did not receive any FIN...
```
If we write slower than ack timeout, we will receive onTimeout in ~5 seconds. We can still write() again and it is possible that onTimeout callback will be called again too, so we depend on whether connection restores itself or not. Maybe we can count timeouts as another broken connection metric.

Ofc ack timeout is an arbitrary value set by the AsyncTCP library. LWIP stack tries re-sending the very first failed data packet in the meantime, which is why the connection is still there. Yesterday I tried ESP8266 boards with lwip1.4 and lwip2.1 options.
lwip1.4 retransmits 12 times (TCP_MAXRTX in lwip code) within 15 minutes, increasing time between attempts. Connection is closed after the final one, resulting in onDisconnect call
lwip2 variant retransmits 7 times and 8th time happens after 15 minutes, onDisconnect is called too.

ref:
https://git.savannah.gnu.org/cgit/lwip.git/
https://github.com/espressif/esp-lwip/
https://lwip.fandom.com/wiki/Raw/TCP

bertmelis, 2019-06-19:
What was I thinking? I meant to add the disconnect in the timeout handler.

mcspr, 2019-06-19:
If both sides experience some brief connectivity problems, packets will still go through eventually because of lwip retransmit mechanism. I think the intent was to allow ping to handle that, which should happen twice (since KEEPALIVE*0.7 interval for sendPing?) and only then close the connection.

bertmelis, 2019-06-19:
Yes, but the MQTT keepalive mechanism relies on onPoll, which returns early.

Or am I missing something?

mcspr, 2019-06-19:
onPoll never stops, since the connection is still in ESTABLISHED state
https://lwip.fandom.com/wiki/Raw/TCP#Application_polling
In the other issue I mentioned debug methods for the app to show lwip connection status (DEBUG_MSG == Serial.printf)
https://github.com/mcspr/espurna/blob/4b5b4ac545387f107aaa89afcc21623a6eefc7dc/code/espurna/config/debug.h#L30

bertmelis, 2019-06-20:
it keeps running but our callbacks doesn't get called as _onPoll returns early if there's an ack timeout.

Because this lib does nothing on timeout and completely relies on mqtt's timeout mechanism, it stalls.

mcspr, 2019-06-20:
But why though? There are only two early returns, and _connected is still true during all of this. With https://github.com/mcspr/async-mqtt-client/commit/62bd62e ping time will be updated too.

bertmelis, 2019-06-20:
Asynctcp's internal onPoll returns early and our onPoll doesn't get called when there's an ack timeout.

Only our ontimeout get called but we don't do anything with it.

mcspr, 2019-06-20:
Right. When timeout happens, asynctcp poll it skips poll_cb user callback. But, it resets the state so that timeout_cb is not called next time and poll_cb continues to be called in the future. _close_pcb is false, ack_timeout is cleared, we don't set rx timeout, - so it drops down to the poll_cb call, doesn't it?

And going back a bit, I think the initial change by @arihantdaga was correct.
Per http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc385349238 , we send ping when server looks inactive. With that change we compliment server activity sendPing trigger, and  received ping response packet resets both conditions for send and disconnect. Unlike my patch, which only tries to work with client side activity

oscarsan1, 2020-01-05:
Hi, 
I've found this bug. I write my findings.
I use Windows 10 host with mosquitto server. 
When I stop mosqitto service or reset windows 10, the TCP close correct and Espurna detect very fast the "TCP Disconnected". Espurna try to connect broker every several seconds. When borker is online again Espurna connects without problems. This are the logs:

[016972] [MQTT] TCP Disconnected
[016972] [MQTT] Disconnected!
[021972] [MQTT] Connecting to broker at ---:1883
[021972] [MQTT] Client ID: Nadal
[021972] [MQTT] QoS: 0
[021973] [MQTT] Retain flag: 0
[021974] [MQTT] Keepalive time: 3600s
[021977] [MQTT] Will topic: {Nadal}/status
[028646] [MQTT] TCP Disconnected
[028646] [MQTT] Disconnected!
[038646] [MQTT] Connecting to broker at ----:1883
[038647] [MQTT] Client ID: Nadal
[038647] [MQTT] QoS: 0
[038647] [MQTT] Retain flag: 0
[038648] [MQTT] Keepalive time: 3600s
[038651] [MQTT] Will topic: {Nadal}/status
[045320] [MQTT] TCP Disconnected
[045321] [MQTT] Disconnected!
[060321] [MQTT] Connecting to broker at ----:1883
[060321] [MQTT] Client ID: Nadal
[060321] [MQTT] QoS: 0
[060321] [MQTT] Retain flag: 0
[060323] [MQTT] Keepalive time: 3600s
[060326] [MQTT] Will topic: {Nadal}/status
[060344] [MQTT] Connected!
[060344] [MQTT] Unsubscribing to # (PID 1)
[060345] [MQTT] Subscribing to {Nadal}/relay/+/set (PID 2)
[060345] [MQTT] Subscribing to {Nadal}/pulse/+/set (PID 3)
[060350] [MQTT] Subscribing to {Nadal}/led/+/set (PID 4)
[060354] [MQTT] Subscribing to {Nadal}/action/set (PID 5)
[060363] [MQTT] Sending {Nadal}/app => ESPURNA (PID 1)
[060365] [MQTT] Sending {Nadal}/version => 1.14.0 (PID 1)
[060370] [MQTT] Subscribe ACK for PID 2
[060375] [MQTT] Subscribe ACK for PID 3
[060376] [MQTT] Subscribe ACK for PID 4
[060379] [MQTT] Subscribe ACK for PID 5
[060384] [MQTT] Sending {Nadal}/board => ITEAD_S20 (PID 1)
[060389] [MQTT] Sending {Nadal}/host => Endoll (PID 1)
[060393] [MQTT] Sending {Nadal}/ssid => OSR (PID 1)
[060397] [MQTT] Sending {Nadal}/ip => 192.168.1.203 (PID 1)
[060402] [MQTT] Sending {Nadal}/mac => ----- (PID 1)
[060411] [MQTT] Sending {Nadal}/rssi => -69 (PID 1)
[060413] [MQTT] Sending {Nadal}/uptime => 21060 (PID 1)
[060419] [MQTT] Sending {Nadal}/datetime => 2020-01-04 12:09:11 (PID 1)
[060424] [MQTT] Sending {Nadal}/freeheap => 19704 (PID 1)
[060429] [MQTT] Sending {Nadal}/relay/0 => 0 (PID 1)
[060435] [MQTT] Sending {Nadal}/vcc => 3236 (PID 1)
[060438] [MQTT] Sending {Nadal}/status => 1 (PID 1)
[060442] [MQTT] Sending {Nadal}/loadavg => 2 (PID 1)
[068867] [MQTT] Received {Nadal}/relay/0/set => 2
[068868] [RELAY] #0 scheduled ON in 0 ms
[068871] [RELAY] #0 set to ON
[068872] [MQTT] Sending {Nadal}/relay/0 => 1 (PID 1)
[069479] [MQTT] Received {Nadal}/relay/0/set => 2
[069480] [RELAY] #0 scheduled OFF in 0 ms
[069483] [RELAY] #0 set to OFF
[069485] [MQTT] Sending {Nadal}/relay/0 => 0 (PID 1)
[070486] [RELAY] Setting relay mask: 0b0

But when I power down Windows 10 the Espurna TCP connection is not closed. On mqtt.info the response is, eventhougt the borker is down:

[514309] [WEBSOCKET] Requested action: dbgcmd
[514310] [MQTT] AsyncMqttClient, SSL DISABLED, Autoconnect ENABLED
[514310] [MQTT] Client ENABLED, CONNECTED
[514312] [MQTT] Retry WAITING (Now 21514311, Last 21060344, Delay 5000, Step 5000)
[... 80 lines cut ...]
[787452] [MQTT] Sending {mesa}/host => Mesa (PID 1)
[787455] [MQTT] Sending {mesa}/ssid => OSR (PID 1)
[787460] [MQTT] Sending {mesa}/ip => 192... (PID 1)
[787465] [MQTT] Sending {mesa}/mac =>...(PID 1)
[787473] [MQTT] Sending {mesa}/rssi => -67 (PID 1)
[787475] [MQTT] Sending {mesa}/uptime => 128787 (PID 1)
[787481] [MQTT] Sending {mesa}/datetime => 2020-01-05 00:54:32 (PID 1)
[787486] [MQTT] Sending {mesa}/freeheap => 19616 (PID 1)
[787491] [MQTT] Sending {mesa}/relay/0 => 0 (PID 1)
[787497] [MQTT] Sending {mesa}/vcc => 3196 (PID 1)
[787500] [MQTT] Sending {mesa}/status => 1 (PID 1)
[787504] [MQTT] Sending {mesa}/loadavg => 2 (PID 1)
[876485] [MQTT] Received {mesa}/relay/0/set => 2
[876486] [RELAY] #0 scheduled ON in 0 ms
[876489] [RELAY] #0 set to ON
[876490] [MQTT] Sending {mesa}/relay/0 => 1 (PID 1)
[877101] [MQTT] Received {mesa}/relay/0/set => 2
[877102] [RELAY] #0 scheduled OFF in 0 ms
[877105] [RELAY] #0 set to OFF
[877106] [MQTT] Sending {mesa}/relay/0 => 0 (PID 1)
[878108] [RELAY] Setting relay mask: 0b0

Executing mqtt.reset in debug window solve this and force reconection to broker. 

Have you found a solution for this?

mcspr, 2020-01-05:
@oscarsan1 
https://github.com/marvinroger/async-mqtt-client/pull/156 should be a solution, still pending though

bertmelis, 2021-07-22:
Should be fixed, at least in `develop`. Feel free to reopen if the problem persists.

closed 2021-07-22 by bertmelis