marvinroger/async-mqtt-client #107: setWill() cause mqtt could not connect with SSL enable

dzungpv, 2018-07-20:
I am switch form PubSubClient becasue it block cpu. But i have a problem.
I enable SSL by change ASYNC_TCP_SSL_ENABLED to 1.
Then i set this line:
```
mqttClient.setWill(MQTT_LIGHT_AVAILABILITY_TOPIC, 0, true, MQTT_LIGHT_AVAILABILITY_LWTMESSAGE);
```
after 
```
mqttClient.setServer(MQTT_HOST, MQTT_PORT);
```
But it will not connect to my server, remove the line it connected.

I have try to setWill() in onMqttConnect() but it not work.

rneurink, 2018-08-25:
I am also having this problem. I can connect over SSL perfectly but when using the SetWill command it will not connect with an disconnect with reason 0

bertmelis, 2018-09-02:
I'm working without SSL but can't connect after `setWill`. After commenting out, it works just fine!

bertmelis, 2018-09-02:
I figured out my problem. AsyncMqttClient doesn't copy the last will *topic* or *value* but only saves a pointer. And after `setWill` is called, *topic* and *value* might get out of scope.

I've marked made topic and payload `static` to keep them available for when the mqtt connection is established (which is async and not in the same scope as `setWill`).

rneurink, 2018-09-04:
@bertmelis 
Could you post your code? I still seem to have the problem
This is my code https://pastebin.com/T4e9Ezc8

bertmelis, 2018-09-04:
You're using mosquitto? What does the log of your broker say?

rneurink, 2018-09-04:
1536078725: Socket error on client <unknown>, disconnecting.
1536078725: New connection from 192.168.0.107 on port 8883.
It is kind of weird how it tries to connect twice just after each other

bertmelis, 2018-09-04:
Without SSL it is working? It sure seems like some buffer got overwritten.

rneurink, 2018-09-04:
It is working without SSL and it is working without the `mqttClient.setWill`

rneurink, 2018-09-04:
This is definitely a bug in my code, mosquitto SSL or the lib. After reflashing this multiple times I get around 50% succeeded connections with working will

sooyewguan, 2018-09-12:
Same problem, setwill doesn't work with SSL

timpur, 2018-09-13:
Sadly ATM I lack time to investigate, but I will try to... When I can

sooyewguan, 2018-12-23:
Try change the lwIP to v2 Higher Bandwidth

jeroenst, 2018-12-23:
My error was caused because the setwill function doesn't copy the *topic data but just stores the pointer. Because I called this from a separate mqttinit function, the topic was lost because the variable containing the topic was not static. So it was a different error which has nothing to do with SSL. Altough, when using setwill in combination with ssl the connection fails with also the same error in the mosquitto logging.

jeroenst, 2018-12-23:
With ssl enabled I get this error:

`(connectToMqtt) Connecting mqtt

Exception (29):
epc1=0x4021d770 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000204 depc=0x00000000

ctx: sys 
sp: 3fffe5a0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffe740:  0d31f0e0 0b204f19 eef326de 3fff186c  
3fffe750:  3fffba74 00000408 3fffba74 40100698  
3fffe760:  00000204 4022c0ca 00009c10 00001382  
3fffe770:  00000082 3fffb81c 3fffbda4 4022c2e0  
3fffe780:  fed37fa0 00000000 00000001 00000001  
3fffe790:  3fffb81c 00000082 3fffba5c 4022c0ca  
3fffe7a0:  3fffb81c 3fffba5c 00000081 4022cdca  
3fffe7b0:  3fff186c 000013c3 000013c3 4010020c  
3fffe7c0:  3fffbda4 3fffbc7c 00000040 00000000  
3fffe7d0:  0000000c 3fffb02c 00000041 00000041  
3fffe7e0:  00000000 00000001 00000001 4022c358  
3fffe7f0:  3fffb93c 3fffb02c 00000000 0000007f  
3fffe800:  3fffba5c 00000081 3fffba5c 0000007f  
3fffe810:  3fffba5c 3fffb02c 3fffb02c 4022d08d  
3fffe820:  00000002 00000001 3fffecac 4022c170  
3fffe830:  00000100 00000100 3fffb714 3fffb08c  
3fffe840:  3fffb2c4 00000100 3fffb02c 4022e249  
3fffe850:  3fffb714 3fff7e7c 00000003 40231ebd  
3fffe860:  00000002 00000079 00000003 00000000  
3fffe870:  00000100 3fff8e5b 3fffecac 40232359  
3fffe880:  3fff7e7c 3fffb714 3ffeb482 402321e5  
3fffe890:  3fffb284 0000001a 00000003 40232086  
3fffe8a0:  4022a214 3fffb284 3fff8e5b 00000000  
3fffe8b0:  00000004 3fffb284 3fff8e5b 4022a2fe  
3fffe8c0:  000005c0 00000000 a9daf112 91b0ba62  
3fffe8d0:  3ad1d3b7 96826b9b bc5709a4 ba59846e  
3fffe8e0:  82ae2911 14513370 a3c7201e 6dc7c908  
3fffe8f0:  dbbca0f4 0e2c7be9 ed7781d0 ddab3bad  
3fffe900:  8d5bad12 0000801c 00000000 00000000  
3fffe910:  00000000 00000000 00000000 00000000  
3fffe920:  00000000 e82c0000 00000000 ffffff80  
3fffe930:  3fffea00 3fffe9f0 0000000c 00000006  
3fffe940:  3ffea152 00000007 00000000 4024eeff  
3fffe950:  00000000 0000000b 402524b8 00000000  
3fffe960:  3fff1430 3fff1330 3fff1474 402185a0  
3fffe970:  3fff1481 00000002 7ffffff2 ffff0208  
3fffe980:  3fff1474 7fffffff 00000001 3ffefd5c  
3fffe990:  3fff186c 000007f5 000007f5 3fff186c  
3fffe9a0:  00000010 00000010 00000000 3fff186c  
3fffe9b0:  00000010 00000010 00000000 40100579  
3fffe9c0:  3fff186c 40213b83 000007f4 3fff186c  
3fffe9d0:  00000010 00000010 00000000 40100579  
3fffe9e0:  00000010 40213b83 3fff1268 00000000  
3fffe9f0:  00000010 3fff1118 3fff1118 40213b83  
3fffea00:  3ffe9bd8 00000000 3fff1118 3ffefd5c  
3fffea10:  00000010 3fff10a0 3fff10a0 40213b83  
3fffea20:  3ffe9bd8 00000000 3fff10a0 40213bcf  
3fffea30:  3ffe9bd8 3ffe9bd8 3fff10a0 40213c01  
3fffea40:  3fff0a6c 3fff0a98 3fff10a0 40213c4e  
3fffea50:  3ffe9bd8 402068c8 3fff15b8 4020f400  
3fffea60:  00000000 00000000 00000001 40214b45  
3fffea70:  4000b570 3fff0a98 3ffeb55c 40214b7e  
3fffea80:  00000000 00000000 00000000 00000000  
3fffea90:  00000000 00000000 00000000 00000000  
3fffeaa0:  00000000 00000000 00000000 00000000  
3fffeab0:  00000000 00000000 00000000 00000000  
3fffeac0:  00000000 00000000 00000000 00000000  
3fffead0:  00000000 00000000 00000000 00000000  
3fffeae0:  00000000 00000000 00000000 00000000  
[... 311 lines cut ...]
3ffffe60:  00000000 00000000 00000000 fffffffe  
3ffffe70:  ffffffff 3fffc6fc 00000001 3fff10d4  
3ffffe80:  3fff10e8 3fffdad0 3fff196c 00000030  
3ffffe90:  00000000 00000000 00000001 3fff1965  
3ffffea0:  3fff1278 3fffdad0 3fff196c 00000030  
3ffffeb0:  40000ea3 00000000 00000001 00000001  
3ffffec0:  3fffdab0 00000000 3fffd9d0 3fff196c  
3ffffed0:  00000000 00000000 00000001 3fff1965  
3ffffee0:  00000000 3fffdad0 3fff196c 00000030  
3ffffef0:  00000000 00000000 3fff1278 40215360  
3fffff00:  00000000 4bc6a7f0 00001655 3fff196c  
3fffff10:  00000000 00000000 4bc6a7f0 00000000  
3fffff20:  3fff0ac0 00000000 40106576 a3126e97  
3fffff30:  00000000 00000000 00000000 40215eb0  
3fffff40:  3fffdad0 3fff10e8 3fff10d4 4020489e  
3fffff50:  4024c20e 3fff1278 3ffe8624 3fff1278  
3fffff60:  3fffdad0 3fff196c 40214b24 3fffefb0  
3fffff70:  00000000 00000000 3fff1965 3fff196c  
3fffff80:  3fffdad0 3fff1278 3fff1965 40205c14  
3fffff90:  00000000 00000000 00000000 00000000  
3fffffa0:  00000000 00000000 00000001 40214b45  
<<<stack<<<

last failed alloc call: 4022C0CA(1032)
`

Bastelralf, 2019-09-21:
Hi 
is there a workaround to get setwill() with SSL working?

Plggeldy, 2019-09-21:
I would love to know, what I can do for MQTTClient::setwill() with TLS/SSL.. 

Is there a chance to get it working?

Plggeldy, 2019-09-24:
The solution is here:

https://github.com/256dpi/arduino-mqtt

> Set a will message (last testament) that gets registered on the broker after connecting. setWill() has to be called before calling connect():<

jeroenst, 2019-11-01:
> The solution is here:
> 
> https://github.com/256dpi/arduino-mqtt
> 
> > Set a will message (last testament) that gets registered on the broker after connecting. setWill() has to be called before calling connect():<

Is this really non-blocking asynchronous?

jeroenst, 2019-11-01:
I did some investigation, it seems that the password is not send after the username (I receive everything up to the password with ncat). Very strange..

KrYpToDeN, 2020-06-04:
> The problem is that the _onconnect function calls _client.add for each piece of data to send to the mqtt server. When setwill is used, there is to much data to be send by the ASyncTCP module because it also add ssl data to each packet, causing it to drop the password.
> 
> After changing the code to send less packets setwill works with SSL!

Could you help me, how send less packets?
Our problem is the same - https://github.com/eclipse/mosquitto/issues/1715

qcasey, 2020-06-04:
I believe this is ultimately caused by #193. What I've done to get SSL working is implementing a queue and only allowing the packet in the head to be on the wire at any time.

This requires QoS2 and popping packets once they're acknowledged

KrYpToDeN, 2020-06-05:
> I believe this is ultimately caused by #193. What I've done to get SSL working is implementing a queue and only allowing the packet in the head to be on the wire at any time.
> 
> This requires QoS2 and popping packets once they're acknowledged

Hello.

Thank you for your answer, but it's for asynchronous code.

We use asynchronous version, because packets not so much to send/get.

But the problem is the same. Server got Will topic in mosquitto log file. But it publishes it 50/50 times. Also we can't find a way, when it's happed. It's always different. Will topic can go/it may be lost.

Do you know how solve it?

bertmelis, 2021-02-10:
The issue should be gone using the `development` branch.

closed 2021-02-10 by bertmelis