marvinroger/async-mqtt-client #111: ESP32 - Is there SSl TLS or not ?

Netoperz, 2018-08-31:
I'm a bit lost,

An many people are saing different things.

So my Question is simple:

If I want to use ESP32 to run some MQTTclient (publish and receive) is it possible to connect to the broker, using  SSL/TLS ?

My broker is MOSCA, it supports TLS for sure.
 Right now im using Pub Sub client lib, but that lib does not support encription (as most of arduinos do not have memory to handel so no one bothers to add SSL/TLS) 

Question is if this lib actualy works with encrypted connection for ESP 32, and where is the example of use ?

If this lib does not work, do I have any other library around that someone can know about ?

Thank You !!!!! :)

timpur, 2018-09-09:
Our dependency is AsyncTCP lib, please see https://github.com/me-no-dev/AsyncTCP/issues/8

In short sadly not atm ....

fedor1210, 2018-09-26:
PubSubClient actually now supports TLS on ESP32 with WiFiClientSecure (included in Arduino SDK, also available on ESP8266, but without hardware acceleration), the example is available here: http://www.iotsharing.com/2017/08/how-to-use-esp32-mqtts-with-mqtts-mosquitto-broker-tls-ssl.html

DAVIDBETAN2000, 2020-09-25:
asyncmqtt library is SSL or not?

bertmelis, 2020-09-30:
no

closed 2020-09-30 by bertmelis