marvinroger/async-mqtt-client #119: Problem with using fingerprint for SSL

gjt211, 2018-12-01:
Hi all,

I have been using this fantastic lib for quite a while and must say thanks to all the people creating, working on, and updating it.

I am having a problem that I feel should be easy to solve but with my level of knowledge I am struggling with it.

I want to store the fingerprint in a variable that I can change if needed.
The existing hard coded `#define MQTT_SERVER_FINGERPRINT {fingerprint stuff here}` works well for me with `mqttClient.addServerFingerprint((const uint8_t[])MQTT_SERVER_FINGERPRINT);` , but now am trying to store this a different way.

I don't really care how I store this, as long as it can be changed programatically, rathe than being hard coded.
I have tried the following;
`uint8_t fingerprint[32] = {fingerprint here};`
`char fingerprint[32] = "fingerprint here";`

I have tried various casts in the `addServerFingerprint` call also without success.

Hopefully someone can advise how to do this if possible.

Many thanks in advance.

bertmelis, 2018-12-01:
What is the error you get?
Is it failing at compilation or Does it not work?

bertmelis, 2018-12-01:
I see your fingerprint is 32 bytes long, while SHA1 is defined as 20 bytes. Try again after adjusting the size in the header file.

gjt211, 2018-12-01:
Hi @bertmelis,
Thanks for your responses.
I was getting various compile errors about casts and also if it would compile, it would not work.
I have actually got this working now, so I can download a new fingerprint from my server if needed.
My fingerprint is SHA1, and yes it is 20 bytes long. I was worried that the length may change in the future so allowed for extra bytes. A quick search and I realise that I only need 20 bytes.

Using the following in my code now works well. I will endeavour to change the 41 to 20 today to tidy up the code. FYI, the bytes above 20 are filled with zeros in my code.
`uint8_t finger[41];`
`mqttClient.addServerFingerprint((const uint8_t *)finger);`

This fingerprint is used on a specific port of my server (not HTTP/S) so my sensor can connect elsewhere and download the current fingerprint. It then can compare the downloaded one versus the existing stored fingerprint. That way if my fingerprint for that specific port changes, the sensor can obtain a new fingerprint.

gjt211, 2018-12-07:
All working with 20 byte char array, thanks for your help.

closed 2018-12-07 by gjt211