marvinroger/async-mqtt-client #120: onMqttDisconnect doesn't get called until keep alive times out

gjt211, 2018-12-07:
I have found that the `onMqttDisconnect` callback does not get called until the MQTT keep alive time expires.
By default this is 15 seconds, so if I power off my wifi router or pull the ethernet cable connecting the router to the internet, I don't get a onMqttDisconnect callback until the keep alive time passes.
The WiFi callback runs indicating it has been lost but MQTT keeps going.
Is it possible that if WiFi disconnected callback runs, then it can also set MQTT to be disconnected as well?

bertmelis, 2018-12-07:
By design, WiFi can disconnect and reconnect without the broker of client notices anything. MQTT is designed to be used in unstable wireless situations.

Solving this issue would break this design.

bertmelis, 2018-12-07:
Should you still want the behaviour, you can implement it yourself in the onWiFiDisconnect handler.

gjt211, 2018-12-08:
Hi @bertmelis thanks for the information and advice.
I understand and accept what you are saying thank you.
For my situation, WiFi rarely disconnects or drops out but if/when it does it is always off for much longer than the keep alive timeout. I will implement something as you suggest.

closed 2018-12-08 by gjt211

thijstriemstra, 2020-05-02:
> I have found that the onMqttDisconnect callback does not get called until the MQTT keep alive time expires.

Good to know, thanks for info.