marvinroger/async-mqtt-client #124: WIFI continuously disconnecting when using sta+ap wifi mode

gnkarn, 2018-12-15:
as i was not able found a single working case where the esp is actin as sta  on mqtt , and as of AP using UDP, i have the doubt if it is something wrong in the code or if this si not possible to implement as of today.

in my case i have , a master esp and a slave esp
master is STA+AP , and slave is STA only 
slave has no problems , but master is totally unreliable due to wifi continuous dropping and reconnecting .

` // https://github.com/hansjny/Natural-Nerd/blob/master/arduino/soundsread2/sound_reactive.ino
// version async mqtt
#include <FastLED.h>
#include <ESP8266WiFi.h>
//#include <WiFiUDP.h>
#include "reactive_common.h"

#include <DNSServer.h>            //Local DNS Server used for redirecting all requests to the configuration portal
//#include <ESP8266WebServer.h>     //Local WebServer used to serve the configuration portal
#include <ArduinoJson.h>
#define SENSORNAME "led_vertical"

/************ WIFI and MQTT Information (CHANGE THESE FOR YOUR SETUP) ******************/
//const char* ssid = "YourSSID"; //type your WIFI information inside the quotes
//const char* password = "YourWIFIpassword";
//#include <PubSubClient.h> // mqtt

const char* mqtt_username = "xxxxx";
const char* mqtt_password = "xxxx";
const char* mqtt_clientid = SENSORNAME;
#include <AsyncMqttClient.h>    //https://github.com/marvinroger/async-mqtt-client

/// for async mqtt
#define MQTT_HOST IPAddress(192, 168, 0, xx)
#define MQTT_PORT 1883
#include <Ticker.h> /// para leds de status
AsyncMqttClient mqttClient;
Ticker mqttReconnectTimer;

WiFiEventHandler wifiConnectHandler;
WiFiEventHandler wifiDisconnectHandler;
Ticker wifiReconnectTimer;
///

long lastReconnectAttempt = 0; // mqtt sta reconnect
bool stateOn = false;
int transitionTime = 0;
#define BRIGHTNESS  200// set master brightness 0-255 here to adjust power consumption
byte red = 255;
byte green = 255;
byte blue = 255;
byte brightness = 255;
byte myBrightness = BRIGHTNESS;  // global brightness changable from screen
/************* MQTT TOPICS (change these topics as you wish)  **************************/
const char* led_vertical_state_topic = "led_vertical/stat"; // no los uso?
const char* led_vertical_set_topic = "led_vertical/cmnd";

const char* on_cmd = "ON";
const char* off_cmd = "OFF";
const char* effect = "solid";
String effectString = "solid";
String oldeffectString = "solid";

// MSGEQ7 wiring on spectrum analyser shield ********************************
#define MSGEQ7_STROBE_PIN D2//antes 2
#define MSGEQ7_RESET_PIN  D6// antes 1
#define AUDIO_LEFT_PIN    0
#define AUDIO_RIGHT_PIN   0
// storage of the 7 10Bit (0-1023) audio band values
// modified only by AudioRead()
int left[7];
int right[7];
/****************************************FOR JSON***************************************/
// Step 1: DEfinir Json buffer :Reserve memory space
[... 584 lines cut ...]
        if (root.containsKey("color")) {
                red = root["color"]["r"];
                green = root["color"]["g"];
                blue = root["color"]["b"];
        }
        else {
                // Red = red;
                // Green = green;
                // Blue = blue;
        }

        if (root.containsKey("effect")) {
                effect = root["effect"];
                opMode = root["effect"];
                effectString = effect;
                //  gCurrentPatternNumber = getposition();
                //twinklecounter = 0; //manage twinklecounter
        }

        //flash = true;
        //startFlash = true;

        return true;
}
`

luebbe, 2021-02-10:
Possible duplicate of #106

bertmelis, 2021-07-26:
Please provide a **minimal** example of the problem you face.

Closing for now, feel free to reopen.

closed 2021-07-26 by bertmelis