marvinroger/async-mqtt-client #125: Enabling ssl causes exception 29

jeroenst, 2018-12-23:
I don't know why this is happening, but when I enable SSL my ESP8266 crashes and does a stack dump:

(connectToMqtt) Connecting mqtt

Exception (29):
epc1=0x4021d770 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000204 depc=0x00000000

ctx: sys 
sp: 3fffe5a0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffe740:  0d31f0e0 0b204f19 eef326de 3fff1984  
3fffe750:  3fffbb54 00000408 3fffbb54 40100698  
3fffe760:  00000204 4022c0ca 00009ce0 0000139c  
3fffe770:  00000082 3fffb8fc 3fffbe84 4022c2e0  
3fffe780:  fed37fa0 00000000 00000001 00000001  
3fffe790:  3fffb8fc 00000082 3fffbb3c 4022c0ca  
3fffe7a0:  3fffb8fc 3fffbb3c 00000081 4022cdca  
3fffe7b0:  3fff1984 000013dd 000013dd 4010020c  
3fffe7c0:  3fffbe84 3fffbd5c 00000040 00000000  
3fffe7d0:  0000000c 3fffb0fc 00000041 00000041  
3fffe7e0:  00000000 00000001 00000001 4022c358  
3fffe7f0:  3fffba1c 3fffb0fc 00000000 0000007f  
3fffe800:  3fffbb3c 00000081 3fffbb3c 0000007f  
3fffe810:  3fffbb3c 3fffb0fc 3fffb0fc 4022d08d  
3fffe820:  00000002 00000001 3fffecac 4022c170  
3fffe830:  00000100 00000100 3fffb7c4 3fffb8cc  
3fffe840:  3fffb0ac 00000100 3fffb0fc 4022e249  
3fffe850:  3fffb7c4 3fffb3ac 00000003 40231ebd  
3fffe860:  00000002 00000079 00000003 00000000  
3fffe870:  00000100 3fff8ec3 3fffecac 40232359  
3fffe880:  3fffb3ac 3fffb7c4 3ffeb47a 402321e5  
3fffe890:  3fffb06c 0000001a 00000003 40232086  
3fffe8a0:  4022a214 3fffb06c 3fff8ec3 00000000  
3fffe8b0:  00000004 3fffb06c 3fff8ec3 4022a2fe  
3fffe8c0:  000005c0 00000000 a9daf112 91b0ba62  
3fffe8d0:  3ad1d3b7 96826b9b bc5709a4 ba59846e  
3fffe8e0:  82ae2911 14513370 a3c7201e 6dc7c908  
3fffe8f0:  dbbca0f4 0e2c7be9 ed7781d0 ddab3bad  
3fffe900:  8d5bad12 0000801c 00000000 00000000  
3fffe910:  00000000 00000000 00000000 00000000  
3fffe920:  00000000 e82c0000 00000000 ffffff80  
3fffe930:  3fffea00 3fffe9f0 0000000c 00000006  
3fffe940:  3ffea14a 00000007 00000000 4024eeff  
3fffe950:  00000000 0000000b 402524b8 00000000  
3fffe960:  3fff1440 3fff1340 3fff1484 402185a0  
3fffe970:  3fff1491 00000002 7ffffff2 ffff0208  
3fffe980:  3fff1484 7fffffff 00000001 3ffefd5c  
3fffe990:  3fff1984 000007f5 000007f5 3fff1984  
3fffe9a0:  00000010 00000010 00000000 3fff1984  
3fffe9b0:  00000010 00000010 00000000 40100579  
3fffe9c0:  3fff1984 40213a43 000007f4 3fff1984  
3fffe9d0:  00000010 00000010 00000000 40100579  
3fffe9e0:  00000010 40213a43 3fff1278 00000000  
3fffe9f0:  00000010 3fff1128 3fff1128 40213a43  
3fffea00:  3ffe9bd0 00000000 3fff1128 3ffefd5c  
3fffea10:  00000010 3fff10a0 3fff10a0 40213a43  
3fffea20:  3ffe9bd0 00000000 3fff10a0 40213a8f  
3fffea30:  3ffe9bd0 3ffe9bd0 3fff10a0 40213ac1  
3fffea40:  3fff0a6c 3fff0a98 3fff10a0 40213b0e  
3fffea50:  3ffe9bd0 40206924 3fff15c8 4020f400  
3fffea60:  00000000 00000000 00000001 40214a05  
3fffea70:  4000b570 3fff0a98 3ffeb554 40214a3e  
3fffea80:  00000000 00000000 00000000 00000000  
3fffea90:  00000000 00000000 00000000 00000000  
3fffeaa0:  00000000 00000000 00000000 00000000  
3fffeab0:  00000000 00000000 00000000 00000000  
3fffeac0:  00000000 00000000 00000000 00000000  
3fffead0:  00000000 00000000 00000000 00000000  
3fffeae0:  00000000 00000000 00000000 00000000  
[... 310 lines cut ...]
3ffffe50:  40204906 0000165f 00418937 00000000  
3ffffe60:  00004bc6 00000000 00000000 fffffffe  
3ffffe70:  00000000 3fffc6fc 00000000 3fff10d4  
3ffffe80:  3fff10f0 0000165f 3fff181c 00000030  
3ffffe90:  00000000 00000000 00000001 3fff1815  
3ffffea0:  3fff1288 3fffdad0 3fff181c 00000030  
3ffffeb0:  3fff1288 3fffdad0 3fff181c 00000030  
3ffffec0:  3fff1288 3fffdad0 3fff181c 00000030  
3ffffed0:  00000000 00000000 00000001 3fff1815  
3ffffee0:  00000000 3fffdad0 3fff181c 00000030  
3ffffef0:  ffffffff 3fffc6fc 3fff1288 40215360  
3fffff00:  00000000 4bc6a7f0 0000165f 3fff181c  
3fffff10:  00000000 00000000 4bc6a7f0 00000000  
3fffff20:  3fff0ac0 00000000 40106576 bef9db22  
3fffff30:  00000000 00000000 00000000 40215eb0  
3fffff40:  3fffdad0 3fff10f0 3fff10d4 40204906  
3fffff50:  4024c20e 3fff1288 3ffe8624 3fff1288  
3fffff60:  3fffdad0 3fff181c 402149e4 3fffefb0  
3fffff70:  00000000 00000000 3fff1815 3fff181c  
3fffff80:  3fffdad0 3fff1288 3fff1815 40205c70  
3fffff90:  00000000 00000000 00000000 00000000  
3fffffa0:  00000000 00000000 00000001 40214a05  
<<<stack<<<

last failed alloc call: 4022C0CA(1032)

jeroenst, 2018-12-23:
When I disable SSL it works fine:

(connectToMqtt) Connecting mqtt
(onMqttConnect) Connected to MQTT SessionPresent=0.

bertmelis, 2018-12-24:
you have to decoder the stackdump to make something useful out of it.

Muhamed-Reyad, 2019-01-08:
@jeroenst 
I have noticed that the library stopped working for my devices after the new SSL was implemented on the server side ( IBM bluemix deprecated TLS 1.0 and 1.1 ) i think it was the update to TLS 1.2 that caused this since it is probably not supported by the library.

karannadar, 2019-01-14:
Check ESP 8266 Free Heap Memory. It might be low and causing ESP8266 to crash.

jeroenst, 2019-01-23:
I will investigate this more in the following weeks.

jeroenst, 2019-01-28:
I think this has to do with callbacks during delays in code. I removed the delays and replaced them with millis() constructions and the crash doesn't occur anymore.

closed 2019-01-28 by jeroenst

rohitjust24idpl, 2019-03-09:
Still facing issue with SSL **Exception (29):**

```
#include <ESP8266WiFi.h>
#include <Ticker.h>
#include <AsyncMqttClient.h>

#define WIFI_SSID                 "SSID"
#define WIFI_PASSWORD             "PSK"

#define mqtt_username             "username"
#define mqtt_password             "password"
#define MQTT_HOST                 "host"
#define MQTT_PORT                 8883
#define MQTT_SECURE               true

AsyncMqttClient mqttClient;
Ticker mqttReconnectTimer;

WiFiEventHandler wifiConnectHandler;
WiFiEventHandler wifiDisconnectHandler;
Ticker wifiReconnectTimer;

void connectToWifi() {
  Serial.println("Connecting to Wi-Fi...\n");
  Serial.println(ESP.getFreeHeap());
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
}

void onWifiConnect(const WiFiEventStationModeGotIP& event) {
  Serial.println("Connected to Wi-Fi.");
  Serial.println(ESP.getFreeHeap());
  connectToMqtt();
}

void onWifiDisconnect(const WiFiEventStationModeDisconnected& event) {
  Serial.println("Disconnected from Wi-Fi.");
  Serial.println(ESP.getFreeHeap());
  mqttReconnectTimer.detach(); // ensure we don't reconnect to MQTT while reconnecting to Wi-Fi
  wifiReconnectTimer.once(2, connectToWifi);
}

void connectToMqtt() {
  Serial.println("Connecting to MQTT...");
  Serial.println("");
  Serial.println(ESP.getFreeHeap());
  mqttClient.connect();
}

void onMqttConnect(bool sessionPresent) {
  Serial.println("Connected to MQTT.");
  Serial.println("");
  Serial.println(ESP.getFreeHeap());

}

void onMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
  Serial.println("Disconnected from MQTT.");
  Serial.println("");
  Serial.println(ESP.getFreeHeap());
  if (WiFi.isConnected()) {
    mqttReconnectTimer.once(2, connectToMqtt);
  }
}

void onMqttSubscribe(uint16_t packetId, uint8_t qos) {
  Serial.println("Subscribe acknowledged.");
}

void onMqttUnsubscribe(uint16_t packetId) {
[... 361 lines cut ...]
3ffffef0:  00000000 3fffdad0 3ffeed34 00000030  
3fffff00:  00000000 3fffdad0 3ffeed34 00000030  
3fffff10:  3fffdad0 00000005 3ffeec28 402067f4  
3fffff20:  00000009 00000000 3ffeec28 40206a81  
3fffff30:  00000002 00000001 3ffeeba0 3ffeed34  
3fffff40:  00000006 00000000 3ffeebec 00000001  
3fffff50:  4020676c 00000000 3ffeec28 3ffeed34  
3fffff60:  3fffdad0 00000000 3ffeec28 402029a4  
3fffff70:  3fffdad0 00000000 3ffeea40 40202d5b  
3fffff80:  40202908 feefeffe 40207310 402072fc  
3fffff90:  00000000 00000000 00000001 40206f91  
3fffffa0:  3fffdad0 00000000 3ffeed04 40207022  
<<<stack<<<

last failed alloc call: 40215FAE(1024)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld
```