marvinroger/async-mqtt-client #126: TLS 1.2 Support

Muhamed-Reyad, 2019-01-08:
Since TLS 1.0 and 1.1 are now deprecated, is there a chance that we can get an update to TLS 1.2 in this library? 
I'm working with IBM bluemix and i had to disable SSL altogether in order to be able to connect to the IBM Broker again since they no longer support TLS 1.0 and 1.1

bertmelis, 2021-02-25:
Although I'd want to, this has to be solved in the ESPAsnycTCP and AsyncTCP libraries.

closed 2021-02-25 by bertmelis

kleini, 2021-02-25:
And there are some really good things in pull requests for ESPAsyncTCP but the maintainer does not respond at all to anything.

bertmelis, 2021-02-25:
It's always possible to define dependencies from alternative sources. But it can quickly become a mess... Open to suggestions though.

kleini, 2021-02-25:
Right, but multiple people should then decide to create a fork, where all the good work then goes into. Much better would be, if the maintainer finds additional maintainers, just like Marvin did.

luebbe, 2021-02-25:
Marvin didn't find them. They came upon him ;)

kleini, 2021-02-25:
Is the maintainer reachable somewhere? I just noticed, he is not responding at all.

bertmelis, 2021-02-25:
You mean @me-no-dev? I guess he's terribly occupied by all the new Espressif stuff that's coming our way.

Pablo2048, 2021-02-25:
> Is the maintainer reachable somewhere? I just noticed, he is not responding at all.

He actually accepted one pull request here https://github.com/me-no-dev/ESPAsyncWebServer just today. So maybe...