marvinroger/async-mqtt-client #127: fully  featured esp32 compilation error

sheetalsingala, 2019-01-08:
I'm using the fully-featured-ESP32 example on my ESP 32 wroom.
I get this error on compilation : 
**ESPAsyncTCP.h: No such file or directory**

eos1d3, 2019-03-25:
@justoke I just tested this with PlatformIO. And I do not have any compilation error. What I did is to copy the FullyFeatured-ESP32.ino file only. And set the library to AsyncMqttClient@0.8.2.

The dependency I see from compilation:

```
Dependency Graph
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.0.3
|-- <WiFi> 1.0
```

justoke, 2019-06-01:
@eos1d3 Thank you for the tip - I just moved over to Platform IO recently, so this actually makes sense to me now

closed 2020-08-04 by sheetalsingala

justoke, 2020-08-04:
> @eos1d3 Thank you for the tip - I just moved over to Platform IO recently, so this actually makes sense to me now

Glad it helped. Platform IO is really good and I use it almost exclusively for development. It is easier maybe to manage libraries as you can just add links to git repos where you need a specific commit or branch in such situations.