marvinroger/async-mqtt-client #128: MQTT SSL TLS - self-signed CA and certificate

muuuh, 2019-01-12:
Hi,

how would I use the library together with an MQTT server with a self-signed CA and certificate as described here: http://www.steves-internet-guide.com/mosquitto-tls/

Usually, you would at the connection specify the ca certificate or use tls_insecure_set in the MQTT library.

Thank you!

jeroenst, 2019-01-23:
If you don't specify the server fingerprint (don't call mqttClient.addServerFingerprint()) and set setSecure to true (mqttClient.setSecure(true)) ssl is activated without certificate check.

Although in my software which is using async-mqtt-client I get an stack dump when I enable ssl.

jeroenst, 2019-01-23:
I'm wondering if this works for you, reading the code it should work, but at my place I see packets with tcpdump between mosquitto and the esp but connection is not established.

17:22:06.106004 IP 192.168.2.178.49153 > 192.168.2.1.8883: Flags [S], seq 6509, win 2144, options [mss 536], length 0
17:22:06.106075 IP 192.168.2.1.8883 > 192.168.2.178.49153: Flags [S.], seq 2269391815, ack 6510, win 29200, options [mss 1460], length 0
17:22:06.114812 IP 192.168.2.178.49153 > 192.168.2.1.8883: Flags [P.], seq 1:73, ack 1, win 2144, length 72
17:22:06.114880 IP 192.168.2.1.8883 > 192.168.2.178.49153: Flags [.], ack 73, win 29200, length 0
17:22:06.115428 IP 192.168.2.1.8883 > 192.168.2.178.49153: Flags [.], seq 1:537, ack 73, win 29200, length 536
17:22:06.115455 IP 192.168.2.1.8883 > 192.168.2.178.49153: Flags [.], seq 537:1073, ack 73, win 29200, length 536
17:22:06.115619 IP 192.168.2.1.8883 > 192.168.2.178.49153: Flags [P.], seq 1073:1609, ack 73, win 29200, length 536
17:22:06.115658 IP 192.168.2.1.8883 > 192.168.2.178.49153: Flags [.], seq 1609:2145, ack 73, win 29200, length 536
17:22:06.121017 IP 192.168.2.178.49153 > 192.168.2.1.8883: Flags [.], ack 537, win 2144, length 0
17:22:06.121079 IP 192.168.2.1.8883 > 192.168.2.178.49153: Flags [.], seq 2145:2681, ack 73, win 29200, length 536
17:22:06.123599 IP 192.168.2.178.49153 > 192.168.2.1.8883: Flags [.], ack 1073, win 2144, length 0
17:22:06.123658 IP 192.168.2.1.8883 > 192.168.2.178.49153: Flags [P.], seq 2681:2821, ack 73, win 29200, length 140
17:22:06.126394 IP 192.168.2.178.49153 > 192.168.2.1.8883: Flags [.], ack 1609, win 2144, length 0
17:22:06.132858 IP 192.168.2.178.49153 > 192.168.2.1.8883: Flags [.], ack 2145, win 2144, length 0
17:22:06.150628 IP 192.168.2.178.49153 > 192.168.2.1.8883: Flags [.], ack 2681, win 2144, length 0
17:22:06.367124 IP 192.168.2.178.49153 > 192.168.2.1.8883: Flags [P.], seq 73:431, ack 2821, win 2004, length 358
17:22:06.370282 IP 192.168.2.1.8883 > 192.168.2.178.49153: Flags [P.], seq 2821:2912, ack 431, win 30016, length 91
17:22:06.393996 IP 192.168.2.178.49153 > 192.168.2.1.8883: Flags [P.], seq 431:967, ack 2912, win 1913, length 536
17:22:06.437416 IP 192.168.2.1.8883 > 192.168.2.178.49153: Flags [.], ack 967, win 31088, length 0
17:22:06.442001 IP 192.168.2.178.49153 > 192.168.2.1.8883: Flags [P.], seq 967:1429, ack 2912, win 1913, length 462
17:22:06.442053 IP 192.168.2.1.8883 > 192.168.2.178.49153: Flags [.], ack 1429, win 32160, length 0

s00500, 2019-10-11:
Having a similar problem

DAVIDBETAN2000, 2020-09-24:
(.literal._ZN15AsyncMqttClient10_onConnectEP11AsyncClient+0x4)  undefined reference to `ssl_match_fingerprint'
esp32, don´t compile

bertmelis, 2020-09-30:
esp32 async tcp does not support SSL.

you might want to use the builtin C client.

s00500, 2020-10-01:
btw this issue has been fixed I think, at least on the esphome fork...

bertmelis, 2021-02-25:
Meanwhile, I've got an ESP8266 running with TLS and self-signed certificates using the `development` branch.

closed 2021-02-25 by bertmelis