marvinroger/async-mqtt-client #129: An 8266 connect to multiple MQTT servers

Arduino8266, 2019-01-17:
Use an 8266 connect to multiple MQTT servers ， Whether to support ？   Please help me thank you 。

bertmelis, 2019-01-17:
I've got this feeling your question doesn't reveal your real purpose:
Do you want 2 seperate clients? --> just create 2 seperate client instances
Do you want a fallback scenario? --> not part of this lib, can be implemented in user code (connect to other server after a number of failed connect attempts)
Do you want the data to be available on 2 brokers? --> brokers can be bridged.

Arduino8266, 2019-01-18:
> I've got this feeling your question doesn't reveal your real purpose:
> Do you want 2 seperate clients? --> just create 2 seperate client instances
> Do you want a fallback scenario? --> not part of [...]

Thank you for your reply 。
Want to achieve  ，8266 as the client, To connect the two servers ，and  can interact with data from both servers at the same time , this is part of the demo , However, it is error , it can only connect to the second (MQTT_HOST2) server .

-------------------------------------------------------------------------------------------------------------------
  mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  mqttClient.onSubscribe(onMqttSubscribe);
  mqttClient.onUnsubscribe(onMqttUnsubscribe);
  mqttClient.onMessage(onMqttMessage);
  mqttClient.onPublish(onMqttPublish);
  mqttClient.setCredentials("admin","admin");
  mqttClient.setServer(MQTT_HOST, MQTT_PORT);

 Serial.println(F("---------------------------------------------------------------------------------"));

  mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  mqttClient.onSubscribe(onMqttSubscribe);
  mqttClient.onUnsubscribe(onMqttUnsubscribe);
  mqttClient.onMessage(onMqttMessage);
  mqttClient.onPublish(onMqttPublish);
  mqttClient.setCredentials("admin","admin");
  mqttClient.setServer(MQTT_HOST2, MQTT_PORT);
--------------------------------------------------------------------------------------------------END

Arduino8266, 2019-01-18:
@bertmelis       I meams one 8266 connect to 2 different mqtt servers, how can I do this ？ 

In the test:
AsyncMqttClient mqttClient;
AsyncMqttClient mqttClient2;

closed 2019-01-18 by Arduino8266