marvinroger/async-mqtt-client #134: How do I use SSL with esp32 ?

ShivamJoker, 2019-03-03:
Hey I tried setting the build flag to
`build_flags = -DASYNC_TCP_SSL_ENABLED=ASYNC_TCP_SSL_ENABLED`
and replaced the AsyncTCP library with [https://github.com/fremouw/AsyncTCP/tree/mbed-tls](TLS supported version)

but still I am getting the error 
'class AsyncMqttClient' has no member named 'setSecure'

rohitjust24idpl, 2019-03-14:
@marvinroger @ivankravets @skorokithakis 
Put following lines somewhere at the top of the sketch:

```
#ifndef ASYNC_TCP_SSL_ENABLED
#define ASYNC_TCP_SSL_ENABLED 1
#endif
```

solved my issue. But i am getting new error now:

`AsyncMqttClient.hpp:17:23: fatal error: tcp_axtls.h: No such file or directory`

ShivamJoker, 2019-03-14:
What do you mean ? I want fully supported SSL

tve, 2019-04-28:
Works for me: https://github.com/tve/async-mqtt-client
I made more changes to pull PRs in and support PSK cipher suites but maybe this helps you spot what you're missing.

ShivamJoker, 2019-04-29:
@tve I tried everything again and now my board is getting panic attack
Please can you share one example with us

`WiFi connected
IP address:
192.168.31.11
Connecting to MQTT...
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400014fd  PS      : 0x00060630  A0      : 0x80122db1  A1      : 0x3ffb3e50
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb3d10
A10     : 0x3ffc51dc  A11     : 0x00000001  A12     : 0x3f41b4ac  A13     : 0x3ffcaa60
A14     : 0x00000009  A15     : 0xffff0000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x400014fd:0x3ffb3e50 0x40122dae:0x3ffb3e60 0x40122175:0x3ffb3ed0 0x401221f9:0x3ffb3ef0 0

x400f82e6:0x3ffb3f10 0x400f8ed9:0x3ffb3f30 0x400fe843:0x3ffb3f50 0x40103a6e:0x3ffb3f70 0x400f28cb:0x
3ffb3f90 0x400887a1:0x3ffb3fc0`

tve, 2019-04-29:
https://github.com/tve/esp32-secure-base
Note use of PSK cipher suite, though.