marvinroger/async-mqtt-client #143: [QUESTION] Use case for multiple callbacks

bertmelis, 2019-04-30:
All callbacks are stored in std::vectors. So multiple callbacks for the same events are possible.
I have not had the need to attahc multiple callbacks however. And even then, if I only need a few actions I just use an intermediate function.

I know it is a breaking change, but just storing a std::function instead of a vector-of... would mean slightly faster code and less code. Don't know the effect on RAM.

What is the rationale?

jeroenst, 2019-11-30:
For me also multiple callbacks for the same event stored in vectors is overkill. I don't know if anyone uses multiple callbacks the for the same event. And indeed an intermediate can perfectly solve this in a rare occasion when multiple callbacks are needed.

bertmelis, 2021-07-22:
Reviving this question.

A problem I have with the vector of callbacks is that it is not straightforward to **remove** a callback. Not a big problem, but still...

tobiasjaster, 2022-05-29:
I just started a project using multiple callbacks. 

Usecase: different ESP32 use the same functions to publish state and handle OTA-Updater. Therefore on Callback is defined. The ESP32 specific task has its own callbacks and can be added separately.