marvinroger/async-mqtt-client #144: mqtt ondisconnect callback not working

jassi00713, 2019-05-22:
I have a working script based on AsyncMqttClient running on ESP12F.

I had developed the script on platformio

But after a recent update of the mqtt library, the mqtt ondisconnect callback stoped working.

When i backtrack to 0.8.2 version, everything is fine.
As i update to the latest version available here, the ondisconnect callback stops working.

onConnect
onSubscribe
 mqttClient.connect() ... etc ... everything is working.

But on sudden loss of internet at my router, the mqttclient doesnt call the ondisconnect callback.

It remains like this for infinity amount of time, there is no Timeout called, no ondisconnect callback, nothing.

Kindly look into this.
This is weird because i ran a loop to check in realtime wheather the mqttconnection is present.
In case of internet loss, the mqttclient.connected()  -> returns False as expected but the callback doesn't occur.

It was working in 0.8.2

bertmelis, 2019-05-22:
does `mqttclient.connected()` return `false` immideately after losing connection or does it take some time (timeout)?

jassi00713, 2019-05-24:
it takes some time ... doesnt happen immediatelt

voidEvgeny, 2019-05-29:
Force disconnect cable from the broker or internet cable if your broker in internet and you see how this not work.

jeroenst, 2019-11-29:
https://github.com/marvinroger/async-mqtt-client/issues/160

bertmelis, 2021-07-22:
Disconnecting is reworked completely in develop which should solve this issue. Closing.

closed 2021-07-22 by bertmelis