marvinroger/async-mqtt-client #145: onDisconnect called before connection can finish

QuantusHoka, 2019-06-13:
I modified the ESP8266 full example to manage connections to my broker, to send basic environment data for a remote sensor. It worked great for the last few months, until I made a small change, recompiled and uploaded. Now onDisconnect seems to get called before the connect can complete, leading to and endless loop of connect, onDisconnect, connect... If I comment the disconnect callback the ESP8266 connects flawlessly (but I obviously also lose connection handling). Suggestions?

Code
```
#include <Arduino.h>
#include <SPI.h>
#include <ArduinoJson.h>
#include <Wire.h>
#include <BME280I2C.h>
#include <Ticker.h>
#include <ESP8266WiFi.h>
#include <AsyncMqttClient.h>
#include <ArduinoOTA.h>

#define WIFI_SSID "HokasHouse"
#define WIFI_PASSWORD "Vsd5h#5e9"
#define SDA_PIN     D2
#define SCL_PIN     D1
#define BUTTON_PIN  D3

#define DEBUG_SERIAL

#ifdef DEBUG_SERIAL
  #define DEBUG_PRINTLN(x) Serial.println(x)
  #define DEBUG_PRINT(x) Serial.print(x)
#else
  #define DEBUG_PRINTLN(x) 
  #define DEBUG_PRINT(x) 
#endif

/*-----------------------------------MQTT Basic Stuff-------------------------*/
//#define MQTT_HOST IPAddress(999, 999, 999, 999)
#define MQTT_HOST "my.host.com"
#define MQTT_PORT 1883
#define MQTT_ROOT_TOPIC "environment"
#define MQTT_SETTING_SUBTOPIC "set"
#define MQTT_STATUS_SUBTOPIC "status"
#define MQTT_REPORTS_SUBTOPIC "reports"
#define DEVICE_NAME "Outside"
#define UPDATE_INTERVAL 5

AsyncMqttClient mqttClient;
Ticker mqttReconnectTimer;
WiFiEventHandler wifiConnectHandler;
WiFiEventHandler wifiDisconnectHandler;
Ticker wifiReconnectTimer;
Ticker iotUpdateTimer;

float h, t, p;
BME280I2C bme;

void connectToWifi() {
  DEBUG_PRINTLN("Connecting to Wi-Fi...");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
}

void connectToMqtt() {
  DEBUG_PRINTLN("Connecting to MQTT...");
  mqttClient.connect();
}
 
void onWifiConnect(const WiFiEventStationModeGotIP& event) {
  DEBUG_PRINTLN("Connected to Wi-Fi.");
  ArduinoOTA.begin();
  DEBUG_PRINTLN("OTA Ready");
  DEBUG_PRINT("IP address: ");
  DEBUG_PRINTLN(WiFi.localIP());
  connectToMqtt();
}

void onWifiDisconnect(const WiFiEventStationModeDisconnected& event) {
  DEBUG_PRINTLN("Disconnected from Wi-Fi.");
  iotUpdateTimer.detach();
  mqttReconnectTimer.detach(); // ensure we don't reconnect to MQTT while reconnecting to Wi-Fi
  wifiReconnectTimer.once(2, connectToWifi);
}

void updateIOT()
{
  bme.read(p, t, h);
//  p = p;
  #ifdef DEBUG_SERIAL
    Serial.printf("Temp: %.1fC    Press: %.1fmbar    Hum: %.1f%%\n", t, p, h);
  #endif

  char topicBuffer[64];
  sprintf(topicBuffer, "%s/%s/%s", MQTT_ROOT_TOPIC, DEVICE_NAME, MQTT_REPORTS_SUBTOPIC);
  StaticJsonDocument<128> jsonMessage;
  jsonMessage["sensor"] = DEVICE_NAME;
  jsonMessage["temp"] = t;
  jsonMessage["press"] = p;
  jsonMessage["hum"] = h;
  char jsonMessageBuffer[128];
  serializeJson(jsonMessage, jsonMessageBuffer);
  DEBUG_PRINTLN(jsonMessageBuffer);
  mqttClient.publish(topicBuffer, 1, true, jsonMessageBuffer);
}

/*-------------------------------------------------MQTT Stuff----------------------------------------------*/
void onMqttConnect(bool sessionPresent)
{
  DEBUG_PRINTLN("** Connected to the broker **");
  DEBUG_PRINTLN("Session present: ");
  DEBUG_PRINTLN(sessionPresent);
  char topicBuffer[64];
  sprintf(topicBuffer, "%s/%s/%s", MQTT_ROOT_TOPIC, DEVICE_NAME, MQTT_SETTING_SUBTOPIC);
  mqttClient.subscribe(topicBuffer, 1);
  DEBUG_PRINT("Subscribing: ");
  DEBUG_PRINTLN(topicBuffer);
  updateIOT();
  iotUpdateTimer.attach(UPDATE_INTERVAL * 60, updateIOT);
}

void onMqttDisconnect(AsyncMqttClientDisconnectReason reason)
{
  #ifdef DEBUG_SERIAL
    Serial.printf("** Disconnected from the broker ** %d", reason);
    Serial.printf("WiFi: %u", WiFi.isConnected());
  #endif
  iotUpdateTimer.detach();
  if (WiFi.isConnected()) { mqttReconnectTimer.once(2, connectToMqtt);}
}

void onMqttSubscribe(uint16_t packetId, uint8_t qos)
{
  DEBUG_PRINT("** Subscribe acknowledged **");
  DEBUG_PRINT(" packetId: ");
  DEBUG_PRINT(packetId);
  DEBUG_PRINT(" qos: ");
  DEBUG_PRINTLN(qos);
}

void onMqttUnsubscribe(uint16_t packetId)
{
  DEBUG_PRINTLN("** Unsubscribe acknowledged **");
  DEBUG_PRINT("  packetId: ");
  DEBUG_PRINTLN(packetId);
}

void onMqttPublish(uint16_t packetId)
{
  DEBUG_PRINT("** Published");
  DEBUG_PRINT("  packetId: ");
  DEBUG_PRINT(packetId);
  DEBUG_PRINT("\n\n");
}

void onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total)
{
  DEBUG_PRINTLN("** Message received **");
  DEBUG_PRINT("  topic: ");
  DEBUG_PRINT(topic);
  DEBUG_PRINT("  qos: ");
  DEBUG_PRINT(properties.qos);
  DEBUG_PRINT("  dup: ");
  DEBUG_PRINT(properties.dup);
  DEBUG_PRINT("  retain: ");
  DEBUG_PRINT(properties.retain);
  DEBUG_PRINT("  len: ");
  DEBUG_PRINT(len);
  DEBUG_PRINT("  index: ");
  DEBUG_PRINT(index);
  DEBUG_PRINT("  payload: ");
  DEBUG_PRINTLN(payload);

  /* Not implemented Yet
  StaticJsonDocument<128> jsonMessage;
  deserializeJson(jsonMessage, payload);
  */
}

// ---------------------------------------- SETUP -----------------------------------------------
void setup() {
  #ifdef DEBUG_SERIAL
    Serial.begin(115200);
  #endif
  DEBUG_PRINTLN();
  DEBUG_PRINTLN();

  Wire.begin(SDA_PIN, SCL_PIN);
  while(!bme.begin())
  {
    DEBUG_PRINTLN("Could not find BME280 sensor!");
  }

  wifiConnectHandler = WiFi.onStationModeGotIP(onWifiConnect);
  wifiDisconnectHandler = WiFi.onStationModeDisconnected(onWifiDisconnect);

  char topicBuffer[64];
  char msgBuffer[64];
  sprintf(topicBuffer, "%s/%s/%s", MQTT_ROOT_TOPIC, MQTT_STATUS_SUBTOPIC, DEVICE_NAME);
  sprintf(msgBuffer, "%s is dead?", DEVICE_NAME);
  mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  mqttClient.onSubscribe(onMqttSubscribe);
  mqttClient.onUnsubscribe(onMqttUnsubscribe);
  mqttClient.onMessage(onMqttMessage);  
  mqttClient.onPublish(onMqttPublish);
  mqttClient.setServer(MQTT_HOST, MQTT_PORT);
  mqttClient.setCleanSession(false);
  mqttClient.setWill(topicBuffer, 2, true, msgBuffer);

  ArduinoOTA.onStart([]() {
    DEBUG_PRINTLN("Start");
  });
  ArduinoOTA.onEnd([]() {
    DEBUG_PRINTLN("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    #ifdef DEBUG_SERIAL
      Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    #endif
  });
  ArduinoOTA.onError([](ota_error_t error) {
    #ifdef DEBUG_SERIAL
      Serial.printf("Error[%u]: ", error);
    #endif
    if (error == OTA_AUTH_ERROR) DEBUG_PRINTLN("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) DEBUG_PRINTLN("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) DEBUG_PRINTLN("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) DEBUG_PRINTLN("Receive Failed");
    else if (error == OTA_END_ERROR) DEBUG_PRINTLN("End Failed");
  });

  connectToWifi();

}

// ================================================ LOOP =================================================
void loop()
{
  ArduinoOTA.handle();
}
```

QuantusHoka, 2019-06-13:
NM. It appears that the culprit actually lies somewhere in the interaction between the onDisconnect callback and the most recent implementation of ArduinoOTA. I'm blaming ArdunioOTA. I don't know how to fix it, though...

arihantdaga, 2019-08-27:
@QuantusHoka . I know its a bit late. But can you try this PR - https://github.com/marvinroger/async-mqtt-client/pull/148/files. It may help.

luebbe, 2021-02-12:
Should be fixed by PR #165

closed 2021-02-12 by luebbe