marvinroger/async-mqtt-client #147: Calling connect() from _onDisconnect() user callback fails

obrain17, 2019-06-17:
In current master if you call` connect()` directly within a callback routine for `_onDisconnect()` this would do nothing because the callback is called **before** the status was set to `_connected = false;`  with `_clear()`.  ` connect()` will return immediately.

AsyncMqttClient.cpp line 352:

```
    }
    for (auto callback : _onDisconnectUserCallbacks) callback(reason);
  }
  _clear();
 
```
AsyncMqttClient.cpp line 694:

```
void AsyncMqttClient::connect() {
  if (_connected) return;

```
So existing sketches or libraries (e.g.  [https://github.com/homieiot/homie-esp8266/blob/develop/src/Homie/Boot/BootNormal.cpp](url)) using this direct re-connect would not work anymore.

In former versions or releases `_clear()` was called before the callbacks.

If you **use a timer** as in the examples it is still working of course. I suggest this should be the recommendation for the usage. Even if one callback would establish the MQTT reconnect the others (if exist) would run in an undefined status if _connected or not.

obrain17, 2019-06-17:
See pull request #148

jeroenst, 2019-11-30:
Ok, checkout my pull request. I'm running the code now on 20 esp's also with ssl enabled on 1 device.

Also tested several connection losses with successful onDisconnect

obrain17, 2019-12-01:
@jeroenst I checked you pull request. One minor flaw:
`clear()` is now too far up, so it resets the reasons (e. g. _connectPacketNotEnoughSpace) before variable `reason` is set.
`clear()` should be directly before `for (auto callback : … ` (L368 in new code) so it does  `_connected = false;`, what is needed to fix issue #147 / PR #148

jeroenst, 2019-12-01:
You're right, I have fixed it!

jeroenst, 2019-12-14:
Can someone confirm the fix is in master branch and close this bug?

obrain17, 2019-12-15:
I confirm the issue is fixed with PR  #165. Thank you

closed 2019-12-15 by obrain17