marvinroger/async-mqtt-client #150: std::bind() does not work as onConnect() callback

joede, 2019-07-03:
I'm using `AsyncMqttClient` as attribute inside my on class `MqttBindings`. Now I want to setup the callbacks **not** by using static functions. Instead I want to use `std::bind`.

Here a small snippet:

~~~~
class MqttBindings
{
  public:
    void onMqttConnect(bool sessionPresent);
    void onMqttDisconnect(AsyncMqttClientDisconnectReason reason);
    void setupClient()
    {
        m_Client.onConnect(std::bind(&MqttBindings::onMqttConnect,this));
        m_Client.onDisconnect(std::bind(&MqttBindings::onMqttDisconnect,this));
    };

  privat;
    AsyncMqttClient m_Client;
    String m_ClientId;
}
~~~~

The resulting error message is:
~~~~
src/MqttBindings.cpp: In member function 'void MqttBindings::setupClient()':
src/MqttBindings.cpp:10:66: error: no matching function for call to 'AsyncMqttClient::onConnect(std::_Bind_helper<false, void (MqttBindings::*)(bool), MqttBindings* const>::type)'
m_Client.onConnect(std::bind(&MqttBindings::onMqttConnect,this));
^
src/MqttBindings.cpp:10:66: note: candidate is:
In file included from .piolibdeps/AsyncMqttClient_ID346/src/AsyncMqttClient.h:4:0,
from src/MqttBindings.h:9,
from src/MqttBindings.cpp:3:
.piolibdeps/AsyncMqttClient_ID346/src/AsyncMqttClient.hpp:58:20: note: AsyncMqttClient& AsyncMqttClient::onConnect(AsyncMqttClientInternals::OnConnectUserCallback)
AsyncMqttClient& onConnect(AsyncMqttClientInternals::OnConnectUserCallback callback);
^
~~~~

I think is is my fault, but I can't figure out why. *Please can somebody help me!*

By the way: `Ticker::once` is working this way.

joede, 2019-07-04:
To bad... Realy my fault. :-)  I've forgotten to add the `std::parameters` like `_1`! But I wasn't  able to get this out of the error message.

closed 2019-07-04 by joede

Paul-Simpson, 2021-02-15:
Hey, reading your comments help me - thankyou! Did you get this working with mqttClient.onMessage?