marvinroger/async-mqtt-client #153: Connection fault finding

HLNZ, 2019-07-11:
Hi, sorry to make the issue for my question.
I have compiled the code the esp32 following the other instructions you provided.
My esp is connecting to the wifi network OK, but I am not able to make a connection to the windows mqtt server/broker. 
The full featured code example gets stuck in a loop of being disconnected from the broker, then trying to re-connect.
My issue is, how do I figure out what the issue is with connecting to the broker?
Is there some debugging option available to give me some more hints as to why the esp is unable to connect to the broker???

HLNZ, 2019-08-26:
Hi,
Found my issues as a poorly configured mqtt broker :/
Still wanting some more feedback from the esp as to why the device couldn't connect to broker....

bertmelis, 2019-08-26:
Tip: check the logs of the broker.
another tip: for TCP errors, enable errors in the adynctcp-library.

jeroenst, 2019-11-29:
Can this issue be closed?

HLNZ, 2019-12-12:
Just using counters to do reconnects if it fails enough times.

closed 2019-12-12 by HLNZ