marvinroger/async-mqtt-client #160: onDisconnect callbacks not triggered after `client.disconnect(false)`

nougad, 2019-10-06:
It looks like https://github.com/marvinroger/async-mqtt-client/pull/83 changed the behavior of `onDisconnect` hooks. When `client.disconnect(false)` is called `_disconnectFlagged` is set to `true`: https://github.com/marvinroger/async-mqtt-client/blob/master/src/AsyncMqttClient.cpp#L718 and reset back after the disconnect: https://github.com/marvinroger/async-mqtt-client/blob/master/src/AsyncMqttClient.cpp#L675

At least in my case this means https://github.com/marvinroger/async-mqtt-client/blob/master/src/AsyncMqttClient.cpp#L343 will evaluate to `false` and user callbacks in https://github.com/marvinroger/async-mqtt-client/blob/master/src/AsyncMqttClient.cpp#L353 will not be triggered. Not sure how exactly the async part works, it might be just a race-condition.

There is nothing in the PR which indicates this change was intentional. His breaks deep-sleep functionality in https://github.com/homieiot/homie-esp8266/tree/develop-v3 which relies on disconnect callback after calling `disconnect`.

obrain17, 2019-11-25:
You are right it seems to be a race condition. 
In https://github.com/marvinroger/async-mqtt-client/blob/master/src/AsyncMqttClient.cpp#L675 _`disconnectFlagged` is set to` false`. (what is a condition for the `onDisconnect` hooks to be served.)
Unfortunately  `onDisconnect`  gets already called before, assumably from ` _client.send();` or `_client.close(true)`; some lines above. 
Shifting the `_disconnectFlagged = false;` up some lines at least brought the `OnDisconnect` messages back in my case.

A second issue, also connected with this, was that the `disconnect();` in https://github.com/marvinroger/async-mqtt-client/blob/master/src/AsyncMqttClient.cpp#L452 also would not trigger the  `OnDisconnect` callback, thus preventing to automatically re-connect within my code.  

I will do some long-time tests and if it works then create a PR

jeroenst, 2019-11-29:
I thing the _disconnectFlagged is mixed up for 2 things:
1. To start disconnecting in the in the _onPoll() function
2. To prevent calling multiple onDisconnect callbacks.

This causes that when _disconnectFlagged is true, the callback is prevented.

What could help is swap _sendDisconnect(); and _disconnectFlagged = true in the disconnect function. So it's still prevented that the onDisconnect callback is called multiple times, while also doing a disconnect when the ping times out in the onPoll funtion
`
void AsyncMqttClient::disconnect(bool force) {
  if (!_connected) return;

  if (force) {
    _client.close(true);
  } else {
    _sendDisconnect();
    _disconnectFlagged = true;
  }
}
`

I'm not sure this solves the issue, but it seems plausible.

I will test this tomorrow.

jeroenst, 2019-11-30:
Investigating more, even multiple calls to the ondisconnect callbacks are not prevented, because there is a clear in the ondisconnect function...

I think the only function that has to be remained is the prevention of multiple disconnects in the OnPoll function.

I've adapted the code, will run some tests and if it works, I will create a pull request.

obrain17, 2019-11-30:
There are only 2 positions in the code with `_disconnectFlagged = true; `

1. in the ` _onConnAck()` function (L489)
2. in the  `disconnect()` function (L718)

Both prevent the callback being called the **next time only**, because `clear()` resets the flag in `onDisconnect`. For the  `_onConnAck()` function this behavior is OK, because there is a separate call of the callback-functions . ( L488)
But `disconnect()` does not do an extra call and so needs to call the callbacks the first time as well.
Just remove  `_disconnectFlagged = true` in `disconnect()` in L718 and it will work like this.
https://github.com/marvinroger/async-mqtt-client/blob/master/src/AsyncMqttClient.cpp#L718

I am also running some tests, @jeroenst waiting for your pull-request

jeroenst, 2019-11-30:
I also changed the _disconnectFlagged to _disconnectOnPoll because I think that is less confusing.
The onDisconnect callback is only called from the ondisconnect function that is called from the async tcp lib.
I've tested this and it seems to work fine.

obrain17, 2019-11-30:
Looks good.
"The onDisconnect callback is only called from the ondisconnect function … " would return reason 0 (TCP_DISCONNECTED) in case of failed `OnConnect` then.

May I suggest you additionally shift the `clear()` up one line to ensure that in` _onDisconnect()` `_clear() `is done in any case **before** calling user callbacks. (L366 in your new code). This will then also fix the issue #147 and commit my PR #148

jeroenst, 2019-11-30:
Yes, I will change that tomorrow.

My pull request will then solve 3 bugs:
- This one
- https://github.com/marvinroger/async-mqtt-client/issues/107
- https://github.com/marvinroger/async-mqtt-client/issues/147

I hope @marvinroger  has some time to apply the pull request in the following weeks.

closed 2019-12-08 by marvinroger