marvinroger/async-mqtt-client #161: mqttClient.connect() never finishes

tomastrimax, 2019-10-11:
I'm having trouble starting a MQTT connection

I'm using most of the ESP32 example and modified it to connect to Ubidots MQTT broker.

When I call mqttClient.connect(); it never seems to do anything and never gets to onMqttConnect() callback.

Are there any debugging messages I can enable with this library to see where it's getting stuck?
Rest of my code continues to run fine.

Assume it's going to be something simple but not sure how to even begin debugging.

```
#include "Arduino.h"
#include <TinyGPS++.h>
#include "BluetoothSerial.h"
#include "Diagnostics.h"

#include <WiFi.h>
extern "C" {
	#include "freertos/FreeRTOS.h"
	#include "freertos/timers.h"
}
#include <AsyncMqttClient.h>

#define TOKEN "########################" // Your Ubidots TOKEN
#define WIFINAME "############" //Your SSID
#define WIFIPASS "###########" // Your Wifi Pass

#define MQTT_HOST "industrial.api.ubidots.com"
#define MQTT_PORT 1883

static const uint32_t GPSBaud = 9600;
extern boolean serialDebug;

// Objects
TinyGPSPlus gps;
extern BluetoothSerial ESP_BT;
AsyncMqttClient mqttClient;
TimerHandle_t mqttReconnectTimer;
TimerHandle_t wifiReconnectTimer;

void setupDiagnostics()
{
  pinMode(14, INPUT);
  setupMQTT();
}

void connectToWifi() {
 Serial.println("Connecting to Wi-Fi...");
  WiFi.begin(WIFINAME, WIFIPASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(300);
  }
}

void connectToMqtt() {
  Serial.println("Connecting to MQTT..."); 
  mqttClient.connect();
}

void WiFiEvent(WiFiEvent_t event) {
    Serial.printf("[WiFi-event] event: %d\n", event);
    switch(event) {
    case SYSTEM_EVENT_STA_START:
        Serial.println("WiFi Started");
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        if(serialDebug == true){ 
          Serial.println("WiFi connected");
          Serial.print("IP address: ");
          Serial.println(WiFi.localIP()); 
          }
        Serial.println("WiFi connected");
        Serial.print("IP address: ");
        Serial.println(WiFi.localIP());
        connectToMqtt();
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        if(serialDebug == true){ Serial.println("WiFi lost connection"); }
        Serial.println("WiFi lost connection");
        xTimerStop(mqttReconnectTimer, 0); // ensure we don't reconnect to MQTT while reconnecting to Wi-Fi
		    xTimerStart(wifiReconnectTimer, 0);
        break;
    default:
        Serial.println("WiFi default case");
        break;
    }
}

void onMqttConnect(bool sessionPresent) {
  if(serialDebug == true){ 
    Serial.println("Connected to MQTT.");
    Serial.print("Session present: ");
    Serial.println(sessionPresent);
    }
  Serial.println("Connected to MQTT.");
  Serial.print("Session present: ");
  Serial.println(sessionPresent);
  //uint16_t packetIdSub = mqttClient.subscribe("test/lol", 2);
  //Serial.print("Subscribing at QoS 2, packetId: ");
  //Serial.println(packetIdSub);
  char * testTime = (char *) millis();
  mqttClient.publish("/v1.6/devices/MLSTest", 0, true, testTime);
  if(serialDebug == true){ Serial.println("Publishing at QoS 0"); }
  Serial.println("Publishing at QoS 0");
  /*uint16_t packetIdPub1 = mqttClient.publish("test/lol", 1, true, "test 2");
  Serial.print("Publishing at QoS 1, packetId: ");
  Serial.println(packetIdPub1);
  uint16_t packetIdPub2 = mqttClient.publish("test/lol", 2, true, "test 3");
  Serial.print("Publishing at QoS 2, packetId: ");
  Serial.println(packetIdPub2);*/
}

void onMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
  if(serialDebug == true){ Serial.println("Disconnected from MQTT."); }
  Serial.println("Disconnected from MQTT.");

  if (WiFi.isConnected()) {
    xTimerStart(mqttReconnectTimer, 0);
  }
}

void onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {
  Serial.println("Publish received.");
  Serial.print("  topic: ");
  Serial.println(topic);
  Serial.print("  qos: ");
  Serial.println(properties.qos);
  Serial.print("  dup: ");
  Serial.println(properties.dup);
  Serial.print("  retain: ");
  Serial.println(properties.retain);
  Serial.print("  len: ");
  Serial.println(len);
  Serial.print("  index: ");
  Serial.println(index);
  Serial.print("  total: ");
  Serial.println(total);
}

void onMqttPublish(uint16_t packetId) {
  Serial.println("Publish acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
}

void setupMQTT(){ 
  
  mqttReconnectTimer = xTimerCreate("mqttTimer", pdMS_TO_TICKS(10000), pdFALSE, (void*)0, reinterpret_cast<TimerCallbackFunction_t>(connectToMqtt));
  wifiReconnectTimer = xTimerCreate("wifiTimer", pdMS_TO_TICKS(2000), pdFALSE, (void*)0, reinterpret_cast<TimerCallbackFunction_t>(connectToWifi));
  
  WiFi.onEvent(WiFiEvent);

  mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  mqttClient.onMessage(onMqttMessage);
  mqttClient.onPublish(onMqttPublish);
  mqttClient.setServer(MQTT_HOST, MQTT_PORT);
  mqttClient.setCredentials(TOKEN, "a");
  //mqttClient.setKeepAlive(120);
  //mqttClient.setCleanSession(true);

  connectToWifi();
}
```

sskorol, 2020-03-15:
I can confirm the same behaviour. Sometimes it establishes connection, sometimes not.

tomastrimax, 2020-03-15:
@sskorol Have you tried this with more than one device? I'm using a Samsung S9 but have nothing else to test with.

sskorol, 2020-03-15:
Not yet. But I found this interesting PR to play with: https://github.com/marvinroger/async-mqtt-client/pull/156/files

davidkester, 2020-05-10:
I have the same problem, tested on two different ESP32. 
On the ESP-01 it works fine

bertmelis, 2021-02-24:
Should work fine with the `develop` branch. If not, feel free to reopen the issue.

closed 2021-02-24 by bertmelis