marvinroger/async-mqtt-client #166: 2 ESP32s been given the same ClientId

Jasea, 2019-11-04:
Setup
  Latest levels of MQTT and Mosquitto installed on Raspberry PI running latest levels of Buster.
   2 ESP32 are attempting to connect with each other but their respective onMqttDisconnect() routines are being constantly driven. The reason is the both ESP32s are being given the same ClientIDs, see Mosquito log

Mosquitto Log

1572794767: New client connected from 192.168.1.178 as esp32ba3a7d80 (c1, k15).
1572795026: New connection from 192.168.1.84 on port 1883.
1572795026: Client esp32ba3a7d80 already connected, closing old connection.
1572795026: Socket error on client esp32ba3a7d80, disconnecting.
1572795026: New client connected from 192.168.1.84 as esp32ba3a7d80 (c1, k15).
1572795028: New connection from 192.168.1.178 on port 1883.
1572795028: Client esp32ba3a7d80 already connected, closing old connection.
1572795028: Socket error on client esp32ba3a7d80, disconnecting.
1572795028: New client connected from 192.168.1.178 as esp32ba3a7d80 (c1, k15).

The MAC addresses of the ESP32s from WiFi.macAddress() are
     ESP32 1.  80:7D:3A:BA:DB:7C
     ESP32 2.  80:7D:3A:BA:EF:BC

the Chipids from ESP.getEfuseMac() are
     ESP32 1.   7CDBBA3A7D80
      ESP32 2.  BCEFBA3A7D80
So these are the MAC address but in reverse order.

in AsyncMqttClient.hpp There is the following definition

Char _generatedClientId[13+1];  // esp8266abc123

in AsyncMqttClient.cpp  there is the following line of code

#ifdef ESP32
   sprintf(_generatedClientID, “esp32%06x”, ESP.getEfuseMac());  

This I assume should add the reversed MAC address to esp32, but this would be too long for the definition in  AsyncMqttClient.hpp. I am not too sure if the outcome would be the problem.

Regards
John

jeroenst, 2019-11-29:
Maybe remove 32 from esp32 and use %08x?

Or change the size of _generatedClientId? 

You can also use setClientId (be aware to use a static variable because it is a pointer.

jeroenst, 2019-11-30:
https://github.com/marvinroger/async-mqtt-client/pull/103#issuecomment-560028439

closed 2019-12-08 by marvinroger