marvinroger/async-mqtt-client #168: The onMessage callback adds some cyrillic endings to the topic and the payload message

x0tester0x, 2019-11-07:
Example:
I'm sending "LOCATE" and I will get this "LOCATE⸮" or this "LOCATE⸮⸮\l⸮⸮⸮r @" or this "LOCATEXEZT⸮⸮x⸮!@⸮⸮⸮?L⸮⸮?" back.

I couldn't figure out what's the mistake.
Can someone help me, please?

jeroenst, 2019-11-07:
The string is not 0 terminated, so use the parameter len to print the number of characters.

x0tester0x, 2019-11-07:
Ok, I understand. It works fine now, but how can I then store the right value in a variable?

jeroenst, 2019-11-07:
Depends on what you want...

In my code I do this:
String payloadstring = "";
  for (int i = 0; i < len; i++)
  {
    payloadstring += char(payload[i]);
  }

x0tester0x, 2019-11-07:
Later I want to compare it to another String...

jeroenst, 2019-11-07:
You can use my example to compare it to your string.

x0tester0x, 2019-11-07:
Yes I see. Is there not an existing function from Arduino to do this?

jeroenst, 2019-11-07:
You can use strncat... But it makes life more difficult because you must declare an array of characters big enough to hold the incoming payload and prevent an overrun, also you have to use strcmp.

x0tester0x, 2019-11-07:
A ok, but is there no alternative to a loop?

jeroenst, 2019-11-07:
Maybe you can use:

char *str = realloc(payload, (len+1) * sizeof(char));
str[len] = 0;

x0tester0x, 2019-11-07:
Could I not just add the null terminator and convert it with String(payload)?

jeroenst, 2019-11-07:
Nope, because the payload variable has no room left at the end to add the null terminator (that's why you have to do a realloc or copy it character by character to a string)....

x0tester0x, 2019-11-07:
Ok it looks like there is no other way around the loop... 
Then I have to use the loop

Thanks a lot for your help :)

jeroenst, 2019-11-07:
You're welcome ;-)

closed 2019-11-08 by x0tester0x