marvinroger/async-mqtt-client #170: I got a strange data corruption issue

Speedy2k, 2019-11-22:
When i send this payload:

`{"info":{"upTime":["000:00:02:49",2]},"serial":"040300000","timeStamp":"2019-11-22, 12:35:43"}`

I receive on the MQTT server this:

`{"info":{"upTime":["000:00:02:49",2]},"serial":"040300000","timeStamp":"2019-11-22, 12:35:43"}����������������������������������������������������������������������������������������������������������������������L@�@0���0�?# #X��?�L@D�@0(^�`�?���?0��?`�?D��@�?���?���? 
��?��@��@�����N@ĺ@5�?��?������?,��?��?�����?@��?��?������
2���?�/@��?32���?�/@xV������?�@?�D���?��?8��?�/@��?@��?<d�?�xV���E�0�?��?��? ���?@?xV���8��?^^0��?NQ�p�?��?�E�`�?8��?&�p�?�E���?h��?|�? xV��dI���?h��?���?���?T��?R���?��?���?��?12:3*R���?��?���?*R���?��?h��?���?t��?`�?`�?@R�
�?��?P
�?���?�@?�F
�@
�?�
�? ^����?��
�P
�?H�?�
�?GG
�p
�?�
�?p�?܊
��
�?`�?�
�?l�?�@?dJ
��
�?�
�?`�?",2]},"serial":"�
�?�@?���
��
�?�
�?`

The command who send that is this:
`
    char topic[100];
    char payload[payloadSize];
    strcpy(topic, "data/releasers/");
    strcat(topic, serial);
    serializeJson(_doc, payload, sizeof(payload));
    mqttClient.publish(topic, 0, false, payload, sizeof(payload));
`

The max payload size is 1024.

jeroenst, 2019-11-22:
Your payload string is not null terminated. Use strlen instead of sizeof.

https://stackoverflow.com/questions/8590332/difference-between-sizeof-and-strlen-in-c/8590379

Speedy2k, 2019-11-23:
Worked!! thanx a lot!

Speedy2k, 2019-11-24:
Thanx a lot!! It as worked perfectly!!!

https://stackoverflow.com/questions/8590332/difference-between-sizeof-and-strlen-in-c/8590379

jeroenst, 2019-11-29:
Can you please close this issue?

closed 2019-11-29 by Speedy2k