marvinroger/async-mqtt-client #171: Unable to connect to remote MQTT via Internet

sreeramtkd, 2019-12-07:
//All that I get is repeated below.... If i'm on local wifi username & PWD I get connected otheriwse the esp8266 goes on a spin as below.. (eg. hotspot or from some other n/w).
/*
Disconnected from MQTT.
Connecting to MQTT...
Connecting to MQTT...
Disconnected from MQTT.
Connecting to MQTT...
Connecting to MQTT...
*/

//Libs
#include <ESP8266WiFi.h>
#include <Ticker.h>
#include <AsyncMqttClient.h>
// Button Lib for differentiating button actions
#include <AceButton.h>
using namespace ace_button;

//Credentials & Hosts
#define WIFI_SSID "wifi"
#define WIFI_PASSWORD "password"
const char* MQTTUsername="username";
const char* MQTTPassword="password";
#define MQTT_HOST ""
#define MQTT_PORT 5883 // PORT opened in my router
//GPIOs
const int ac_relay1 = 12; // D6
const int ac_relay2 = 13; // D7
const int motionSensor = 5; // D1
//Push button Config
const int BUTTON_PIN = 14; // D5
const int BUTTON_LED = 4; // D2
AceButton button(BUTTON_PIN);
AceButton button2(motionSensor);

void handleEvent(AceButton*, uint8_t, uint8_t);

//Timer & Flags
#define timeSeconds 900 // Time in seconds to keep the Device running
unsigned long now = millis();
unsigned long lastTrigger = 0;
boolean startTimer = false;
unsigned long previousMillis = 0;
String ledState;
int MAIN_FLAG=0;

AsyncMqttClient mqttClient;
Ticker mqttReconnectTimer;

WiFiEventHandler wifiConnectHandler;
WiFiEventHandler wifiDisconnectHandler;
Ticker wifiReconnectTimer;

void connectToWifi() {
  Serial.println("Connecting to Wi-Fi...");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  digitalWrite(BUTTON_LED,!digitalRead(BUTTON_LED));
 }

void onWifiConnect(const WiFiEventStationModeGotIP& event) {
  Serial.println("Connected to Wi-Fi.");
  connectToMqtt();
}

void onWifiDisconnect(const WiFiEventStationModeDisconnected& event) {
  Serial.println("Disconnected from Wi-Fi.");
  mqttReconnectTimer.detach(); // ensure we don't reconnect to MQTT while reconnecting to Wi-Fi
  wifiReconnectTimer.once(2, connectToWifi);
   digitalWrite(BUTTON_LED,!digitalRead(BUTTON_LED));
}

void connectToMqtt() {
  Serial.println("Connecting to MQTT...");
  mqttClient.connect();
}

void onMqttConnect(bool sessionPresent) {
  Serial.println("Connected to MQTT.");
  Serial.print("Session present: ");
  Serial.println(sessionPresent);
  Serial.println ("Subscribing to Superb Urban Tree Messaging Queue...");
  //Subscribing to power topic for complete builing poweroff
  
  uint16_t packetIdSub = mqttClient.subscribe("power/SBUT", 2);
  Serial.print("Subscribing to Power Topic at QoS 2, packetId: ");
  Serial.println(packetIdSub);

  uint16_t packetIdSub2 = mqttClient.subscribe("power/SBUT/GYM", 2);
  Serial.print("Subscribing to GYM Topic at QoS 2, packetId: ");
  Serial.println(packetIdSub2);  
  mqttClient.publish("telem/SBUT/GYM", 2, false, "Alive");
  Serial.println("Published GYM is Alive to MQTT: telem/SBUT/GYM");
}

void onMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
  Serial.println("Disconnected from MQTT.");
  if (WiFi.isConnected()) {
    mqttReconnectTimer.once(2, connectToMqtt);
  }
}

void onMqttSubscribe(uint16_t packetId, uint8_t qos) {
  Serial.println("Subscribe acknowledged by the broker, All good !");
 }

void onMqttUnsubscribe(uint16_t packetId) {
  Serial.println("Unsubscribe acknowledged.");
}

// Need a Seperate Routine to Handle Logic, But for now added the on/off logic in this function itself
void onMqttMessage(char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {
  String messageTemp;
   for (int i = 0; i < len; i++) {
  messageTemp += (char)payload[i];
 }
  if (strcmp(topic, "power/SBUT") == 0){
    if(messageTemp=="OFF"){Serial.println("SHUT DOWN ALL INITIATED");MAIN_FLAG=0;all_off();}
    else if(messageTemp=="ON"){Serial.println("POWER ON ALL INITIATED");MAIN_FLAG=1;digitalWrite(BUTTON_LED,HIGH);}
  }
  if (strcmp(topic, "power/SBUT/GYM") == 0){
    if(messageTemp=="OFF"){Serial.println("POWER OFF for GYM INITIATED");MAIN_FLAG=0;all_off();}
    else if(messageTemp=="ON"){Serial.println("POWER ON for GYM INITIATED");MAIN_FLAG=1;digitalWrite(BUTTON_LED,HIGH);}
  }
  Serial.print("Message received for Topic: ");
  Serial.println(topic);
  Serial.print("Payload = ");
  Serial.println(messageTemp);
  String msg = topic+':'+messageTemp;
  mqttClient.publish("telem/SBUT/GYM", 2, false, msg.c_str());
}

void onMqttPublish(uint16_t packetId) {
  Serial.println("Published Telemetry Data Please check MQTT Host logs for details");
  }
  
void detectsMovement() {
  if (MAIN_FLAG==1){
  digitalWrite(ac_relay1, LOW); 
  digitalWrite(ac_relay2, LOW);
  startTimer = true;
  lastTrigger = millis();
  }
  String telem="Motion Detected Or Button Pressed in GYM & ON/OFF= ";
  telem= telem + MAIN_FLAG;
  mqttClient.publish("telem/SBUT", 2, false, telem.c_str());
  mqttClient.publish("telem/SBUT/GYM", 2, false, "2"); // Status for Sending Motion ON Message
  Serial.println(telem.c_str());
}

void all_off(){
  digitalWrite(ac_relay1, HIGH);
  digitalWrite(ac_relay2, HIGH);
  digitalWrite(BUTTON_LED,LOW);
}

void all_on(){
  digitalWrite(ac_relay1, LOW);
  digitalWrite(ac_relay2, LOW);
  digitalWrite(BUTTON_LED,HIGH);
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  wifiConnectHandler = WiFi.onStationModeGotIP(onWifiConnect);
  wifiDisconnectHandler = WiFi.onStationModeDisconnected(onWifiDisconnect);
  mqttClient.onConnect(onMqttConnect);
  mqttClient.onDisconnect(onMqttDisconnect);
  mqttClient.onSubscribe(onMqttSubscribe);
  mqttClient.onUnsubscribe(onMqttUnsubscribe);
  mqttClient.onMessage(onMqttMessage);
  mqttClient.onPublish(onMqttPublish);
  mqttClient.setServer(MQTT_HOST, MQTT_PORT);
  mqttClient.setCredentials(MQTTUsername,MQTTPassword);
  connectToWifi();
  pinMode(motionSensor, INPUT_PULLUP);
  pinMode(BUTTON_PIN, INPUT_PULLUP);
  pinMode(ac_relay1, OUTPUT);
  pinMode(ac_relay2, OUTPUT);  
  pinMode(2,OUTPUT);
  pinMode(BUTTON_LED,OUTPUT);
  //Setting inital state
  digitalWrite(ac_relay1, HIGH);
  digitalWrite(ac_relay2, HIGH);
  ButtonConfig* buttonConfig = button.getButtonConfig();
  ButtonConfig* buttonConfig2 = button2.getButtonConfig();
  buttonConfig->setEventHandler(handleEvent);
  buttonConfig->setFeature(ButtonConfig::kFeatureDoubleClick);
  buttonConfig->setFeature(ButtonConfig::kFeatureSuppressAfterClick);
  buttonConfig->setFeature(ButtonConfig::kFeatureLongPress);

  buttonConfig2->setEventHandler(handleEvent);
  buttonConfig2->setFeature(ButtonConfig::kFeatureDoubleClick);
  buttonConfig2->setFeature(ButtonConfig::kFeatureSuppressAfterClick);
  buttonConfig2->setFeature(ButtonConfig::kFeatureLongPress);
}

void loop() {
  now = millis();
  unsigned long currentMillis = millis();
  if(startTimer && (now - lastTrigger > (timeSeconds*1000))) {
    Serial.println("Timer stopped...");
    mqttClient.publish("telem/SBUT/GYM", 2, false, "3"); // Status for Sending Motion OFF Message
    digitalWrite(ac_relay1, HIGH);
    digitalWrite(ac_relay2, HIGH);
    startTimer = false;
  } 
  button.check();
  button2.check();
}
// The event handler for the button.
void handleEvent(AceButton* , uint8_t eventType, uint8_t /* buttonState */) {
  switch (eventType) {
    case AceButton::kEventLongPressed:      
      Serial.println("Button Long pressed / Motion Detected");
      for(int i=0;i<=8;i++){
        digitalWrite(BUTTON_LED,LOW);
        delay(100);
        digitalWrite(BUTTON_LED,HIGH);
        delay(100);
      }
      if (MAIN_FLAG==0){digitalWrite(BUTTON_LED,LOW); all_off();}
      else if(MAIN_FLAG==1){digitalWrite(BUTTON_LED,HIGH);}
      detectsMovement();
      break;
    case AceButton::kEventDoubleClicked:
      if (MAIN_FLAG==0){
      digitalWrite(BUTTON_LED,HIGH);
      Serial.println("Turned ON Using Button");
      MAIN_FLAG=1;
      mqttClient.publish("power/SBUT/GYM", 2, true, "ON");
      } 
      else if (MAIN_FLAG==1){
      Serial.println("Turned Off Using Button");
      MAIN_FLAG=0;
      mqttClient.publish("power/SBUT/GYM", 2, true, "OFF");
      }      
      break;
  }
}

jeroenst, 2019-12-07:
Some logging would be nice..

bertmelis, 2019-12-07:
Don't forget the logging from the MQTT broker. And please use code tags.

sreeramtkd, 2019-12-08:
sorry about that !

Here is the MQTT logs. Nothing suspicious though ! The weird thing is the same port & host-name works if I connect the wemos to my home n/w wifi credentials where the rpi is connected. however if I use another n/w like mobile hotspot / wifi this behaviour happens. But when I tried to Pub/Sub using the PC terminal it works like a charm on any network.

**Serial Output:**
09:04:59.066 -> sd $⸮⸮| ⸮l⸮|⸮$⸮c|ǃ⸮⸮{⸮c⸮b⸮⸮og⸮lgn⸮⸮⸮cp⸮⸮dsdrd8⸮g⸮⸮$⸮⸮#g⸮|d⸮⸮c⸮⸮'o⸮ d⸮⸮$`⸮gnd`orǛ⸮'c⸮l{ۓo#⸮d⸮csc⸮⸮⸮⸮`⸮⸮g⸮
09:04:59.188 -> 
09:04:59.188 -> Connecting to Wi-Fi...
09:05:03.099 -> Disconnected from Wi-Fi.
09:05:05.101 -> Connecting to Wi-Fi...
09:05:11.006 -> Connected to Wi-Fi.
09:05:11.006 -> Connecting to MQTT...
09:05:29.455 -> Disconnected from MQTT.
09:05:29.455 -> Connecting to MQTT...
09:05:31.456 -> Connecting to MQTT...
09:05:50.196 -> Disconnected from MQTT.
09:05:50.196 -> Connecting to MQTT...
09:05:52.198 -> Connecting to MQTT...
09:06:10.939 -> Disconnected from MQTT.
09:06:10.939 -> Connecting to MQTT...
09:06:12.939 -> Connecting to MQTT...

**MQTT Broker Logs**
1575774609: New connection from ::1 on port 1883.
1575774609: New client connected from ::1 as d8b45186-739b-415c-907c-c0a112de740c (c1, k60, u'usrname').
1575774609: Client d8b45186-739b-415c-907c-c0a112de740c disconnected.
1575774613: Client mosqsub/19715-hostname disconnected.
1575774661: New connection from ::1 on port 1883.
1575774661: New client connected from ::1 as mosqsub/19747-hostname (c1, k60, u'usrname').
1575774673: Client mosqsub/19747-hostname disconnected.
1575774721: New connection from ::1 on port 1883.
1575774721: New client connected from ::1 as mosqsub/19808-hostname (c1, k60, u'usrname').
1575774733: Client mosqsub/19808-hostname disconnected.
1575774781: New connection from ::1 on port 1883.
1575774781: New client connected from ::1 as mosqsub/19829-hostname (c1, k60, u'usrname').
1575774793: Client mosqsub/19829-hostname disconnected.
1575774798: New connection from 27.62.75.50 on port 1883.
1575774798: New client connected from 27.62.75.50 as b1add319-dd6d-493f-8f83-aacba442ef6d (c1, k60, u'usrname').
1575774821: New connection from 182.65.179.59 on port 1883.
1575774821: New client connected from 182.65.179.59 as mosqsub|2464-usrname-lpt (c1, k60, u'usrname').
1575774841: New connection from ::1 on port 1883.
1575774841: New client connected from ::1 as mosqsub/19843-hostname (c1, k60, u'usrname').
1575774853: Client mosqsub/19843-hostname disconnected.
1575774902: New connection from ::1 on port 1883.
1575774902: New client connected from ::1 as mosqsub/19880-hostname (c1, k60, u'usrname').
1575774909: New connection from ::1 on port 1883.
1575774909: New client connected from ::1 as 47f7f600-7cb3-4a2b-963a-0bf9811429c7 (c1, k60, u'usrname').
1575774909: Client 47f7f600-7cb3-4a2b-963a-0bf9811429c7 disconnected.
1575774909: New connection from ::1 on port 1883.
1575774909: New client connected from ::1 as d3f2c2ad-190c-4de3-8ec8-42a5c5622f60 (c1, k60, u'usrname').
1575774909: Client d3f2c2ad-190c-4de3-8ec8-42a5c5622f60 disconnected.
1575774909: New connection from ::1 on port 1883.
1575774909: New client connected from ::1 as 2ed0487a-09a5-4deb-bd25-ff095a801f5a (c1, k60, u'usrname').
1575774909: Client 2ed0487a-09a5-4deb-bd25-ff095a801f5a disconnected.
1575774909: New connection from ::1 on port 1883.
1575774909: New client connected from ::1 as e2f448eb-03cd-47d7-8ba9-a81db59905df (c1, k60, u'usrname').
1575774909: Client e2f448eb-03cd-47d7-8ba9-a81db59905df disconnected.
1575774909: New connection from ::1 on port 1883.
1575774909: New client connected from ::1 as 22ea3ea9-4d8c-4a20-b3dc-562a82ebefd1 (c1, k60, u'usrname').
1575774909: Client 22ea3ea9-4d8c-4a20-b3dc-562a82ebefd1 disconnected.
1575774909: New connection from ::1 on port 1883.
1575774909: New client connected from ::1 as cd4310b7-0584-400c-a1ee-0e783ba29b45 (c1, k60, u'usrname').
1575774909: Client cd4310b7-0584-400c-a1ee-0e783ba29b45 disconnected.
1575774909: New connection from ::1 on port 1883.
1575774909: New client connected from ::1 as 67ad2017-ff3e-406d-b96b-80d8c21c356b (c1, k60, u'usrname').
1575774909: Client 67ad2017-ff3e-406d-b96b-80d8c21c356b disconnected.
1575774913: Client mosqsub/19880-hostname disconnected.
1575774918: New connection from 192.168.1.3 on port 1883.
1575774918: New client connected from 192.168.1.3 as mosqpub|2651-usrname-lpt (c1, k60, u'usrname').
1575774918: Client mosqpub|2651-usrname-lpt disconnected.
[... 56 lines cut ...]
1575775509: New client connected from ::1 as efa33f65-d037-45a8-9d6a-272de90e8518 (c1, k60, u'usrname').
1575775509: Client efa33f65-d037-45a8-9d6a-272de90e8518 disconnected.
1575775509: New connection from ::1 on port 1883.
1575775509: New client connected from ::1 as 1f1d6b3f-e244-43fb-84f5-66a8431566e3 (c1, k60, u'usrname').
1575775509: Client 1f1d6b3f-e244-43fb-84f5-66a8431566e3 disconnected.
1575775509: New connection from ::1 on port 1883.
1575775509: New client connected from ::1 as 330023ad-8361-4ecd-ba2e-8a9d82dbb81b (c1, k60, u'usrname').
1575775509: Client 330023ad-8361-4ecd-ba2e-8a9d82dbb81b disconnected.
1575775509: New connection from ::1 on port 1883.
1575775509: New client connected from ::1 as 3f02cb4c-28e2-4c29-9656-6757eb724f52 (c1, k60, u'usrname').
1575775509: Client 3f02cb4c-28e2-4c29-9656-6757eb724f52 disconnected.
1575775509: New connection from ::1 on port 1883.
[... cut ...]
1575775509: New connection from ::1 on port 1883.
1575775509: New client connected from ::1 as 8fc334ff-3730-4b99-afe7-354ca8af3d91 (c1, k60, u'usrname').
1575775509: Client 8fc334ff-3730-4b99-afe7-354ca8af3d91 disconnected.
1575775513: Client mosqsub/20112-hostname disconnected.
1575775562: New connection from ::1 on port 1883.
1575775562: New client connected from ::1 as mosqsub/20157-hostname (c1, k60, u'usrname').
1575775573: Client mosqsub/20157-hostname disconnected.
^[[D1575775621: New connection from ::1 on port 1883.

bertmelis, 2019-12-08:
I didn't look at the code example but imho there's something going on with the hostname. Why is it not always the same?

jeroenst, 2019-12-08:
Try to leave onMqttConnect empty (or only set a flag for the main loop). Doing things in callbacks is always tricky.

sreeramtkd, 2019-12-09:
> I didn't look at the code example but imho there's something going on with the hostname. Why is it not always the same?

I just hid the hostname for privacy purpose.

sreeramtkd, 2019-12-09:
> Try to leave onMqttConnect empty (or only set a flag for the main loop). Doing things in callbacks is always tricky.

Tried that too still the same behaviour

sreeramtkd, 2019-12-09:
Okie... some improvements I was able to rebuild the whole code from scratch (Fully Featured Example) & it worked until I triggered events using Button & MQTT started crashing, see the Serial Output. Any thoughts ?

`
22:37:24.374 -> Connecting to Wi-Fi...
22:37:29.081 -> Disconnected from Wi-Fi.
22:37:31.077 -> Connecting to Wi-Fi...
22:37:37.170 -> Connected to Wi-Fi.
22:37:37.170 -> Connecting to MQTT...
22:37:37.703 -> Connected to MQTT.
22:37:37.703 -> Session present: 0
22:37:37.703 -> Published GYM is Alive to MQTT: telem/SBUT/GYM
22:37:37.936 -> Subscribe acknowledged by the broker, All good !
22:37:37.936 ->   packetId: 1
22:37:37.936 ->   qos: 1
22:37:37.936 -> Subscribe acknowledged by the broker, All good !
22:37:37.936 ->   packetId: 2
22:37:37.936 ->   qos: 1
22:38:07.134 -> Button Long pressed / Motion Detected
22:38:08.032 -> Motion Detected Or Button Pressed in GYM & ON/OFF= 0
22:38:20.253 -> Turned ON Using Button
22:38:20.486 -> Disconnected from MQTT:Connecting to MQTT...
22:38:23.016 -> Connected to MQTT.
22:38:23.016 -> Session present: 0
22:38:23.016 -> Published GYM is Alive to MQTT: telem/SBUT/GYM
22:38:23.249 -> Subscribe acknowledged by the broker, All good !
22:38:23.249 ->   packetId: 1
22:38:23.314 ->   qos: 1
22:38:23.314 -> Subscribe acknowledged by the broker, All good !
22:38:23.314 ->   packetId: 2
22:38:23.314 ->   qos: 1
22:38:39.897 -> Button Long pressed / Motion Detected
22:38:40.829 -> Motion Detected Or Button Pressed in GYM & ON/OFF= 1
22:38:47.855 -> Turned Off Using Button
22:38:48.121 -> Disconnected from MQTT:Connecting to MQTT...
22:38:50.618 -> Connected to MQTT.
22:38:50.618 -> Session present: 0
22:38:50.618 -> Published GYM is Alive to MQTT: telem/SBUT/GYM
22:38:50.851 -> Subscribe acknowledged by the broker, All good !
22:38:50.851 ->   packetId: 1
22:38:50.851 ->   qos: 1
22:38:50.851 -> Subscribe acknowledged by the broker, All good !
22:38:50.851 ->   packetId: 2
22:38:50.904 ->   qos: 1
`

sreeramtkd, 2019-12-10:
This can be closed for time being, Cleared Arduino build & cache folders and it seems to be connecting now. Possible build issue ? not sure will keep an eye !

closed 2019-12-10 by sreeramtkd

uddhavdave1, 2020-04-28:
Hi sreeram, Are there any updates regarding the issue you were facing? I am also facing similar issue where my `onPublish()` callback never triggers and it returns `Session present: 0`. Any feedback would be helpful.