marvinroger/async-mqtt-client #172: Anyway of Knowing when the mqtt Ping is happening??

jassi00713, 2019-12-21:
Hi..

I am developing an mqtt based application on ESP8266.
One of the routines i am running is keeping a check on free heap in ESP8266.

I notice when there is a mqtt ping going out, there is a dip in free heap.

Is the a callback or function to know when this ping takes place.
I checked out the library and found certain variables like lastClientActivit, lastServerActivity, lastPingRequestTime but they are all private to the library.

Any way of knowing it directly in my application code when this ping takes place?

Thank you
Really hoping for a solution to this

kleini, 2019-12-21:
I have no idea, what you mean by a MQTT ping? If you're talking about the statistics of every Homie device, that is send regularly, then there is an event where you can listen on. See on page https://homieiot.github.io/homie-esp8266/docs/develop-v3/advanced-usage/events/ the event `HomieEventType::SENDING_STATISTICS`.

bertmelis, 2019-12-21:
No, it is not exposed. Underlying lib asynctcp calls _onPoll every 125ms (?) and this lib decides when to send the Ping based on activity and keepalive.

The only way to know is to change the code  and implement it yourself.

jassi00713, 2019-12-22:
ok ... what minimal thing should i do to achieve that?
will making those variables public in the library do it?

jeroenst, 2019-12-22:
You would have to create a callback like onPing and onPingAck, and call them in the library at the point where the ping is send and where the ping ack is received.

jassi00713, 2020-01-09:
Thanx ... i will look into it

luebbe, 2021-02-10:
Closing old issue

closed 2021-02-10 by luebbe