marvinroger/async-mqtt-client #175: How to combine asysncmqttclient with ESP8266HTTPClient?

AlbanT, 2020-01-05:
I'm using the FullyFeatures-ESP8266.ino example file and I try to perform an http get to the url received thru mqtt.
I used the BasicHttpClient.ino example (from the ESP8266HTTPClient library) and that one works for me on my WemosD1. When I add it to FullyFeatures-ESP8266.ino it works but then it runs in the loop with a delay:
`void HttpGet(String url) {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {
    WiFiClient client;
    HTTPClient http;
  
    Serial.print("[HTTP] begin...\n");
    if (http.begin(client, url)) {  // HTTP
      Serial.print("[HTTP] GET...\n");
      // start connection and send HTTP header
      int httpCode = http.GET();
  
      // httpCode will be negative on error
      if (httpCode > 0) {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTP] GET... code: %d\n", httpCode);
  
        // file found at server
        if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
          String payload = http.getString();
          Serial.println(payload);
        }
      } 
      else {
        Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
      }
      http.end();
    } 
    else {
      Serial.printf("[HTTP} Unable to connect\n");
    }
  }
}

void loop() {
  HttpGet("http://www.reptile-addict.nl/test.htm");
  delay(10000);
}`

But when I try to call HttpGet() from the loop() to onMqttMessage() it fails with **[HTTP] GET... failed, error: connection refused**

Any ideas on why this is?

aliusc, 2020-01-20:
Hello, I had same issue few days ago and successfully made it work. Problem was fixed when I stopped calling to update function directly from any place of code but set update flag to 'true' to which is checked in loop and making update call.
Basic logics example:
```
bool checkForUpdate = false;
long lastUpdateTime;

void UpdateFirmware() {
  if (checkForUpdate) {
    checkForUpdate = false;
    /* perform http update actions */
  }
}

void loop() {
  if (millis() > lastUpdateTime+10000) {
    lastUpdateTime = millis();
    checkForUpdate = true;
  }
  UpdateFirmware();
}
```

I think that may be somehow connected with unfinished http request, not prepared headers etc. When I call it not in any place of loop but by flag it works as desired. by the way, working with such libraries as http etc try to avoid delay command as it may block simultaneous processes. Good luck with your code.

tokra, 2020-11-07:
I am fighting with same issue: Any HTTP call to some remote server simply fails from onMqttMessage().
I tried both WifiClient example, also HttpClient - they both work when calling from main loop.
I would like to call worldtimeapi.org when i receive MQTT message. Is this bug or simply not possible feature ?
Thank you in advance!

Pablo2048, 2020-11-07:
Async mean that the callback is... well asynchronous. So you simple can not use synchronous objects like HttpClient (and delay() for example) from inside callback. Best practice is set some sort of flag, or put the received data into some queue and handle this flag and/or the queue content in the loop().

tokra, 2020-11-07:
@AlbanT @Pablo2048 Finally found solution using: https://github.com/boblemaire/asyncHTTPrequest

closed 2021-07-25 by bertmelis