marvinroger/async-mqtt-client #176: Esp32 over websocket?

soheils2, 2020-01-14:
Hi is there any way your code connect mqtt over "ws://" cause this is what I've looked everywhere and found nothing with any library or add-on!

Can you help me with this? 

******Ps:*****
I also ran to some s.hit from espressif  called mqtt-clinet.h. That i didn't succed to run it correctly (somehow event handlesr are not implimented !
Im tired of searching and founding literally nothing 😔

https://github.com/espressif/arduino-esp32/issues/3639

soheils2, 2020-01-17:
nothing?

bertmelis, 2020-01-17:
this lib doesn't support websocket connections.

jigneshk5, 2020-04-05:
@bertmelis Do you know any arduino library that support websocket connections for Mqtt?

bertmelis, 2020-04-05:
You can use the lib in IDF for esp32. It even has secure websockets. Example here: https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt

closed 2021-01-27 by bertmelis