marvinroger/async-mqtt-client #177: ESP32 loses connection periodically

Clon1998, 2020-01-25:
Hi folks,
I run into the problem that for the last three weeks, all of my Esp32 always disconnected after beeing connected for 3hr 20min and 30sec with Reason `TCP_DISCONNECTED `. They reconnect and after 3hr 20min and 30 sec they disconnect again.

I verified that all Esp32 use a unique Client Id.
My MqttBroker is running on my Pi in my local Network.

So anyone has an idea why they always disconnect with the same period?
Changing the keepAlive to a higher/lower value did not change anything.

obrain17, 2020-01-29:
I had exactly the same issue. Only with ESP32 not with ESP8266, but with both async-mqtt-client and pubsubclient, different keepAlive:  
Connection regularly disconnected after ~ 3 1/2 hr and reconnected.
And then it vanished one weekend in October 2019, Friday evening to be exact.
No "client disconnected" in mosquitto logs anymore

I tried to find out what had changed that weekend ? End of DST ? No - this was on Sunday.
The only thing I could imagine was: I had removed MQTT_BRIDGE from my FHEM running on the same Pi as mosquitto and replaced it with MQTT_GENERIC_BRIDGE. 

Since then no disconnects anymore. I did not try, however, to do a check if they would come back if I revert to MQTT_BRIDGE.

dx168b, 2020-02-15:
On my ESP8266 the same problem. It gets even worse if I forcefully disconnect the connection to the broker (restarting the mosquitto service) - in this case, AsyncMqttClient starts randomly connecting and disconnecting from the broker. In some cases, he may not even subscribe to the topics.

```
[WiFI_STA] Connection success!
[WiFI_STA] IP address = 192.168.1.40
[WiFI_STA] NETMASK = 255.255.255.0
[WiFI_STA] GATEWAY = 192.168.1.1
[WiFI_STA] DNS = 192.168.1.1
[MQTT] config: Loading configuration...
[MQTT] config: clentID = NodeMCU
[MQTT] config: hosthame = server.lan
[MQTT] config: port = 1883
[MQTT] config: keepAlive = 20
[MQTT] config: user = user
[MQTT] config: password = password
[MQTT] config: baseTopic = /NodeMCU
[MQTT] config: Secured connection = disabled
[MQTT] config: Configuration completed

[SYSTEM] Running...
[MQTT] connection: Success connected
[MQTT] subscribing: Subscribed

Stop the Mosquitto service on my server

[MQTT] connection: Disconnected = TCP_DISCONNECTED (0)

Start the Mosquitto service on my server after 10 seconds

[MQTT] connection: Success connected
[MQTT] subscribing: Subscribed
[MQTT] connection: Disconnected = TCP_DISCONNECTED (0)
[MQTT] connection: Success connected
[MQTT] subscribing: Subscribed
[MQTT] connection: Disconnected = TCP_DISCONNECTED (0)
[MQTT] connection: Success connected
[MQTT] connection: Success connected
[MQTT] subscribing: Subscribed
[MQTT] connection: Disconnected = TCP_DISCONNECTED (0)
[MQTT] connection: Success connected
[MQTT] connection: Success connected
[MQTT] connection: Success connected
[MQTT] connection: Disconnected = TCP_DISCONNECTED (0)
[MQTT] connection: Success connected
[MQTT] subscribing: Subscribed
```

dx168b, 2020-02-16:
Hi! I seem to have found the cause of this problem.
This can happen if you still call the "connect ()" method between the first call and the fact that the connection was established.  I solved this problem by blocking reconnections of the connection if the method was already called once.  The corrected library is at the link below.  I also requested to add these changes to this repository.
https://github.com/dx168b/async-mqtt-client

luebbe, 2021-02-12:
Closing, because #180 has been closed

closed 2021-02-12 by luebbe