marvinroger/async-mqtt-client #178: async_tcp task triggers watchdog

Clon1998, 2020-02-02:
In the startup-sequence of my program, I have to publish a lot of values to different topics. The payload of most of these messages only contains around 30 bytes.

I created a separate task that is responsible for doing that and after everything got published it gets deleted.
The problem is, that after some topics were published the task-watchdog gets triggered by the async_tcp task.
I thought maybe it it is a problem to publish that many messages in a short time, so I went ahead and tried only publishing 4 messages each 250ms. The result was that it published fewer messages and still triggered the watchdog.
I went ahead and tried sending only one message every 5000ms and it published even fewer messages and still triggered the watchdog.

So it seems like the number of messages published every second doesn't affect that problem. 
But what causes the watchdog to trigger?

jeroenst, 2020-02-02:
I use qos 1 and wait for ack before publishing next value.

Clon1998, 2020-02-02:
How can I wait for ack?
And what do you mean by "doing tasks in the callbacks"? Do you mean resuming tasks?

Clon1998, 2020-02-02:
I guess i found out what you meant by that.
In my onMqttConnectCallback I called the method that started the setup. That took to long and caused the async_tcp task to stop/trigger the watchdog.
I moved the code to its own Task and start the new task from the onMqttConnectCallback.
Now everything works fine..

closed 2020-02-02 by Clon1998

jeroenst, 2020-02-02:
👍