marvinroger/async-mqtt-client #179: Two brokers connections

brunohorta82, 2020-02-13:
Hello it’s possible to create two connections, I already try on ESp8266 but seems very unstable

bertmelis, 2020-02-13:
It should be possible. What do you mean with very unstable? Does it crash, disconnect, something else?

Maybe another direction: why do you connect to two brokers? Can't you bridge the brokers?

brunohorta82, 2020-02-17:
> It should be possible. What do you mean with very unstable? Does it crash, disconnect, something else?
> 
> Maybe another direction: why do you connect to two brokers? Can't you bridge the brokers?

I need to brokers, one is for the Cloud communication and other is if the user want user one private broker in parallel

kleini, 2020-02-17:
Why does the user want another private broker in parallel? There is still no reason to set up things like that. Instead it would be more useful to connect two brokers and share only the necessary topics. The brokers are able to handle full TCP stacks and multiple TCP connections easily, while ESP based devices not even have a TCP stack, there is only something, that tries to behave like a TCP stack.

bertmelis, 2021-02-11:
It should be possible now, try the develop branch. There are in my memory no things in this code that prevents you from having two clients.

If they're connecting to the same broker (for whatever reason?) you have to set a unique clientid.

bertmelis, 2021-07-22:
There are no singletons in the code to my knowledge. I can't think of a reason why **this library** doesn't allow to create two client objects.

Feel free to reopen, but please give us some more info on what symptoms prevent you from creating two connections.

closed 2021-07-22 by bertmelis