marvinroger/async-mqtt-client #181: ESP8266 not publishing after wifi lost and reconnection

ZoranCom, 2020-02-20:
Sorry, I am totally new at this...

I need to monitor and collect timestamp data when some machine sends ON signal. ESP8266 with AsyncMqttClient lib. is attached to machine and monitors that signal. I have WiFi connection with PC that runs mosquito mqtt and node red.
Everything works perfectly, node red's mqtt-in node receives published timestamp on each event as desired. 
However, sometimes WiFi connection gets lost for couple of minutes (due to quality of signal, interference...).  But, no data of events that **occurred during WiFi disconnected period** is published when ESP automatically connects to WiFi again, so that data gets lost. Timestamps of events that occurred after that are normally published again.
Publish QoS is set to 1, subscribe QoS is set to 2.

Can you comment on this, please!!!
Regards

bertmelis, 2020-02-20:
Obviously you cannot send data when WiFi is disconnected. When a message is sent, a valid packetId is returned. You can check this and build your firmware around this.

ZoranCom, 2020-02-20:
Thanks for yor reply.
I expected that my ESP, once it reconnects after temporary wifi signal loss, will publish all data collected while wifi was not available. Hope that makes sense.

bertmelis, 2020-02-20:
It doesn't. You have to do that yourself.
what I would do:
- define a struct (class?) that holds your data and all properties (topic, payload, qos, flags...)
- define a queue (std::queue?) to hold your struct
- don't publish in your code but add to the queue
- check every loop() if there's something in the queue and publish if so.
- if publish was not succesfull, don't remove from queue.

Obviously, you'd only check your queue when WiFi and MQTT are connected (mind: MQTT can think it's still connected while WiFi is already gone).

ZoranCom, 2020-02-21:
Thanks for your reply, much appreciated!
It's clearer now. Are you aware of some existing project/code in which this or similar kind of procedure was implemented, so I can study it?
Regards!

bertmelis, 2020-02-21:
I'm not aware of any such code, but it's on my todo list. Unfortunately, nothing on the horizon yet.