marvinroger/async-mqtt-client #185: Push new release to Arduino/PlatformIO?

nullstalgia, 2020-04-06:
It's been a couple years since the last Release, and maybe those commits since then would be a good thing to send out to the public?

hachi, 2020-04-16:
Yes, please.

bluelu, 2020-05-18:
Yes, this would be great!

thijstriemstra, 2021-06-09:
+1. ping @marvinroger

marvinroger, 2021-06-10:
`master` is 2 commits ahead of `develop`, and there are merge conflicts, so I'm not able create a new release right now.
@bertmelis thanks for all the work done here 😃 can you merge `master` into `develop`, so we can prepare the release (bump the version) on `develop` before we merge it to `master`?

bertmelis, 2021-06-14:
I'll look into it. Holiday season is coming up though and I've got some catching up to do now everything is opening up again 😄 .

Documentation also has to be updated.

bertmelis, 2021-06-22:
After checking my calendar, I can set myself a deadline by the end of July.

bertmelis, 2021-07-21:
So I'm back.

Few lasts checkes:
- I thought about adding a "clearQueue" method, to be called when **disconnected**. This is not MQTT compliant, but might be handy in some cases.
- API is not changed but it is also more then a bugfix release, so v0.9.0 would fit?
- Anything missing?

We're not launching a new Apollo mission but I also want to make sure we're not missing anything obvious.

proddy, 2021-07-21:
Agree, the develop branch is so much more than a patch release. so 0.9.0 seems appropriate. I've been using the dev branch in all my projects and some are being used by 2000+ users so it's stable enough. If you add functions you'll also need to update the docs and example code which is a bit more housekeeping.

bertmelis, 2021-07-21:
There you have it, I forgot about updating the docs.

bertmelis, 2021-07-25:
Release 0.9.0 will be picked up soon by the Pio library bot (I hope).

closed 2021-07-25 by bertmelis