marvinroger/async-mqtt-client #186: Basic example crashes when wrong port is set (1880 where node-red is running).

olonsoft, 2020-04-14:
Today while testing the "FullyFeatured-ESP8266.ino" example, I made a mistake and instead of using port 1883, I set 1880 where my node-red is running.
The program crashes:
```
Connecting to Wi-Fi...
Connected to Wi-Fi.
Connecting to MQTT...
Publish acknowledged.
  packetId: 21584
Publish acknowledged.
  packetId: 29556

Exception (28):
epc1=0x4020340f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffec20 end: 3fffffb0 offset: 01a0
3fffedc0:  3ffefaec 4bc6a7f0 40202610 402081b4  
3fffedd0:  402035bc 00000000 3ffee538 00000000
3fffede0:  0000001a 00000000 401002d4 000000fe
3fffedf0:  3ffee6c5 3ffee638 401026ce 00000100
3fffee00:  00012646 7fffffff 00002200 3ffefa6c
3fffee10:  00000000 3fffee60 3ffee538 4020345c
3fffee20:  00000000 3fffee60 3ffee538 4020759a
3fffee30:  3ffef86c 2c9f0300 4000050c 3fffc278
3fffee40:  4010000a 3fffc200 00000022 3ffef86c
3fffee50:  3ffefa6c 00000000 3ffee538 40207852
3fffee60:  3ffef464 3ffef474 00000001 3ffeeb48
3fffee70:  3ffeecf0 3ffeedec 3ffefa6c 3ffeeb48
3fffee80:  3ffef86c 3ffeeb2c 3ffeeb2d 402120d0  
3fffee90:  00000c60 00000000 00000020 401009de
3fffeea0:  3ffeecf0 b113a8c0 00000012 3ffef37c
3fffeeb0:  00000000 00000000 00000020 3ffeedec
3fffeec0:  3ffeecf0 3ffefa8a 3ffefa6c 40217b3e
3fffeed0:  00000014 3ffeecf0 00000065 3ffef37c
3fffeee0:  00000000 3ffef30c 00000020 401009a7
3fffeef0:  40229d20 3ffef9d4 3ffef30c 3ffef37c
3fffef00:  00000008 3ffeecf0 3ffefa6c 4020ed6d
3fffef10:  3fffdc80 3ffef30c 3ffef9d4 4020eb90  
3fffef20:  402341ca 3ffef30c 3ffef9d4 402341db
3fffef30:  3ffefa7c 3ffefa6c 00000000 3ffe84e8
3fffef40:  4022994b 00000000 3ffef9d4 4022f9e3
3fffef50:  40000f49 3fffdab0 3fffdab0 40000f49
3fffef60:  40000e19 40001878 00000002 3fffffb0
3fffef70:  3fffff10 aa55aa55 00000001 4010482d
3fffef80:  40104833 00000002 3fffffb0 4023eb48  
3fffef90:  4010000d 00000000 00000000 00000000
3fffefa0:  00000000 3fffef3c 00000000 00000000
3fffefb0:  3fffffc0 00000000 00000000 feefeffe
3fffefc0:  feefeffe feefeffe feefeffe feefeffe
...
3ffffc80:  feefeffe feefeffe feefeffe feefeffe
3ffffc90:  feefeffe feefeffe feefeffe feefeffe  
3ffffca0:  feefeffe 80af1999 feefeffe feefeffe
3ffffcb0:  00000a20 00000000 40240000 ffffff1a
3ffffcc0:  00000a48 80be5999 00000018 00000000
3ffffcd0:  00000000 00000020 4022aaa7 00000001
3ffffce0:  ffffffff 00000000 3ffe8d31 00000008
3ffffcf0:  4022aaf6 3ffef7ec 3ffef7ec 00000000
3ffffd00:  fffffffe 00000000 401015cf 3ffef7ec
3ffffd10:  000000c0 4022383c 00000000 3ffef854  
3ffffd20:  3ffe8be0 40229bb3 3ffef7ec 3ffecfbc  
3ffffd30:  00000000 4021db7b 3ffecfbc 3ffef7ec
3ffffd40:  3ffe8895 0000001c 00000001 0000000a
3ffffd50:  3ffef885 3ffeef44 00000007 3ffed700
3ffffd60:  3ffe8895 3ffe8895 3ffef33f 4021f9bb
3ffffd70:  3ffed6b4 4021f58c 40221bec 3ffed6b4
3ffffd80:  40105111 0000c358 3ffed7e0 00000000  
3ffffd90:  3ffedef0 3ffed7e0 00000001 00000001
3ffffda0:  00000002 00000000 00000020 401001dc
3ffffdb0:  401018c5 40221bec 00000002 40101490
3ffffdc0:  3ffe8be2 4010472f 3ffec378 401001dc
3ffffdd0:  00000001 4010390a 3ffecda8 40101490
3ffffde0:  00000005 00000000 00000020 401001dc
3ffffdf0:  00000002 00000000 00000005 40101490
3ffffe00:  3ffe8be5 40104747 3ffec670 40101490  
3ffffe10:  4010202b 3ffec670 3ffec378 40101490  
3ffffe20:  fffffffc 0003d132 3ffed0b8 4010220c
3ffffe30:  3ffe9478 00000000 00000000 4010220c
3ffffe40:  fffffffc 0003d132 401026ce 00000100
3ffffe50:  3ffe9478 7fffffff 00002200 00000001
3ffffe60:  00000001 00004208 4000050c 3fffc278
3ffffe70:  3ffe9478 00080000 00000022 0003d132  
3ffffe80:  3ffe9484 2c9f0300 4000050c 3fffc278  
3ffffe90:  401023f0 3fffc200 00000022 3fffc278
3ffffea0:  40207004 00000030 0000001b ffffffff
3ffffeb0:  40205ab4 00000000 40205ac4 00000000
3ffffec0:  00000000 00000000 00000000 fffffffe
3ffffed0:  ffffffff 3fffc6fc 00000001 3ffee808
3ffffee0:  00000000 3fffdad0 3ffee848 00000030
3ffffef0:  00000000 3fffdad0 3ffee848 00000030  
3fffff00:  00000000 3fffdad0 3ffee848 00000030
3fffff10:  3ffe8622 00000001 00000010 3ffee848
3fffff20:  40204d8c 3ffee71c 3ffe8802 40204d98
3fffff30:  40204d00 3ffee71c 3ffe8881 00000002
3fffff40:  00000006 00000000 3ffee6e4 00000001
3fffff50:  40207f40 40207f54 3ffee71c 3ffee848
3fffff60:  007a1200 6eb043e4 3ffee500 4020104a  
3fffff70:  00000000 40207f54 3ffee538 4020163f
3fffff80:  00000000 00000000 00000001 401001dc  
3fffff90:  3fffdad0 00000000 3ffee808 401001fd
3fffffa0:  3fffdad0 00000000 3ffee808 40205aed
<<<stack<<<
```
When I set port 1883 it works. When I also set any other not used port, it normally disconnects.

jeroenst, 2020-04-14:
Port 1880 is the web interface of nodered not the MQTT port so that is why...

The crash should not happen, but it in normal situation it will be fine...

olonsoft, 2020-04-14:
> The crash should not happen, but it in normal situation it will be fine...
This crash happened in my application and I couldn't find what was wrong until I noticed the wrong port. Then I run the basic example to be sure that was not my code's bug.
I hope this to be fixed soon.

bertmelis, 2020-04-14:
Is it possible to decode the stack trace? I'm quite interested in how this happens. I can't reproduce.

olonsoft, 2020-04-15:
The code is this:
```
#include <ESP8266WiFi.h>
#include <Ticker.h>
#include <AsyncMqttClient.h>

#define WIFI_SSID "Support"
#define WIFI_PASSWORD "12345678"

#define MQTT_HOST IPAddress(192, 168, 19, 10)
#define MQTT_PORT 1880

AsyncMqttClient mqttClient;
Ticker mqttReconnectTimer;

WiFiEventHandler wifiConnectHandler;
WiFiEventHandler wifiDisconnectHandler;
Ticker wifiReconnectTimer;

void connectToWifi() {
  Serial.println("Connecting to Wi-Fi...");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
}

void connectToMqtt() {
  Serial.println("Connecting to MQTT...");
  mqttClient.connect();
}

void onWifiConnect(const WiFiEventStationModeGotIP& event) {
  Serial.println("Connected to Wi-Fi.");
  connectToMqtt();
}

void onWifiDisconnect(const WiFiEventStationModeDisconnected& event) {
  Serial.println("Disconnected from Wi-Fi.");
  mqttReconnectTimer.detach(); // ensure we don't reconnect to MQTT while reconnecting to Wi-Fi
  wifiReconnectTimer.once(2, connectToWifi);
}

void onMqttConnect(bool sessionPresent) {
  Serial.println("Connected to MQTT.");
  Serial.print("Session present: ");
  Serial.println(sessionPresent);
  uint16_t packetIdSub = mqttClient.subscribe("test/lol", 2);
  Serial.print("Subscribing at QoS 2, packetId: ");
  Serial.println(packetIdSub);
  mqttClient.publish("test/lol", 0, true, "test 1");
  Serial.println("Publishing at QoS 0");
  uint16_t packetIdPub1 = mqttClient.publish("test/lol", 1, true, "test 2");
  Serial.print("Publishing at QoS 1, packetId: ");
  Serial.println(packetIdPub1);
  uint16_t packetIdPub2 = mqttClient.publish("test/lol", 2, true, "test 3");
  Serial.print("Publishing at QoS 2, packetId: ");
  Serial.println(packetIdPub2);
}

void onMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
  Serial.println("Disconnected from MQTT.");

  if (WiFi.isConnected()) {
    mqttReconnectTimer.once(2, connectToMqtt);
  }
}

void onMqttSubscribe(uint16_t packetId, uint8_t qos) {
  Serial.println("Subscribe acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
  Serial.print("  qos: ");
[... 112 lines cut ...]
0x40103d7b: lmacRecycleMPDU at ??:?
0x4010363c: lmacProcessTXStartData at ??:?
0x40101490: pp_post at ??:?
0x40104747: lmacRxDone at ??:?
0x4010202b: rcReachRetryLimit at ??:?
0x4010220c: rcReachRetryLimit at ??:?
0x401026ce: wDev_ProcessFiq at ??:?
0x4000050c: ?? ??:0
0x401023f0: wDev_ProcessFiq at ??:?
0x40100ca4: cont_wrapper at ??:?
0x40205977: loop_task(ETSEventTag*) at core_esp8266_main.cpp:?
0x40100ca4: cont_wrapper at ??:?
0x40205ac4: loop_wrapper() at core_esp8266_main.cpp:?
0x40204d8c: HardwareSerial::write(unsigned char const*, unsigned int) at ??:?
0x40204d98: HardwareSerial::write(unsigned char const*, unsigned int) at ??:?
0x40204d00: AsyncMqttClientInternals::UnsubAckPacket::UnsubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>) at ??:?
0x40207f40: std::_Function_handler<void (unsigned short), void (*)(unsigned short)>::_M_invoke(std::_Any_data const&, unsigned short) at ??:?
0x40207f54: std::_Function_base::_Base_manager<void (*)(unsigned short)>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
0x4020104a: connectToWifi() at ??:?
0x40207f54: std::_Function_base::_Base_manager<void (*)(unsigned short)>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at ??:?
0x4020163f: setup at ??:?
0x401001dc: ets_post at ??:?
0x401001fd: esp_schedule at ??:?
0x40205aed: loop_wrapper() at core_esp8266_main.cpp:?
```