marvinroger/async-mqtt-client #187: onMqttMessage rogue characters

mugginsjm, 2020-04-20:
I am receiving mqtt messages like these:
Message arrived [stat/frontFlood/RESULT]  : {"Var1":"1.406436"}:"1.398073"}ﾭ⸮xV⸮⸮?0>⸮?4⸮⸮
Message arrived [JMscrollT]  : 19.3o(⸮⸮d8⸮G⸮:I⸮ﾭ⸮xV⸮⸮⸮t<⸮?4⸮⸮

The rogue characters contain some elements of previous messages and is probably due to the asynchronous nature of async-mqtt-client.
I don't know how to sort it. Can anyone help?
thanks

jeroenst, 2020-04-20:
You must take the payload length into account.

qcasey, 2020-04-20:
I experienced this on my ESP32, it's due to this issue #104 (also #35 and #42)

[Hacky solution here](https://github.com/marvinroger/async-mqtt-client/issues/104#issuecomment-388981994) that worked for me, essentially copy the payload and ensure it's null terminated.

jeroenst, 2020-04-20:
Payload is not null terminated because payload is not a string but an array of bytes. It doesn't have to be a string, it can also be binary data.

mugginsjm, 2020-04-20:
Maybe I'm not seeing this correctly, but I'm sending strings via mqtt that are of varying lengths. The fixes seem to suggest truncating payload to a fixed length.

bertmelis, 2020-04-20:
As jeroenst already commented, MQTT's payload is binary. It doesn't know c-strings, characters or whatever you write. That's why you get a length along with the data. You even get an index, should the message be split into multiple TCP packets.

So it's not the lib that introduces rogue characters, it's you who is reading parts that are NOT part of the payload.

Easiest is to do `Serial.write(payload, length);`

mugginsjm, 2020-04-20:
OK thanks

closed 2020-04-20 by mugginsjm