marvinroger/async-mqtt-client #194: OTA over MQTT

timurufa86, 2020-05-28:
is it possible to implement this with this library?
if not, how can you do it?

timurufa86, 2020-05-28:
forums suggested 

but I can’t understand how it will work

static void onMqttMessage(char *topic, char *payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {

  // update requested?
  if (strcmp(topic, MQTT_CLIENT "/update") == 0) {
    int _lastError = 0;
    // start update
    if (index == 0) {
      Update.runAsync(true);
      if (!Update.begin(total, U_FLASH)) {
        _lastError = Update.getError();
        Update.printError(Serial);
        // ignore the rest
        topic[0] = 0;
      }
    }
    // do update in chunks
    if (!_lastError) {
      if (Update.write((uint8_t *)payload, len) != len) {
        _lastError = Update.getError();
        Update.printError(Serial);
        // ignore the rest
        topic[0] = 0;
      }
    }
    // finish update
    if (!_lastError) {
      if (index + len == total) {
        if (!Update.end()) {
          _lastError = Update.getError();
          Update.printError(Serial);
        } else {
          // restart?
          ESP.restart();
        }
      }
    }
  }

  // ...
}

luebbe, 2020-05-29:
Take a look a t the homie framework: https://github.com/homieiot/homie-esp8266
It implements OTA over MQTT

closed 2021-02-24 by bertmelis