marvinroger/async-mqtt-client #198: LWT (Last Will and Testament" Example for ESP32 in Arduino IDE?

TSJim, 2020-08-31:
I'm having a difficult time trying to get LWT to work with async-mqtt-client.

I added this line into the "`void onMqttConnect(bool sessionPresent)`" function in the ESP32 example sketch:

     `uint16_t packetIdWill = mqttClient.setWill("myTopicName", 2, true, "ESP32 has disconnected!");`

but I get the error "`cannot convert 'AsyncMqttClient' to 'uint16_t {aka short unsigned int}' in initialization`".

I looked in the library files to see if I could figure out what the "" variable type needs to be, but I couldn't figure it out.  I am not sure what to do to fix this.

Any ideas?

Thanks!

Pablo2048, 2020-08-31:
According to this line https://github.com/marvinroger/async-mqtt-client/blob/7f1ba481a22d56ccf123e4b2f6e555d134c956d0/src/AsyncMqttClient.hpp#L59 `setWill()` function return reference to the `AsyncMqttClient` object. Why do you need `packetIdWill`?

TSJim, 2020-09-02:
Hey Pablo2048...

Thanks, I just wasn't thinking.  I was using the example from the message sending, but you straightened me out...working now!

closed 2020-09-02 by TSJim