marvinroger/async-mqtt-client #203: HEAP issue - general and even more on reconnection attempt

bastianhjaeger, 2020-10-11:
Running an MQTT client on an ESP12f. I noticed that I have a heap issue and after some investigation I found something here.

I created a minimal example and attached the serial output.
Every once in a while without any event I can see, we loose some heap. Once it is 528 byte and once only 40 byte. But after a day or a week the device will crash due to heap. 

```
#include <Arduino.h>

#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <AsyncMqttClient.h>

#define logf(msg, ...) \
  Serial.printf("%d ", millis()); \
  Serial.printf("[%s] ", __FUNCTION__); \
  Serial.printf(msg, __VA_ARGS__); \
  Serial.printf("\n");
#define log(msg) \
  Serial.printf("%d ", millis()); \
  Serial.printf("[%s] ", __FUNCTION__); \
  Serial.printf(msg); \
  Serial.printf("\n");

AsyncMqttClient mqttClient_;

void sendData() {
  noInterrupts();

  constexpr size_t BUFFER_SIZE = 842;
  DynamicJsonDocument root(BUFFER_SIZE);

  root["time"] = millis();

  root["battery"] = 99;

  JsonObject sensor1 = root.createNestedObject("sensor1");
  sensor1["data1"] = 0.123f;
  sensor1["data2"] = 0.123f;
  JsonObject sensor2 = root.createNestedObject("sensor2");
  sensor2["data1"] = 0.123f;
  sensor2["data2"] = 0.123f;

  char jsonChar[BUFFER_SIZE];

  serializeJson(root, jsonChar);

  interrupts();

  if ( mqttClient_.connected() ) {
    mqttClient_.publish("data", 0, false, &(jsonChar[0]));
  }
}

void OnWiFiEvent(WiFiEvent_t event) {
  switch(event) {
  case WiFiEvent_t::WIFI_EVENT_STAMODE_CONNECTED:
    log("WIFI connected.");
    wifi_set_sleep_type(LIGHT_SLEEP_T);
    break;
  case WiFiEvent_t::WIFI_EVENT_STAMODE_DISCONNECTED:
    log("WIFI disconnected.");
    break;
  case WiFiEvent_t::WIFI_EVENT_STAMODE_GOT_IP:
    log("WIFI got ip.");
    break;
  default:
    printf("WIFI unhandled event %d", event);
  }
}

void OnMqttConnect(bool sessionPresent) {
  log("Connected");

  mqttClient_.publish("connect", 0, false, "");
}

void OnMqttDisconnect(AsyncMqttClientDisconnectReason reason) {
  const int r = (int8_t)reason;
  logf("Disconnected %d", r);
}

void OnMqttMessage(char* topic, char* payload,
    AsyncMqttClientMessageProperties properties, size_t len, size_t index,
    size_t total) {
  logf("Got %s: '%s'", topic, payload);
}

void setup() {
  pinMode(2, OUTPUT);
  digitalWrite(2, true);

  Serial.begin(115200);

  WiFi.onEvent(OnWiFiEvent, WiFiEvent_t::WIFI_EVENT_STAMODE_CONNECTED);
  WiFi.onEvent(OnWiFiEvent, WiFiEvent_t::WIFI_EVENT_STAMODE_DISCONNECTED);
  WiFi.onEvent(OnWiFiEvent, WiFiEvent_t::WIFI_EVENT_STAMODE_GOT_IP);
  WiFi.begin("jarvis_net", "PasswortNixda123");

  while ( WiFi.status() != WL_CONNECTED ) {
    log("Waiting on Wifi connect");
    delay(1000);
  }

  mqttClient_.setServer("h2891062.stratoserver.net", 1884);
  mqttClient_.setCredentials("pirate", "PirateInnovation123");
  mqttClient_.setClientId("DummyMqttClient");
  mqttClient_.onConnect(
      [] (bool sessionPresent) {OnMqttConnect(sessionPresent);});
  mqttClient_.onDisconnect(
      [] (AsyncMqttClientDisconnectReason reason) {OnMqttDisconnect(reason);});
  mqttClient_.onMessage(
      [] (char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total)
      { OnMqttMessage(topic, payload, properties, len, index, total);});
  mqttClient_.connect();

  while ( !mqttClient_.connected() ) {
    log("Waiting on Wifi connect");
    delay(1000);
  }
}

void loop () {
  logf("heap %d, wifi %d, mqtt %d", ESP.getFreeHeap(), WiFi.isConnected(), mqttClient_.connected());

  if ( !mqttClient_.connected() ) {
    mqttClient_.connect();
  }

  sendData();

  delay(5000);
}
```

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 2.6.2 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - tool-mklittlefs 1.203.200522 (2.3) 
 - tool-mkspiffs 1.200.0 (2.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0

```

```
7074 [loop] heap 45224, wifi 1, mqtt 1
12076 [loop] heap 45224, wifi 1, mqtt 1
17078 [loop] heap 45224, wifi 1, mqtt 1
22080 [loop] heap 45176, wifi 1, mqtt 1
27082 [loop] heap 45176, wifi 1, mqtt 1
32084 [loop] heap 45176, wifi 1, mqtt 1
37086 [loop] heap 45176, wifi 1, mqtt 1
42089 [loop] heap 45176, wifi 1, mqtt 1
47091 [loop] heap 45176, wifi 1, mqtt 1
52093 [loop] heap 45176, wifi 1, mqtt 1
57095 [loop] heap 45176, wifi 1, mqtt 1
62097 [loop] heap 45176, wifi 1, mqtt 1
67099 [loop] heap 45176, wifi 1, mqtt 1
72101 [loop] heap 45176, wifi 1, mqtt 1
77103 [loop] heap 45176, wifi 1, mqtt 1
82105 [loop] heap 45176, wifi 1, mqtt 1
87107 [loop] heap 45176, wifi 1, mqtt 1
92109 [loop] heap 45176, wifi 1, mqtt 1
97111 [loop] heap 45176, wifi 1, mqtt 1
102113 [loop] heap 45176, wifi 1, mqtt 1
107115 [loop] heap 45176, wifi 1, mqtt 1
112117 [loop] heap 45176, wifi 1, mqtt 1
117119 [loop] heap 45176, wifi 1, mqtt 1
122121 [loop] heap 45176, wifi 1, mqtt 1
127123 [loop] heap 45176, wifi 1, mqtt 1
132126 [loop] heap 45176, wifi 1, mqtt 1
137128 [loop] heap 45176, wifi 1, mqtt 1
142130 [loop] heap 45176, wifi 1, mqtt 1
147132 [loop] heap 45176, wifi 1, mqtt 1
152134 [loop] heap 45176, wifi 1, mqtt 1
157136 [loop] heap 45176, wifi 1, mqtt 1
162138 [loop] heap 45176, wifi 1, mqtt 1
167140 [loop] heap 45176, wifi 1, mqtt 1
172142 [loop] heap 45176, wifi 1, mqtt 1
177144 [loop] heap 45176, wifi 1, mqtt 1
182146 [loop] heap 45176, wifi 1, mqtt 1
187148 [loop] heap 45176, wifi 1, mqtt 1
192151 [loop] heap 45176, wifi 1, mqtt 1
197153 [loop] heap 45176, wifi 1, mqtt 1
202155 [loop] heap 45176, wifi 1, mqtt 1
207157 [loop] heap 45176, wifi 1, mqtt 1
212159 [loop] heap 45176, wifi 1, mqtt 1
217161 [loop] heap 45176, wifi 1, mqtt 1
222163 [loop] heap 44648, wifi 1, mqtt 1
227165 [loop] heap 44648, wifi 1, mqtt 1
232167 [loop] heap 44648, wifi 1, mqtt 1
237170 [loop] heap 44648, wifi 1, mqtt 1
242172 [loop] heap 44648, wifi 1, mqtt 1
247174 [loop] heap 44648, wifi 1, mqtt 1
252176 [loop] heap 44648, wifi 1, mqtt 1
257178 [loop] heap 44648, wifi 1, mqtt 1
262180 [loop] heap 44648, wifi 1, mqtt 1
267182 [loop] heap 44648, wifi 1, mqtt 1
272184 [loop] heap 44648, wifi 1, mqtt 1
277186 [loop] heap 44648, wifi 1, mqtt 1
282189 [loop] heap 44648, wifi 1, mqtt 1
287191 [loop] heap 44648, wifi 1, mqtt 1
292193 [loop] heap 44648, wifi 1, mqtt 1
297195 [loop] heap 44648, wifi 1, mqtt 1
302197 [loop] heap 44648, wifi 1, mqtt 1

```

bertmelis, 2020-10-11:
Could you try not to use delay()?
Furthermore, it is not advisable to rely on `.connected()` in the `loop` to reconnect. Better is to reconnect in the `onDisconnect` handler.
In extremis you could try to connect twice and who knows how TCP will react.

While it doesn't solve your problem, it may distract.

```
void loop () {
  static uint32_t lastMillis = 0;
  if (millis() - lastMillis > 5000) {
    lastMillis = millis();

    logf("heap %d, wifi %d, mqtt %d", ESP.getFreeHeap(), WiFi.isConnected(), mqttClient_.connected());

    if ( !mqttClient_.connected() ) {
      mqttClient_.connect();  // <-- move this to onDisconnect
    }

    sendData();

  }
}
```

bastianhjaeger, 2020-10-12:
Thanks, I will try to work on that.
But can you give me some background. I never read somthing about
1. Not rely on the .connected() method. Why is that?
2. I may have use cases where I do not want to reconnect immediately. So I need to call my .connect() in the loop ot in some timer event outside the disconnect event. Is this bad?

bertmelis, 2020-10-12:
`connect()` return immediately and does not block until connected. So when you call `connect()` again before the connection is established, you connect twice because `connected()` is still returning `false`. I don't know how the TCP lib handles that. Will there be two connections? Will the first one be dropped? IF so, will this trigger the `disconnect` callback?

It is perfectly fine (I would even say recommended) to start a timer in the `onDisconnect` callback. When the timer fires, you can connect again.

bastianhjaeger, 2020-10-12:
That dies not solve my heap issues, bit great advises. Thanks!

bertmelis, 2020-10-14:
How much time is there between heap issues?
IS your keepalive still at the default level?

bastianhjaeger, 2020-10-14:
I did not change the keepalive with that code, but I updated my original stuff with those findings and there it seems better. 
As soon as I have a test device more, I will give the sample code with keepalive modification a try. 

About the time: I can not tell you. It did not seem to be deterministic. But I will give you details next time.

bertmelis, 2020-10-14:
How well are your c++ skills? There is a known bug in this lib causing dangling pointers, but it should not lead to heap issues like this.

(fix would be to copy data while writing to sending buffer instead of just passing the pointer)

proddy, 2020-10-16:
I've never had any problems with this library and heap or even heap fragmentation. The problem is likely elsewhere, perhaps in your logf function. Also use StaticJsonDocument for any payloads <1kb. That way they'll won't eat into your heap and leave holes.

bastianhjaeger, 2020-10-16:
@proddy Thanks, I will have a look into it. 

@bertmelis (My c++ skills are quite ok, I would say). 

But it seems that the connection/disconnection/reconnection issue was the reason for the heap. ... I think. 

I modified my code
- keep alive is now bigger
- reconnection is only triggered at disconnect (with a timer)

Now my heap seems stable.

Thanks.

closed 2020-10-16 by bastianhjaeger