marvinroger/async-mqtt-client #204: Whel client-space() get's low

bastianhjaeger, 2020-10-22:
Maybe someone can solve / explain this quickly.

I have the AsyncMqttClient in use. Working so far with and without TLS enabled. But ...

I need to send some 10 - 30 messages (< 60 bytes each) in a callback routine. 
With TLS enables, I run out of client space after 5-10 messages, as it takes arrount 300 byte per message. 

A simple `delay(..)` is not working form me. Crashes happen. 

What can I do to send this many messages "as fast as possible" without running out for client space?

dm2302git, 2021-01-12:
Same problem here. Anyone an Idea?

bertmelis, 2021-01-12:
https://github.com/marvinroger/async-mqtt-client/issues/212

Mind that besides client space, overall memory space can hit a hard limit too.