marvinroger/async-mqtt-client #205: Program gets caught in reconnection loop of death

VanKurt, 2020-11-05:
I'm currently playing with the example code on an ESP32. It works fine at first, but after some time the program gets caught in a loop:

Connecting to MQTT...
Connected to MQTT.
Session present: 0
Connecting to Wi-Fi...
Disconnected from MQTT.
[WiFi-event] event: 5
WiFi lost connection
[WiFi-event] event: 4
[WiFi-event] event: 7
WiFi connected
IP address:
192.168.178.56
Connecting to MQTT...
Connected to MQTT.
Session present: 0
Connecting to Wi-Fi...
Disconnected from MQTT.
[WiFi-event] event: 5
WiFi lost connection
[WiFi-event] event: 4
[WiFi-event] event: 7
WiFi connected
IP address:
192.168.178.56
Connecting to MQTT...
Connected to MQTT.
Session present: 0
...

This happens reliably after some minutes.

cyber-junkie9, 2021-06-01:
is there any fix on this ?
i have to reboot my router or disable WiFi then enable again for 1-2 seconds

luebbe, 2021-06-01:
Please try the develop branch

luebbe, 2021-06-02:
Strange. I use async-mqtt-client as a part of the homie framework for years now and this hasn't happened to me yet. I also have some ESP projects that don't use homie, but async-mqtt-client, e.g. this: https://github.com/luebbe/WordClock. It also uses Arduino OTA and up to now it has always rebooted and reconected after an OTA update.
Not exactly a minimal example, but maybe if you compare how you setup everything to how I do it helps you to find something

nottledim, 2022-06-11:
It really needs a WiFi.disconnect() in the ```case SYSTEM_EVENT_STA_DISCONNECTED``` block otherwise it continually disconnects and reconnects after a disconnect.

olie-o, 2023-01-02:
> It really needs a WiFi.disconnect() in the `case SYSTEM_EVENT_STA_DISCONNECTED` block otherwise it continually disconnects and reconnects after a disconnect.

This seems to have solved this problem for me on ESP32