marvinroger/async-mqtt-client #206: revive project and fix issues

bertmelis, 2020-11-05:
I'm currently occupied on a different project, but I'm getting some free time and I want to spend it on this project. Who can join me in the effort to revive this project?

What I have in mind:
- dangling pointer issue --> add copy flag to TCP writes (is enabled by default for ESP32)
- document issues with QOS > 0 --> sending messages with QOS != 0 is not supported as it should be
- merge the stability improvement PRs (#105 #156 #180)

I don't know if I want to go the distance of implementing QoS2 fully. I don't even know if it is possible without limiting the API wrt large payloads (not fitting in memory).

kleini, 2020-11-05:
I would like to contribute https://github.com/kleini/async-mqtt-client/commit/f1b42054815ad82fed4519bce7febb7f1601560f
Some of my devices have a lot of nodes and publication of them during startup is a size problem for me.

luebbe, 2020-11-05:
I can test on ESP8266 using the homie framework. I also have an unused ESP32 in my shelf, which I can fire up for this purpose.

bertmelis, 2020-11-09:
My (hardware) debugger should arrive today. Then I'm going for it.

I'll put improvements in a separate branch.

proddy, 2020-11-12:
@bertmelis I'd like to help.

I'm using an custom version of the asyncmqtt library with a few of the PRs added (including the TCP copyflag). Like @kleini I'm still fighting with heap and mem fragmentation issues on the ESP8266 and have made only small steps there. For example one of my applications publishes 70+ MQTT topics for Home Assistant (part of their MQTT discovery component) which almost cripple's the ESP8266, even if its gracefully queued. I have an ESP-prog board for debugging an ESP32 but sadly no JTAG for the 8266.

bertmelis, 2020-11-12:
Made #207 to copy data instead of keeping pointers only. Please review.

I remember a lot of people struggle with the last will topic and payload as these are only sent upon connection and not on "setup". Is is preferable to copy this data and store this in the lib or leave it as is?

proddy, 2020-11-12:
> I remember a lot of people struggle with the last will topic and payload as these are only sent upon connection and not on "setup". Is is preferable to copy this data and store this in the lib or leave it as is?

think most people worked around it, either by having a static or global string. Not high priority for me.

Here's an idea, create separate GitHub issues for the all enhancements, stick them in a github kanban project so we can prioritize and assign to contributors.

bertmelis, 2020-11-12:
> Here's an idea, create separate GitHub issues for the all enhancements, stick them in a github kanban project so we can prioritize and assign to contributors.

I'm not up to speed with Github and it's features. I'll see if I can manage to do that.

closed 2020-12-26 by bertmelis