marvinroger/async-mqtt-client #208: Compilation error on Arduino IDE

bronks75, 2020-11-16:
Hello,
no success on compilation of async-mqtt-client with Arduino IDE.
Tried with example and my own code, same result.

Thanks for your help.
`
libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.literal._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x54): undefined reference to `AsyncMqttClientInternals::ConnAckPacket::ConnAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (bool, unsigned char)>)'
libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.literal._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x58): undefined reference to `AsyncMqttClientInternals::PingRespPacket::PingRespPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void ()>)'
libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.literal._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x5c): undefined reference to `AsyncMqttClientInternals::SubAckPacket::SubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short, char)>)'
libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.literal._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x60): undefined reference to `AsyncMqttClientInternals::UnsubAckPacket::UnsubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.literal._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x64): undefined reference to `AsyncMqttClientInternals::PublishPacket::PublishPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short)>, std::function<void (unsigned short, unsigned char)>)'
libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.literal._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x68): undefined reference to `AsyncMqttClientInternals::PubRelPacket::PubRelPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.literal._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x6c): undefined reference to `AsyncMqttClientInternals::PubAckPacket::PubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.literal._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x70): undefined reference to `AsyncMqttClientInternals::PubRecPacket::PubRecPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries\AsyncMqttClient\AsyncMqttClient.cpp.o:(.literal._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x74): undefined reference to `AsyncMqttClientInternals::PubCompPacket::PubCompPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries\AsyncMqttClient\AsyncMqttClient.cpp.o: In function `AsyncMqttClient::_onData(AsyncClient*, char*, unsigned int)':
R:\Livier\Documents\Arduino\libraries\AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::ConnAckPacket::ConnAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (bool, unsigned char)>)'
Plusieurs bibliothèque trouvées pour "WiFi.h"
R:\Livier\Documents\Arduino\libraries\AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::PingRespPacket::PingRespPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void ()>)'
Utilisé : C:\Users\Livier\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
Non utilisé : C:\Program Files (x86)\Arduino\libraries\WiFi
R:\Livier\Documents\Arduino\libraries\AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::SubAckPacket::SubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short, char)>)'
R:\Livier\Documents\Arduino\libraries\AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::UnsubAckPacket::UnsubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
R:\Livier\Documents\Arduino\libraries\AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::PublishPacket::PublishPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short)>, std::function<void (unsigned short, unsigned char)>)'
R:\Livier\Documents\Arduino\libraries\AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::PubRelPacket::PubRelPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
R:\Livier\Documents\Arduino\libraries\AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::PubAckPacket::PubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
R:\Livier\Documents\Arduino\libraries\AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::PubRecPacket::PubRecPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
R:\Livier\Documents\Arduino\libraries\AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::PubCompPacket::PubCompPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
collect2.exe: error: ld returned 1 exit status
exit status 1
`

Vfgandara, 2020-12-08:
I had a somewhat similar mistake when trying to compile a project using espmake:
```
$ espmake BOARD=nodemcuv2
cont.S
core_esp8266_flash_quirks.cpp
core_esp8266_wiring_analog.cpp
base64.cpp
cencode.cpp
cdecode.cpp
libc_replacements.cpp
Crypto.cpp
flash_hal.cpp
core_esp8266_wiring_digital.cpp
spiffs_api.cpp
time.cpp
Stream.cpp
cbuf.cpp
Updater.cpp
TypeConversion.cpp
Esp-version.cpp
core_esp8266_phy.cpp
gdb_hooks.cpp
sqrt32.cpp
MD5Builder.cpp
heap.cpp
Esp.cpp
core_esp8266_flash_utils.cpp
core_esp8266_noniso.cpp
core_esp8266_wiring.cpp
crc32.cpp
core_esp8266_wiring_pulse.cpp
FS.cpp
core_esp8266_wiring_pwm.cpp
spiffs_cache.cpp
spiffs_nucleus.cpp
spiffs_gc.cpp
spiffs_check.cpp
spiffs_hydrogen.cpp
core_esp8266_app_entry_noextra4k.cpp
core_esp8266_eboot_command.cpp
core_esp8266_spi_utils.cpp
Tone.cpp
StreamString.cpp
HardwareSerial.cpp
core_esp8266_si2c.cpp
debug.cpp
core_esp8266_features.cpp
sntp-lwip2.cpp
uart.cpp
Schedule.cpp
core_esp8266_wiring_shift.cpp
IPAddress.cpp
Print.cpp
WMath.cpp
core_esp8266_postmortem.cpp
umm_info.c
umm_local.c
umm_malloc.cpp
umm_poison.c
umm_integrity.c
core_esp8266_timer.cpp
FunctionalInterrupt.cpp
core_esp8266_main.cpp
core_esp8266_i2s.cpp
FSnoop.cpp
StackThunk.cpp
core_esp8266_waveform.cpp
abi.cpp
Esp-frag.cpp
WString.cpp
[... 83 lines cut ...]
/home/vfgandara/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/vfgandara/Arduino/libraries/async-mqtt-client-master/src/AsyncMqttClient.cpp:353: undefined reference to `AsyncClient::add(char const*, unsigned int, unsigned char)'
/home/vfgandara/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/vfgandara/Arduino/libraries/async-mqtt-client-master/src/AsyncMqttClient.cpp:628: undefined reference to `AsyncClient::send()'
/home/vfgandara/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /tmp/mkESP/cpdigifirmware_nodemcuv2/user_obj.ar(AsyncMqttClient.cpp.o): in function `AsyncMqttClient::_sendDisconnect()':
/home/vfgandara/Arduino/libraries/async-mqtt-client-master/src/AsyncMqttClient.cpp:670: undefined reference to `AsyncClient::space()'
/home/vfgandara/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/vfgandara/Arduino/libraries/async-mqtt-client-master/src/AsyncMqttClient.cpp:685: undefined reference to `AsyncClient::add(char const*, unsigned int, unsigned char)'
/home/vfgandara/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/vfgandara/Arduino/libraries/async-mqtt-client-master/src/AsyncMqttClient.cpp:683: undefined reference to `AsyncClient::send()'
/home/vfgandara/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /home/vfgandara/Arduino/libraries/async-mqtt-client-master/src/AsyncMqttClient.cpp:685: undefined reference to `AsyncClient::close(bool)'
/home/vfgandara/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /tmp/mkESP/cpdigifirmware_nodemcuv2/user_obj.ar(AsyncMqttClient.cpp.o): in function `AsyncMqttClient::disconnect(bool)':
/home/vfgandara/Arduino/libraries/async-mqtt-client-master/src/AsyncMqttClient.cpp:727: undefined reference to `AsyncClient::close(bool)'
/home/vfgandara/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /tmp/mkESP/cpdigifirmware_nodemcuv2/user_obj.ar(AsyncMqttClient.cpp.o): in function `std::vector<std::function<void (unsigned short)>, std::allocator<std::function<void (unsigned short)> > >::~vector()':
/home/vfgandara/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_vector.h:418: undefined reference to `AsyncClient::~AsyncClient()'
/home/vfgandara/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /tmp/mkESP/cpdigifirmware_nodemcuv2/user_obj.ar(AsyncMqttClient.cpp.o): in function `AsyncMqttClient::~AsyncMqttClient()':
/home/vfgandara/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_construct.h:102: undefined reference to `AsyncClient::~AsyncClient()'
/home/vfgandara/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /tmp/mkESP/cpdigifirmware_nodemcuv2/user_obj.ar(AsyncMqttClient.cpp.o):(.text._ZN15AsyncMqttClient9_sendAcksEv+0x1b): undefined reference to `AsyncClient::space()'
/home/vfgandara/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /tmp/mkESP/cpdigifirmware_nodemcuv2/user_obj.ar(AsyncMqttClient.cpp.o): in function `AsyncMqttClient::_sendAcks()':
[... cut ...]
/home/vfgandara/CTA/makeEspArduino/makeEspArduino.mk:328: recipe for target '/tmp/mkESP/cpdigifirmware_nodemcuv2/cpdigifirmware.bin' failed
make: *** [/tmp/mkESP/cpdigifirmware_nodemcuv2/cpdigifirmware.bin] Error 1
```
I know this probably won't help you in any way, but I'm hoping that by also commenting this error here the developer may be more inclined to take a look at this.

I can't be 100% sure that this is solely an error with async-mqtt-client, but I think it might be since as long as I don't create any instance of the AsyncMqttClient class the program compiles.

bertmelis, 2020-12-26:
I cannot reproduce this. I must admit I only use Platformio to build my firmware.

bertmelis, 2021-02-08:
I just tested on Arduino IDE 1.8.13

I couldn't reproduce the error. Feel free to reopen if the issue persists.

closed 2021-02-08 by bertmelis