marvinroger/async-mqtt-client #209: ESP8266 keeps connecting and disconnecting with reason **TCP_DISCONNECTED**

Abhimanyu6676, 2020-12-04:
On my ESP8266 the same problem. It gets even worse if I forcefully disconnect the connection to the broker (restarting the mosquitto service) - in this case, AsyncMqttClient starts randomly connecting and disconnecting from the broker. In some cases, he may not even subscribe to the topics.

```
[MQTT] connection: Success connected
[MQTT] subscribing: Subscribed
[MQTT] connection: Disconnected = TCP_DISCONNECTED (0)
[MQTT] connection: Success connected
[MQTT] subscribing: Subscribed
[MQTT] connection: Disconnected = TCP_DISCONNECTED (0)
[MQTT] connection: Success connected
[MQTT] connection: Success connected
[MQTT] subscribing: Subscribed
[MQTT] connection: Disconnected = TCP_DISCONNECTED (0)
[MQTT] connection: Success connected
[MQTT] connection: Success connected
[MQTT] connection: Success connected
[MQTT] connection: Disconnected = TCP_DISCONNECTED (0)
[MQTT] connection: Success connected
[MQTT] subscribing: Subscribed
```

re3lex, 2020-12-15:
yeah, same issue here. 
There is no issue with WiFI connection (I can reach running http server on ESP8266), as mqtt broker is accessible.

bertmelis, 2020-12-15:
Does the broker's log say "socket error"? (in case of mosquito)

re3lex, 2020-12-16:
Yes. 

I suspect something wrong with TCPAsync library but  not with mqtt client.

bertmelis, 2020-12-16:
You can add a third parameter to every `_client.add()` statement in this lib (this lib, not the asynctcp) link this for example:
https://github.com/marvinroger/async-mqtt-client/blob/7f1ba481a22d56ccf123e4b2f6e555d134c956d0/src/AsyncMqttClient.cpp#L313

`_client.add(fixedHeader, 1 + remainingLengthLength);`
becomes
`_client.add(fixedHeader, 1 + remainingLengthLength, 0x01);`

Adding this parameter makes the TCP library copy the data instead of taking a reference to the data and hance avoids invalid pointer problems like you're suffering from.

PS This flag is added by default on ESP32 but not on ESP8266. So ESP32 does not suffer from this issue.

re3lex, 2020-12-16:
@bertmelis 
> So ESP32 does not suffer from this issue.
well, looks like I see the same issue on my ESP32 setup too. I didn't dig it in detail though, but as per behavior mqttClient is disconnecting time to time. Added `ESP.restart()` for  now as quick workaround.

Anyway thanks for tip. I will apply it and report back about results.

BTW: found that if I call `mqttClient.setWill()` before calling `mqttClient.connect()` then mqttClient cannot connect to server in 100% cases.

re3lex, 2020-12-16:
> Anyway thanks for tip. I will apply it and report back about results.

yeah, that fixed the issue looks like!
I will test it further but mqtt server restart doesn't cause infinite disconnect-connect loop after this fix.

Thanks, @bertmelis !

bertmelis, 2020-12-16:
The will issue is something that might need better documentation.

The will topic and payload is not saved within the library. It is also not "used" when yo user it, or even on calling connect.

So you have to make sure the pointer you passed on to the lib is still valid throughout the life of the MQTT client object. Easiest is to make it a global or static variable.

re3lex, 2020-12-16:
Yeah, already found out that at least Will topic name must be declared as global variable if it is generated dynamically.

thanks for your help!

closed 2020-12-26 by bertmelis

Gornerrrrr, 2024-05-21:
> @bertmelis
> 
> > So ESP32 does not suffer from this issue.所以 ESP32 不会遇到这个问题。
> 
> well, looks like I see the same issue on my ESP32 setup too. I didn't dig it in detail though, but as per behavior mqttClient is disconnecting time to time. Added `ESP.restart()` for now as quick workaround.嗯，看来我在 ESP32 设置上也看到了同样的问题。虽然我没有详细挖掘它，但根据行为 mqttClient 会不时断开连接。现在添加为快速解决方法。
> 
> Anyway thanks for tip. I will apply it and report back about results.不管怎样，谢谢你的提示。我将应用它并报告结果。
> 
> BTW: found that if I call `mqttClient.setWill()` before calling `mqttClient.connect()` then mqttClient cannot connect to server in 100% cases.顺便说一句：发现如果我在调用之前调用，那么 mqttClient 在 100% 的情况下无法连接到服务器。

> [re3lex](/re3lex)

Same problem in 2024. 
Calling 'mqttClient.setWill()' and `mqttClient.connect()` then mqttClient cannot connect to server. 
Error message "asyncTCP: pcb is NULL".