marvinroger/async-mqtt-client #213: broken WiFi-connection will not handled while publish a new message

arduz, 2021-01-22:
Hello, i tried this library with ESP32 D1 mini. At first this works fine. But then i tested the error handling. My client publishes a new message each 2 seconds. While it works i put the ESP-Board in a metal box - there WiFi was not available. My MQTT broker ist mosquitto, installed on Raspi 4. Via SSH I can see each incoming publish-message on the choosen topic. Immediatly, when i put the board in the metal box I can see, the broker gets no new messages. But the client doesn't feel, that the connection was broken, he sends all 2 senconds a new message. The  mqttClient.publish returns with a valid packetIdPub1 also in the metal box.
I initiated the client with QOS1. Only 10 seconds later he realize, that the connection was broken. More then 5 messages was lost. Thats not, what QOS1 means. How can I monitor, if publish was really succesfull? Yes, I can subscribe to all self published messages and match the received messages with the self published, but this would be handmade-QOS - not my intention!

bertmelis, 2021-01-22:
QoS > 0 is somewhat broken at the moment.
As long as your WiFi is still connected, the client uses the keep alive mechanism to check the connection. Untill the mechanism signals a connection issue, the clients acts as if it was connected. You can change to time this takes be setting the keepalive.

arduz, 2021-01-24:
Hello, what means somewhat broken? I would expect, that a publish statement with QoS1 returns an error, if QoS1 is not suported. I's in opposite to the library description.
But anyway: Even with QoS0 I would suggest, that a .publish fails, if the connection is down (my WiFi board in a metal box). But it doesn't. It's the same behavior, if I use a simple TCP client, which sends each second a message to a server: (standard library WiFi.h). If I put the board in the box the next client.write or client.print statements returns the number of written bytes allthough there is no connetion to the server and nothing was submitted. Not till any seconds he realize, that the connection was broken. How can I check, which messages are certainly transferred? The response codes generated from publish, write or print are unsuitable. Could you help?

bertmelis, 2021-01-24:
What does your program do when WiFi is lost?

arduz, 2021-01-25:
That's the problem - nothing. The program does'nt detect, that WiFi was lost, it sends and sends and so on. See this code-snippet:
WiFiClient client;
sentBytes = 0;
sentBytes = client.print(String(timestamp));
Serial.println(WiFi.status());
if (sentBytes > 0)
  {
  Serial.print("sent ");Serial.print(String(timestamp));Serial.print(" to    ");
  Serial.println(client.remoteIP());  
  }
else
  {
  Serial.print("not success to send "); Serial.println(String(timestamp));
  Serial.print("reconnect to host "); Serial.println(client.connected());
  connectToHost();
  }
sentBytes is allways > 0 also when WiFi is lost. 
I's the same while using MQTT.
uint16_t packetIdPub1 = mqttClient.publish("test", 1, true, message);
The packetIDPub1 was allways > 0.

bertmelis, 2021-01-25:
So your code doesn't detect that WiFi was lost, then how should the mqtt-client know?

Maybe add an "on disconnect" event to the WiFi event handler and disconnect the client? Maybe also use the develop branch. It has a fix to handle half open connections.

arduz, 2021-01-26:
Hello, the actual library has such an event handler, but the event SYSTEM_EVENT_STA_DISCONNECTED was raised to late. After I broke the connection the client has more then 3 messages "successful" sent. This messages wasn't received by the MQTT broker. Not till then these DISCONNECTED event was raised. I assume, the underlaying TCP stack hasn't given the transmission error to the WiFi client. If I see that right the TCP stack works directly with sockets but I can't find details for transmission control in the library. I only have found this interesting detail: void WiFiClient::flush() {
  // TODO: a real check to ensure transmission has been completed} If there is real an open todo this may be the explanation for my problem?

bertmelis, 2021-01-26:
why don't you force disconnect the mqtt client when WiFi disconnects.

and as soon as WiFi connects again, try to reconnect to the broker.

arduz, 2021-01-26:
Sorry, I was wrong. TCP is on layer 4, WiFi on layer 2. Therfore the sequence for error handling must start by WiFi. It must give a broken connection to ICSP (layer 3) and this to TCP. I not understand, because WiFi detects the broken connection so late. Immediatly when I put the sending ESP32-board in a insulating box the message broker gets not any new messages but the client sends for any seconds with "success" without detecting the broken connection.  I'm helpless.

bertmelis, 2021-01-26:
It's because TCP doesn't know yet it is disconnected. And the mqtt client can still fill the outgoing buffer untill the connection is broken.

If you want a faster response, you have to close the connection yourself, in your code.

bertmelis, 2021-01-26:
https://superuser.com/questions/911808/what-happens-to-tcp-connections-when-i-remove-the-ethernet-cable/911829

closed 2021-01-26 by bertmelis

arduz, 2021-01-27:
Thanks for both answers. Now I understand, how it works in an error case. Meanwhile I have found a fulfilling solution: Obviously MQTT registers as first the broken connection. Now I evaluate the onMqttPublish-event before sending a new message. So I can check, if a message was succesful sent or not. (You can see this in the appended protocol: Between Nachricht28 and Nachricht29 I have broken the connection. At first the client was disconnected from MQTT, the publish acknowledged for Nachricht29 was missing. Then he checks, that WiFi was broken and reconnect. Nachricht 29 was stacked, I try to publish it  again. Not before it reconnect succesful to MQTT it gets an publish acknowledged. The subscription client gets each message only once.)
Thanks for your help. I think, you can close this issue.   
[MQTT Protokolle.txt](https://github.com/marvinroger/async-mqtt-client/files/5879702/MQTT.Protokolle.txt)