marvinroger/async-mqtt-client #214: ESP32 fail to publish message to mqtt server but onMqttPublish is executed and returns a valid packetId

elbueno222, 2021-01-27:
I am having an issue with an ESP32 that fails to send messages to the MQTT broker but the code doesn´t realise and it keeps executing and sending more messages as if everything was fine but none of those message are received by the MQTT broker. This issue just happens randomly around 10-15% of the times the ESP32 starts (the rest of the times everything works and the MQTT broker receives every message without issues), but when it happens the only way to solve the issue is to manually restart the ESP32.

Everything in the execution looks fine but the messages don´t appear in the MQTT broker. The connection to wifi and mqtt returns an OK, function mqttClient.connected() returns true, the mqttClient.publish function returns a packetId that is incrementing for every message sent, and even the function onMqttPublish is called and also returns the same packetId, so I don´t have a way to notice the issue within the code to reset the connections or even the ESP32.

bertmelis, 2021-01-28:
Are you sure, the broker returns a PUBACK without having received the message. Allow me to say this is hard to believe.

How would the broker have knowledge of the packetId it is returning? This can only because it has received the message, right?

elbueno222, 2021-01-28:
That is why this issue is driving me crazy. The ESP32 is powered with batteries and I have included this code into the onMqttPublish function to make the led blink when the function is called:

```
void onMqttPublish(uint16_t packetId) {
  Serial.print("Publish acknowledged.");
  Serial.print("  packetId: ");
  Serial.println(packetId);
  for (int i=0; i<packetId; i++){
    digitalWrite(LED_GPIO, HIGH);
    delay(50);
    digitalWrite(LED_GPIO, LOW);
    delay(50);
  }
}
```

This is the only place in the code where the LED blinks, and I can see the led blinking 1 time in the first iteration of the loop, 2 times in the second and so on, what means the function is receiving the packetId 1, 2, 3, etc...

I am using Mosquitto broker within Home Assistant, and to ensure Home Assistant was not having some issue with the messages I actually check the messages with MQTT Explorer and they simply don´t arrive. In the Mosquitto broker logs I can see the "New client connected" message what means the connection happens, but then the messages don´t show up which is really weird. Then without updating the code or changing anything I just restart the ESP32 and 80% of the times after restart the messages arrive to the broker without any issue (until next restart, so it is basically a lottery with no consistency).

bertmelis, 2021-01-28:
There's not much in the puback code in this lib. When the lib receives a puback, it forwards the packet id to the callback in the user code.

You are sending with QoS 1 right?

elbueno222, 2021-01-28:
Yes I am sending the messages with QOS 1, I also tried QOS 2 but the behaviour is the same.

elbueno222, 2021-02-01:
@bertmelis the puback is sent by the broker confirming the message has been received, right? Because if that is the case I guess the issue will be in the Mosquito broker not processing these messages even it sends the puback, am I right?

bertmelis, 2021-02-01:
that's right: http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718101

elbueno222, 2021-02-02:
After doing more testing it seems that the issue happen more often when the ESP32 is connected to batteries and it wakes up from a deep sleep (I have a motion sensor connected to it to wake up). I am just thinking loud because I don´t have enough knowledge about it, but is it possible that the ESP is using some type of old/obsolete session or something like that so the broker doesn´t consider the messages as valid?

This is the part of the code that handles the connection to WIFI and MQTT, just in case someone can spot any issue:

```
void connectToWifi() {
  Serial.println("Connecting to Wi-Fi...");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  WiFi.setHostname(hostname);
}

void connectToMqtt() {
  Serial.println("Connecting to MQTT...");
  mqttClient.connect();
  mqttClient.setClientId(hostname);
}

void WiFiEvent(WiFiEvent_t event) {
    Serial.printf("[WiFi-event] event: %d\n", event);
    switch(event) {
    case SYSTEM_EVENT_STA_GOT_IP:
        Serial.println("WiFi connected");
        Serial.println("IP address: ");
        Serial.println(WiFi.localIP());
        Serial.print("Hostname: \t");
        Serial.println(WiFi.getHostname());
        connectToMqtt();
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        Serial.println("WiFi lost connection");
        xTimerStop(mqttReconnectTimer, 0); // ensure we don't reconnect to MQTT while reconnecting to Wi-Fi
        xTimerStart(wifiReconnectTimer, 0);
        break;
    case SYSTEM_EVENT_STA_START:
        Serial.println("STA Start");
        tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_STA, hostname);
        break;
    }
}

void onMqttConnect(bool sessionPresent) {
    Serial.println("Connected to MQTT.");
    Serial.print("Session present: ");
    Serial.println(sessionPresent);
}
```

luebbe, 2021-02-03:
If an ESP wakes up from deep sleep it should behave like it's powered on. So everything should be fresh, no old, half open connections.

bertmelis, 2021-02-03:
By default, cleanSession is set to `true`. So you should not receive any messages you missed while being asleep.
Also, you have to set the mqtt clientId **before** connecting.

Do you have clean session set to `false`?

elbueno222, 2021-02-03:
> clientId

I don´t set the cleanSession in my code so it will be set to true by default.

I will set the mqtt cliendId before connecting to see if that fix the issue.

elbueno222, 2021-02-04:
@bertmelis after some testing I can say you are my hero, after setting clientId before connecting, no more messages lost!! Thank you very much

closed 2021-02-04 by elbueno222