marvinroger/async-mqtt-client #215: Compile error on Arduino IDE

feddynventor, 2021-01-31:
The library is correctly installed, I tried different core versions and ESPAsyncTCP library is correctly installed and its examples compile correctly. The log shows the bug is trivial, but I don't know how to edit the source code. Has anyone experienced the same compilation issue?

Arduino:1.8.13 (Linux), Scheda:"Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, QIO, 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"
====================================
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x54): undefined reference to `AsyncMqttClientInternals::ConnAckPacket::ConnAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (bool, unsigned char)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x58): undefined reference to `AsyncMqttClientInternals::PingRespPacket::PingRespPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void ()>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x5c): undefined reference to `AsyncMqttClientInternals::SubAckPacket::SubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short, char)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x60): undefined reference to `AsyncMqttClientInternals::UnsubAckPacket::UnsubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x64): undefined reference to `AsyncMqttClientInternals::PublishPacket::PublishPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short)>, std::function<void (unsigned short, unsigned char)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x68): undefined reference to `AsyncMqttClientInternals::PubRelPacket::PubRelPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x6c): undefined reference to `AsyncMqttClientInternals::PubAckPacket::PubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x70): undefined reference to `AsyncMqttClientInternals::PubRecPacket::PubRecPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o:(.text._ZN15AsyncMqttClient7_onDataEP11AsyncClientPcj+0x74): undefined reference to `AsyncMqttClientInternals::PubCompPacket::PubCompPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o: In function `AsyncMqttClient::_onData(AsyncClient*, char*, unsigned int)':
libraries/AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::ConnAckPacket::ConnAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (bool, unsigned char)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o: In function `std::_Function_base::_Base_manager<std::_Bind<std::_Mem_fn<void (AsyncMqttClient::*)(bool, unsigned char)> (AsyncMqttClient*, std::_Placeholder<1>, std::_Placeholder<2>)> >::_M_init_functor(std::_Any_data&, std::_Bind<std::_Mem_fn<void (AsyncMqttClient::*)(bool, unsigned char)> (AsyncMqttClient*, std::_Placeholder<1>, std::_Placeholder<2>)>&&, std::integral_constant<bool, false>)':
libraries/AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::PingRespPacket::PingRespPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void ()>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o: In function `std::_Function_base::_Base_manager<std::_Bind<std::_Mem_fn<void (AsyncMqttClient::*)()> (AsyncMqttClient*)> >::_M_init_functor(std::_Any_data&, std::_Bind<std::_Mem_fn<void (AsyncMqttClient::*)()> (AsyncMqttClient*)>&&, std::integral_constant<bool, false>)':
libraries/AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::SubAckPacket::SubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short, char)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o: In function `AsyncMqttClient::_onData(AsyncClient*, char*, unsigned int)':
libraries/AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::UnsubAckPacket::UnsubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::PublishPacket::PublishPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short)>, std::function<void (unsigned short, unsigned char)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o: In function `function<std::_Bind<std::_Mem_fn<void (AsyncMqttClient::*)(char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, short unsigned int)>(AsyncMqttClient*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>, std::_Placeholder<5>, std::_Placeholder<6>, std::_Placeholder<7>, std::_Placeholder<8>, std::_Placeholder<9>)>, void>':
libraries/AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::PubRelPacket::PubRelPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o: In function `AsyncMqttClient::_onData(AsyncClient*, char*, unsigned int)':
libraries/AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::PubAckPacket::PubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o: In function `_Bind<AsyncMqttClient* const, const std::_Placeholder<1>&>':
libraries/AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::PubRecPacket::PubRecPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
libraries/AsyncMqttClient/AsyncMqttClient.cpp.o: In function `_Head_base<AsyncMqttClient* const, void>':
libraries/AsyncMqttClient/AsyncMqttClient.cpp:107: undefined reference to `AsyncMqttClientInternals::PubCompPacket::PubCompPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>)'
collect2: error: ld returned 1 exit status
exit status 1

luebbe, 2021-02-08:
To me this looks like a linking issue, but it is so badly formatted that my eyes hurt looking at it. 

I don't use the Arduino IDE, so I can't tell you, where to look there. I can only tell you that develop and master branches compile fine in PlatformIO.

bertmelis, 2021-02-08:
I just tested on Arduino IDE 1.8.13

could reproduce the error. Feel free to reopen if the issue persists.

closed 2021-02-08 by bertmelis