marvinroger/async-mqtt-client #219: Please test #develop

bertmelis, 2021-02-08:
I'm been doing some work on the https://github.com/marvinroger/async-mqtt-client/tree/develop branch lately.

It should solve lots of the current issues. The main improvement is that outgoing messages are queued and the memory is only freed after the TCP ack allowes to do so.

- :exclamation: Sending in QoS 0, 1 and 2 is supported for messages as large as memory permits.
- Receiving is supported in all three QoS levels. Messages are still directly passed to the API, in chuncks (you get an index, length and total). QoS 2 messages will be delivered "once completely", meaning that it is possible that you get *parts of QoS 2 messages* more than once. (OTA over MQTT is still possible)
- :exclamation: You don't have to wait for messages to be sent before disconnecting. The disconnect signal will be queued and the client will only disconnect once the previously queued messages are sent.

:exclamation: : although the API remained unchanged, the exlamation mark denotes a change in behaviour.

I'm looking forward to feedback. I do like positive feedback :smile:, but I'm more interested in the negative ones. I still have to go through the list of mandatory statements of the MQTT specification. TLS is also untested but there has been no change in that part of the code.

bertmelis, 2021-02-08:
A sample sketch with the development code:
```
RAM:   [====      ]  36.5% (used 29936 bytes from 81920 bytes)
Flash: [===       ]  28.1% (used 292984 bytes from 1044464 bytes)
```

Master branch
```
RAM:   [====      ]  36.7% (used 30104 bytes from 81920 bytes)
Flash: [===       ]  27.6% (used 288428 bytes from 1044464 bytes)
```

luebbe, 2021-02-08:
I added https://github.com/marvinroger/async-mqtt-client.git#develop to the platformio.ini of my https://github.com/luebbe/homie-node-collection compiled and uploaded the sensor-nodes demo to a d1 mini.

This one contains a lot of nodes and many mqtt messages have to be sent upon startup. I'm running mosquitto on a raspberry Pi 1 and MQTT Explorer on Windows for monitoring.

The only notable difference when the device is reset is that it seems to take longer until all the initial messages have been received by MQTT explorer. I guess this is to be expected since they are queued up. Perhaps before messages were lost when the buffer was full.

I have also added "freeheap" to $stats in my fork of homie-esp2866, so I can monitor if anything runs away there.

The graph looks like this (around 15 minutes of history)
[image]

There is a significant drop directly after the start (expected) and then it is stable at around 30KB.

It is also interesting to see that when the $stats are sent, they now trickle in over the course of 1-3 seconds. With the previous version they came in the same second.

bertmelis, 2021-02-08:
In which qos are you sending?

luebbe, 2021-02-08:
Homie default. I didn't change anything. So probably qos 1?

bertmelis, 2021-02-08:
> Homie default. I didn't change anything. MQTT Explorer says qos 0?

qos subscribing might differ from qos publishing.

Stats are sent qos 1 (https://github.com/homieiot/homie-esp8266/blob/9cd83972f27b394eab8a5e3e2baef20eea1b5408/src/Homie/Boot/BootNormal.cpp#L180). So yes, this is slower than before.

luebbe, 2021-02-08:
The homie convention states https://homieiot.github.io/specification/#qos-and-retained-messages that it uses qos 1, but I have to check the source to confirm. I don't know from where MQTT Explorer gets the qos 0 info.

luebbe, 2021-02-08:
> qos subscribing might differ from qos publishing.
guess so

> Stats are sent qos 1 (https://github.com/homieiot/homie-esp8266/blob/9cd83972f27b394eab8a5e3e2baef20eea1b5408/src/Homie/Boot/BootNormal.cpp#L180). So yes, this is slower than before.

Fine by me. Better slow & certain than quick & dirty

bertmelis, 2021-02-08:
I'll anyhow add speed up to the wishlist.

luebbe, 2021-02-08:
Just FYI, the sensor-demo has 84 subtopics. This is the second largest amount of my homie based applications.

kleini, 2021-02-08:
For me this results in a crash. Please see attached serial output from my [relays-device](https://github.com/kleini/homie-relays-device/). I am using SSL encrypted connection.
`💡 Firmware relays-device (0.0.9)
🔌 Booting into normal mode 🔌
SSL is: 1
{} Stored configuration
  • Hardware device ID: cc50e309ea0b
  • Device ID: heating-control
  • Name: Relay device
  • Device Stats Interval: 60 sec
  • Wi-Fi: 
    ◦ SSID: MySSID
    ◦ Password not shown
    ◦ IP: 192.168.0.196
    ◦ Mask: 255.255.255.0
    ◦ Gateway: 192.168.0.1
  • MQTT: 
    ◦ Host: 192.168.0.6
    ◦ Port: 1883
    ◦ SSL enabled: true
    ◦ Base topic: homie/
    ◦ Auth? yes
    ◦ Username: heating-control
    ◦ Password not shown
  • OTA: 
    ◦ Enabled? yes
  • Custom settings: 
    ◦ adcCorrect: 0.96 (set)
    ◦ battMin: 2.60 (default)
    ◦ battMax: 3.30 (default)
• Internal ADC:
  ◦ Send interval: 30 s
↕ Attempting to connect to Wi-Fi...
✔ Wi-Fi connected, IP: 192.168.0.196
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3fffeb80 end: 3fffffb0 offset: 01a0
3fffed20:  3fff4734 00000000 3ffeefe4 40218b01  
3fffed30:  00000004 00000054 3fff1704 00000000  
3fffed40:  3fff468c 3fff14bc 3ffeefe4 40218c99  
3fffed50:  3fff468c 3fff14bc 3ffeefe4 40218d8b  
3fffed60:  00000001 3ffe87a2 00000000 0000000f  
3fffed70:  3fff1134 e39842fb 368805d0 aedbd6f2  
3fffed80:  7182366a fef3f4bb 9aaf5c15 ef7ff88f  
3fffed90:  00000000 3fff1644 000000ff 00000000  
3fffeda0:  3fffedf0 00000014 00000010 00000000  
3fffedb0:  3fff468c 3fff1644 3fff12bc 40218dc8  
3fffedc0:  3fffedf0 3fff1644 3fff1704 40100179  
3fffedd0:  00000001 3fff16be 00000020 4021bcca  
3fffede0:  3fff468c 3fff1644 3fff12bc 4020135c  
3fffedf0:  00000000 00000000 4bc6a7f0 00000000  
3fffee00:  3ffefb38 00000000 401005aa 7e76c8b4  
3fffee10:  00000000 3fff16be 3fff468c 3fff468c  
3fffee20:  00000000 3fffee60 3fff11bc 4021bf49  
3fffee30:  3fff1644 00000001 00000022 3ffef9a0  
3fffee40:  3fff0092 00000000 00000020 3fff1644  
3fffee50:  3fff468c 00000000 3fff11bc 4021c170  
3fffee60:  3ffefd84 3fff144c 00000000 3ffef9a0  
3fffee70:  3ffefb38 3ffefc34 3fff468c 3ffef9a0  
3fffee80:  3fff1644 3ffef984 3ffef985 4022a600  
3fffee90:  00004a30 00000000 00000020 40100da2  
3fffeea0:  3ffefb38 c4a8a8c0 00000949 3fff01c4  
3fffeeb0:  00000000 00000000 00000020 3ffefc34  
[... 258 lines cut ...]
3ffffee0:  00000000 00000000 00000001 3ffef540  
3ffffef0:  00000000 3fffdad0 3ffef580 00000030  
3fffff00:  00000000 3fffdad0 3ffeec10 40209132  
3fffff10:  3ffef2b8 3fff0380 3ffeec10 40204b94  
3fffff20:  00000000 00000001 3ffff93c 3ffff955  
3fffff30:  3fff053c 00000015 3fff0380 00000003  
3fffff40:  00000000 4bc6a7f0 00001482 3ffef580  
3fffff50:  3ffeeb78 00000000 4bc6a7f0 00000000  
3fffff60:  007a1200 669798eb 40100500 3ffef580  
3fffff70:  00000000 00000000 3ffeec00 40201aa2  
3fffff80:  00000000 00000000 00000001 40100230  
3fffff90:  3fffdad0 00000000 3ffef540 40100251  
3fffffa0:  3fffdad0 00000000 3ffef540 4020dbe1  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld`

bertmelis, 2021-02-08:
Do you run your ESP on 160MHz? A soft watchdog reset on the sys stack during connecting is probably caused by the handshake taking too much time. Off course I can be wrong.

kleini, 2021-02-08:
It is a standard ESP12-F. This time with decoded crash:
[crash.log](https://github.com/marvinroger/async-mqtt-client/files/5946521/crash.log)

bertmelis, 2021-02-08:
gotta look into that. But still, it's a watchdog reset, so something is taking too long.
I workaround could be to not send immediately after connecting but on the first onPoll. You could try with commenting out line 382.
I don't know how long a broker waits to receive a CONNECT before disconnecting.

bertmelis, 2021-02-08:
It's been a while since I've used TLS on the esp8266. How did you solve this error:
```
ESPAsyncTCP/src/ESPAsyncTCP.cpp:1325:31: error: no matching function for call to 'AsyncClient::_recv(tcp_pcb*&, pbuf*&, int)'           
         c->_recv(pcb, p->pb, 0);  
```

luebbe, 2021-02-08:
Still running fine after 3,5 hours. FreeHeap still at 30k, goes down to 29,4k .. 29,6k when values are sent and always goes back to the same max value. 
No SSL.

kleini, 2021-02-08:
> It's been a while since I've used TLS on the esp8266. How did you solve this error:
> 
> ```
> ESPAsyncTCP/src/ESPAsyncTCP.cpp:1325:31: error: no matching function for call to 'AsyncClient::_recv(tcp_pcb*&, pbuf*&, int)'           
>          c->_recv(pcb, p->pb, 0);  
> ```

https://github.com/me-no-dev/ESPAsyncTCP/pull/151 <- that requires a fix from @mhightower83, which is not merged for a 3/4 year now.

bertmelis, 2021-02-08:
Shoot me, with logging enable it gives this output:
```
CONNECTING                          <-- connect call by the user
TCP conn, MQTT CONNECT   <-- TCP is connected (TLS handshake done)
new front #1                              <-- CONNECT packet is added to the queue
snd #1: 101/40                          <-- TCP sends 101 bytes whereas the CONNECT was only 40.
```
So the TCP lib is not returning the actual written payload when using encryption???

kleini, 2021-02-09:
Should TCP library normally return exact the amount of bytes, that have been sent? I saw the similar behaviour some comments above.

bertmelis, 2021-02-09:
It does, but only the non-tls. I guess it returns the encrypted TCP payload size.

bertmelis, 2021-02-09:
@luebbe To speed up Homie you can use `set_tcp_nodelay` to `true` if you use mosquitto as broker. But I added "inflight messages > 1" to the TODO.

bertmelis, 2021-02-09:
I'm thinking about a workaround for TLS. It wont be pretty as it is impossible to know how much data actually has been sent: We can wait to send for the entire TCP buffer to be empty. Then **copy** a fixed amount to LWIP and "ack" our own data.

bertmelis, 2021-02-09:
I've added the hacky solution for TLS. I've tested it with the example in the repo + added a 2000 byte message for chuncked sending.

Did it like described above.

kleini, 2021-02-09:
It works!
[works.log](https://github.com/marvinroger/async-mqtt-client/files/5952924/works.log)

bertmelis, 2021-02-09:
Yes, there is room for improvement. I didn't want to touch the existing code too much. _acked is keeping track of the acked data in the mqtt client. But since we're copying the data to TCP, we can already consider it acked. On non-tls versions, _acked is only incremented when the real zck comes in.

I also realized I can use client->space() instead of the fixed number.

bertmelis, 2021-02-10:
I'm going to give it a rest for a week or two (covid restrictions a eased a bit, whoohoo). I'll keep an eye on the issue and PR tracker and read whatever is going on. I'll keep supporting!

Meanwhile I'll think about possible improvements (like the max inflight messages > 1 for qos > 0).

kleini, 2021-02-10:
Awesome work so far, I must say! Thank you very much @bertmelis !

luebbe, 2021-02-10:
I'll go through old issues backwards and check if I can reproduce them. If not I'll tag them with "to be closed" and we can review together. Won't test TLS related stuff though. I didn't want to jump through that hoop yet.

kleini, 2021-02-10:
@luebbe I can do the TLS related stuff.

bertmelis, 2021-02-12:
> I can do the TLS related stuff.

There might be a serious issue with TLS though. The last thing I was testing was to destroy the client. On non-TLS is was not problem. On TLS it crashed. Now because of a bug in the ESPAsyncTCP library, I had to have the tcp client on the heap to be able to destroy and recreate on each (re)connect. Should destroying the mqttclient's crash be related to the destruction of the tcp client, it will also crash when the mqtt client disconnects. I didn't test this.

It is solveable because the bug is related to acking data and we bypass this when using TLS. However, it involves a lot of preprocesser stuf...

bertmelis, 2021-02-23:
Question: somebody pointed me to the fact that on both ESP8266 and ESP32, data is **copied** to LWIP instead of the data being referenced. So there never were dangling pointers. This means I can severely simplify the `develop` code as I don't need to keep data only for acking (It is still needed for retries etc).

see:
https://git.savannah.nongnu.org/cgit/lwip.git/tree/src/core/tcp_out.c?h=STABLE-2_1_2_RELEASE#n422
https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/7421258237b7c8f61629226961af498a0a6e0096/glue-lwip/arduino/lwipopts.h#L1665
https://github.com/espressif/esp-idf/blob/6e776946d01ec0d081d09000c36d23ec1d318c06/components/lwip/port/esp32/include/lwipopts.h#L449
The COPY flag is set by default with the configuration LWIP is built for ESP8266 and ESP32 and cannot be changed. So When we would keep the data, it will in fact be kept twice.

Hence, I'll simplify the code and keep only the data that has to be acked by the MQTT protocol.

luebbe, 2021-02-24:
Thanks @bertmelis 

My test rig runs stable with develop, but there's not heavy load.

Two weeks ago I wrote:
> I'll go through old issues backwards and check if I can reproduce them. If not I'll tag them with "to be closed" and we can review together.

I did this and tagged a handful of tickets. If someone could review these: https://github.com/marvinroger/async-mqtt-client/labels/to%20be%20closed and close where appropriate?

bertmelis, 2021-02-24:
:+1: I'm currently trying to implement the "don't wait for confirmation before sending the next". This does require some next level thinking. Will take a few days depending on how much time I can spend.

kleini, 2021-02-24:
For me everything works as well. Even OTA over MQTT over TLS improved and works better now.

bertmelis, 2021-02-24:
lovely to hear :smile:

Pablo2048, 2021-02-24:
I also want to report - I have develop branch in 5 devices in two different locations, all of them connected to my MQTT VPN. So far so good - uptime is about 4 days now. VPN network interface statistics on my PC is:
[image]
Everything seems to be working smoothly - OTA from PIO, OTA via web page, web page access, ping, ...
So I can say great job and big thank You...

OldGreyCells, 2021-02-24:
Hi Bert
Thanks for doing this. Only issue I've found so far is the char* payload onMessage is often over length. eg 43 chars instead of 32:
```
Received MQTT topic: homeassistant/localtime 
payload* length: 43 
2021-02-24T20:28:20.236189+00:0�?w␂d␂�� @
```
This is from the master branch:
```
Received MQTT topic: homeassistant/localtime 
payload* length: 32 
2021-02-24T20:33:05.467993+00:00
```
All the best

bertmelis, 2021-02-24:
Strange... I didn't touch that part of the code (not that I can remember, that is :wink:)

However, I did spot some other bugs. On disconnect with a session present, it might block and crash by the watchdog... Fixing it as we speak.

I will look into your issue also!

OldGreyCells, 2021-02-24:
Thank you Bert - apologies for no test script (recovering from Covid, so energy levels are low) but it is a very trivial MQTT payload that doesn't change between async-mqtt-client builds. I'm only publishing and subscribing, so not testing all the features!

bertmelis, 2021-02-24:
no worries, although it would be easier if you could test again with logging enabled. A question though: it is always over length or only sometimes?

OldGreyCells, 2021-02-24:
It appears to always be the same over length (for this particular payload). Forgive my laziness but how do I enable logging? I see log_i and log_w but how to turn it on?

bertmelis, 2021-02-24:
Do you use Arduino IDE or something else (VSCode/Platformio)?

OldGreyCells, 2021-02-24:
I'm using platformio for the first time.

bertmelis, 2021-02-24:
my platformio.ini:
```
[env:esp8266]
platform = espressif8266
board = d1_mini
build_flags =
  -Tesp8266.flash.4m1m.ld
  -D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH
  ;-D ASYNC_TCP_SSL_ENABLED
  -D DEBUG_ESP_PORT=Serial
  -D DEBUG_ASYNC_MQTT_CLIENT
monitor_filters = esp8266_exception_decoder
monitor_speed = 115200
```

proddy, 2021-02-24:
FYI I've started testing on the ESP32 - will report back any issues here

OldGreyCells, 2021-02-24:
> my platformio.ini:
Thank you.

OldGreyCells, 2021-02-24:
I hope this helps (have deleted some chars as they were the bulk of the mqtt broker password!)
A different wrong length with this build.
```
Subscribing to: homeassistant/localtime... SUBSCRIBE
new back #8
snd #8: 30/30
success
PUBLISH
new back #3
ack 30 (30)
data rcv (5)
rcv SUBACK
SUBACK
SUB released
rmv #8
snd #3: 35/35
p #3 rel
ack 35 (35)
data rcv (59)
rcv PUBLISH
Received MQTT topic: homeassistant/localtime 
payload* length: 61 
str::payload length: 61 
2021-02-24T21:56:09.727588+00:00alt����reqũ�� <!!!part of mqtt password!!!>
setLocaltime(): 2021-02-24T21:56:09.727588+00:00alt����reqũ�� <!!!part of mqtt password!!!> len: 61 
 pm open,type:2 0
PING
new back #12
snd #12: 2/2
```

bertmelis, 2021-02-25:
I updated the `develop` code again to simplify and iron out some bugs.

@OldGreyCells You receive data with a total length 59 (that is: including MQTT headers), how do you manage to get a payload of 61 out of it. Could you share your callback code? Mind that the payload is binary and may not include a 0-terminator for c-strings. So you cannot do `strlen(payload)`.

luebbe, 2021-02-25:
> Mind that the payload is binary and may not include a 0-terminator for c-strings. So you cannot do `strlen(payload)`.

Exactly my thoughts.

OldGreyCells, 2021-02-25:
> I updated the `develop` code again to simplify and iron out some bugs.
>
> @OldGreyCells You receive data with a total length 59 (that is: including MQTT headers), how do you manage to get a payload of [...]

Aha! That may well be the cause - I am doing a strlen(payload). My apologies if I've wasted your time - I've been a little* sub-par.

*understatement - this is my 'distraction'.

OldGreyCells, 2021-02-25:
I'm not sure why master didn't highlight my embarrassing "Doh!" but this now sorts my dodgy code out:
```
    char payloadc[len+1];
    for (size_t i = 0; i < len; i++) {
        payloadc[i] = ((char)payload[i]);
    }
    payloadc[len] = '\0';
```
As and aside, is there a way to identify the payload 'type' eg whether it stays a binary stream (eg an image) or text/json?

bertmelis, 2021-02-25:
There probably are, but a bit out of scope for this lib imho. The MQTT protocol doesn't specify anything about the payload format.

An enhancement could be to create a separate file/class with helper methods. It's somewhere on the to do, but without guarantees nor deadline.

OldGreyCells, 2021-02-25:
Thank you for you patience Bert. A comment on the example(s) ```onMqttMessage()``` might help. I kinda just looked at ```char* topic``` and ```char* payload``` in the middle of the night and thought: "They're just regular char arrays" - although I did wonder why the payload wasn't logged. RTFM as bedtime reading is underrated.

bertmelis, 2021-02-25:
Willdo

I'm after another bug now:
```
Exception (9):
epc1=0x4020205e epc2=0x00000000 epc3=0x00000000 excvaddr=0x000002d9 depc=0x00000000

LoadStoreAlignmentCause: Load or store to an unaligned address
  epc1=0x4020205e in AsyncMqttClient::publish(char const*, unsigned char, bool, char const*, unsigned int, bool, unsigned short) at ??:?
```

OldGreyCells, 2021-02-25:
Eeek! That looks way above my pay grade!

proddy, 2021-02-25:
I'm getting regular crashes too on the ESP32 so getting my esp-prog debugger out to find where it's bombing.

Pablo2048, 2021-02-25:
Same problem here - I've just replaced "old" working develop with the new one, and got periodic:
```
Exception (9):
epc1=0x40225596 epc2=0x00000000 epc3=0x00000000 excvaddr=0x000006bf depc=0x00000000

LoadStoreAlignmentCause: Load or store to an unaligned address
  epc1=0x40225596 in AsyncMqttClient::publish(char const*, unsigned char, bool, char const*, unsigned int, bool, unsigned short) at lib/async-mqtt-client-develop/src/AsyncMqttClient.cpp:744

```
crashes...

luebbe, 2021-02-25:
Wouldn't it make sense to convert all the char* payloads into byte* to avoid the confusion that we are dealing with strings?

bertmelis, 2021-02-25:
> Wouldn't it make sense to convert all the char* payloads into byte* to avoid the confusion that we are dealing with strings?

That would indeed make sense.

bertmelis, 2021-02-25:
> Same problem here - I've just replaced "old" working develop with the new one, and got periodic:
>
> ```
> Exception (9):
> epc1=0x40225596 epc2=0x00000000 epc3=0x00000000 excvaddr=0x000006bf [...]

Thank you, here, the line number isn't displayed. Furthermore, when I enable logging, the error is gone.

Now, if the line number is correct, it is on the `packetId()` call. It's a virtual call which returns the value `_packetId` when pointed to a Publish packet or zero when there's no method overloaded. Mind that the base class does not have a member variable _packetId. Am I doing something wrong here?

Pablo2048, 2021-02-25:
Well the complete stack trace is:
```
0x40225548 in AsyncMqttClient::publish(char const*, unsigned char, bool, char const*, unsigned int, bool, unsigned short) at lib/async-mqtt-client-develop/src/AsyncMqttClient.cpp:744
0x40260b90 in mqtt_if_output(netif*, pbuf*, ip4_addr const*) at lib/MQTTVPN/src/MQTTVPN.cpp:72
0x4025a69d in cnx_update_bss_more at ??:?
0x40260bc4 in mqtt_if_output(netif*, pbuf*, ip4_addr const*) at lib/MQTTVPN/src/MQTTVPN.cpp:78
0x40100e3f in umm_free_core at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:351
0x40101108 in free at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:398
0x402501d4 in mem_free at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:237
0x40254459 in ieee80211_parse_beacon at ??:?
0x4024f6a8 in ip4_output_if_opt_src at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:1764
0x40258de7 in sta_input at ??:?
0x402393ee in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, long) at .pio/libdeps/esp12e/ESPAsyncTCP/src/ESPAsyncTCP.cpp:741
0x4024f6f0 in ip4_output_if_opt at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:1577
0x402501d4 in mem_free at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:237
0x4024f716 in ip4_output_if at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:1550
0x4024fd1d in icmp_input at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/icmp.c:291
0x40100e3f in umm_free_core at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:351
0x4024f451 in ip4_input at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:1473
0x40246090 in esp2glue_ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:469
0x4026cb33 in ethernet_input at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:373
0x40260b25 in mqtt_if_Task(ETSEventTag*) at lib/MQTTVPN/src/MQTTVPN.cpp:182
0x40105029 in call_user_start_local at ??:?
0x4010502f in call_user_start_local at ??:?
0x4010000d in call_user_start at ??:?
0x401007c8 in ets_post at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:177
0x40101143 in malloc at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:552
0x4022ddeb in esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::addDomainCacheItem(void const*, bool, unsigned short) at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS/src/LEAmDNS_Structs.cpp:2441 (discriminator 1)
0x40235300 in operator new(unsigned int) at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/abi.cpp:39
0x4022ddeb in esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::addDomainCacheItem(void const*, bool, unsigned short) at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS/src/LEAmDNS_Structs.cpp:2441 (discriminator 1)
0x4022eb8d in esp8266::MDNSImplementation::MDNSResponder::_writeMDNSRRDomain(esp8266::MDNSImplementation::MDNSResponder::stcMDNS_RRDomain const&, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp:1342
0x4022ed71 in esp8266::MDNSImplementation::MDNSResponder::_writeMDNSServiceDomain(esp8266::MDNSImplementation::MDNSResponder::stcMDNSService const&, bool, bool, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp:1431
0x4022e882 in esp8266::MDNSImplementation::MDNSResponder::_udpAppend8(unsigned char) at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp:1112
0x4022e9e9 in esp8266::MDNSImplementation::MDNSResponder::_write8(unsigned char, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp:1254
0x4022ed20 in esp8266::MDNSImplementation::MDNSResponder::_writeMDNSServiceDomain(esp8266::MDNSImplementation::MDNSResponder::stcMDNSService const&, bool, bool, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp:1425
0x40101143 in malloc at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:552
0x402501b4 in mem_malloc at /local/users/gauchard/arduino/arduino_esp8266/esp8266-lwip/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:210
[... cut ...]
0x401008c2 in millis at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_wiring.cpp:188 (discriminator 3)
0x401007c8 in ets_post at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:177
0x401007e9 in esp_schedule at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:125
0x40235740 in loop_wrapper() at /home/pablo2048/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:19
```
but - because it's from my MQTTVPN I don't think that it will help...

[26 more replies]

closed 2021-07-25 by bertmelis